hoylabs/OpenDTU-OnBattery #730: live data partially not auto-updated when reloading page

IngoEF, 2024-03-08:
### What happened?

when page update happens the Live-Data page doesn't actualize all values 

based on too fast closed issue #728 which continues (cant find how to re-open)

### To Reproduce Bug

 verify OpenDTUs live-data page and find different data when comparing upper with lower part 

### Expected Behavior

shoud update the whole data not only some values

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

2024.02.19.post1 to 2024.03.07

### Relevant log/trace output

```Shell
see picture in comment and attachment
```

### Anything else?

[image]
[OpenDTU-onBattery-Live.pdf](https://github.com/helgeerbe/OpenDTU-OnBattery/files/14538910/OpenDTU-onBattery-Live.pdf)

Solarteur, 2024-03-08:
Two screenshots taken **without (manually) reloading** the page (2024.03.07, Firefox 123.0.1)

[image]
[image]

From my point of view everything is ok.

schlimmchen, 2024-03-08:
> based on too fast closed issue https://github.com/helgeerbe/OpenDTU-OnBattery/issues/728 which continues (cant find how to re-open)

You can't re-open other people's issues.

@IngoEF I am very sure that, and others do confirm, that this problem is resolved in version 2024.03.07. Please double-check that this version is indeed installed on your ESP32 (see Info -> System in the web app).

If you continue to have this issue, please try using Ctrl + F5 (Strg + F5) on your keyboard to force Firefox to reload the web application. There is a slim chance that your Browser tries to be clever and it may use the outdated version of the web application.

closed 2024-03-08 by schlimmchen

IngoEF, 2024-03-10:
Issue continued even on 2 other PCs and different browser.
Got time today to go back to version 2024-02-09 (b3d8e984). 
Browser reloads now work as expected.
I find closing this issue is kind of arrogant. I did not open it for joke.

schlimmchen, 2024-03-10:
> I find closing this issue is kind of arrogant.

Yeah, well, somebody needs to start closing issues someday, or otherwise this beautiful project will be flooded with stale issues.

I really don't know whether I should argue with you or just ignore you. The latter would probably be the sane option... However, let me try to lecture you a little bit of decency.

You do realize that I took the time answer to your issue and gave options to try? I did not just close the issue for no reason.

> based on too fast closed issue https://github.com/helgeerbe/OpenDTU-OnBattery/issues/728 which continues

How are you deciding whether or not the issue was closed too soon? Did you realize that the reporter of the issue marked it as solved and closed it, not a maintainer? That guy's problem was indeed solved, he told us himself (maybe herself, don't know...). And did you realize that two other users confirmed that the issue is gone in 2024.03.07? So maybe, just maybe, the issue is with something you did wrong?

> I do not depend on newer versions so for me this issue might continue as you wish.

Now this is just childish, is it not?

> Install Method
> Self-Compiled

This better not turn out as another "You forgot to compile the webapp before building the firmware", because that would be on you alone. Do not self-compile the firmware if you don't understand what you are doing.

Have you compiled the webapp before building the firmware?

Use the binaries from the [release page](https://github.com/helgeerbe/OpenDTU-OnBattery/releases) and you will find that your issue is indeed gone.

IngoEF, 2024-03-10:
Issue might be gone, can't test because none of the binaries works for my fusion-v2-board.
When building firmware by myself the way is (works best since at least one year with nearly every update)
git pull in CLI
then in VS-Code 
clean
build
then in OpenDTU web upload new firmware.bin
Where are webapps to be built exclusivly in this process (would be new -at least to me- and should be well documented)?
The issue remains the same. Only automatic page updates for the inverter values, but not for mppt (victron on pins 44+44) nor Leistung/Netz. 
Only after pressing "Reload" or "Ctrl-F5" or cleaning cache and restart browser.
All works fine in 20240209

schlimmchen, 2024-03-10:
> none of the binaries works for my fusion-v2-board

You mean [this one](https://shop.allianceapps.io/products/opendtu-fusion-community-edition)? That would be the [opendtu-onbattery-generic_esp32s3_usb.bin](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/download/2024.03.07/opendtu-onbattery-generic_esp32s3_usb.bin) release, which works out of the box. Use a respective [pin_mapping.json](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/docs/DeviceProfiles/opendtu_fusion.json), which is also available without a need for modifications. Don't tell me something different, I use this board myself. You should not rely on the opendtufusionv2 environment to be around forever.

> works best since at least one year with nearly every update

Yeah, sorry, I refuse to commit a new version of the transpiled web app for each change to the web application source files, and I broke your workflow. I also refuse to commit a binary version of the firmware for each source code change, because it makes no sense. I don't know why tbnobody does it (sometimes).

> and should be well documented

[Like so](https://www.opendtu.solar/firmware/compile_webapp/)?

> The issue remains the same.

Yes, because you did not [compile the web application](https://www.opendtu.solar/firmware/compile_webapp/) before you compiled the firmware.

IngoEF, 2024-03-11:
forget my question, i've found the solution

IngoEF, 2024-03-11:
worked.
But I had to fetch a new git clone and make webapp before compiling the firmware