hoylabs/OpenDTU-OnBattery #732: Live-Data: Grid Power wird nicht aktualisiert

cerise21, 2024-03-08:
### What happened?

Grid Power -Wert wird nur bei manuellem Reload aktualisiert.

### To Reproduce Bug

Live-Data

### Expected Behavior

Aktualisierung des Grid-Power-Werts alle 15s, so wie in der Console.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

2024.03.07

### Anything else?

Verwendetes Powermeter: SDM630 via RS485-Adapter.
Möglicherweise wird auch der MPPT Total Power-Wert nicht aktualisiert.
Die Inverter-Werte werden jetzt zuverlässig aktualisiert (im Gegensatz zum vorigen Release).

schlimmchen, 2024-03-08:
Das mag ich nicht so recht glauben, auch wenn du jetzt der zweite bist, weil immer noch deutlich mehr als zwei schon zurückgemeldet haben, dass es mit dem gestrigen Release (wieder) einwandfrei funktioniert.

Kannst du einmal sicherstellen, dass du wirklich die 2024.03.07 installiert hast (Screenshot?)?

Passiert dir das auf mehreren Geräten? Wenn ja auf welchen?

Kannst du den Browser einmal schließen und neu öffnen und/oder Strg + F5 benutzen und zurückmelden, ob das Problem dann verschwindet? Der Browser könnte eine alte Version der WebApp ge-cache't haben.

> Verwendetes Powermeter: SDM630 via RS485-Adapter.

Wenn du im Log neue Werte siehst, dann sollten die auch in der WebApp ankommen. Trotzdem danke für diese Info, vielleicht liegt es doch am Typ des PowerMeters.

cerise21, 2024-03-08:
[image]

Neulich habe ich hier irgendwo gelesen, dass beim Compilieren die Webapp ggf. nicht aktualisiert wird und man das separat machen muss; ist das hier der Fall?

Hmm, denke schon, dass die Version aktuell ist; in der Konsole gibt's auch etliche neue, mir bisher unbekannte Log-Meldungen des DPL.

Bisher getestet unter Android 14, Graphene OS, Vanadium 122 und Android, Fire OS 5.1.7.0, Firefox 124.

cerise21, 2024-03-09:
Jetzt, mit Sonnenlicht, kann ich bestätigen, dass der MPPT Total Power-Wert auch nicht aktualisiert wird (im Live Data -Block).

schlimmchen, 2024-03-09:
> Jetzt, mit Sonnenlicht, kann ich bestätigen, dass der MPPT Total Power-Wert auch nicht aktualisiert wird (im Live Data -Block).

Das ist der gleiche Fehler.

Hast du die Firmware selbst gebaut oder ein Release heruntergeladen?

Bei den Releases wird die Webapp jeweils gebaut und dann erst mit einkompiliert (es sei denn ich hab was übersehen). Aber es gäbe keine Bestätigungen, dass alles gut sei wenn der github Build einen Defekt hätte.

cerise21, 2024-03-09:
Die Firmware ist selbst compiliert.

Ich habe Deinen Beitrag [zum separaten Bauen der  Webapp gefunden](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/685#issuecomment-1964164736), es so erstmals gemacht und mit der anschließend neu compilierten Firmware werden die MPPT-Werte in 'Live Data' wieder aktualisiert.

Auch die 'Grid Power'-Werte werden aktualisiert, allerdings nur sporadisch, längst nicht so oft, wie in der Konsole zu sehen, und nur selten auf genau die gleichen Werte wie in der Konsole (teilweise liegen die 'Live'-Werte in der Nähe der Konsolen-Werte, teilweise haben die gar nichts miteinander zu tun, ab und zu kommt tatsächlich der gleiche Wert).
Meist wird der Live-Wert nach ca. 1min aktualisiert, manchmal innerhalb weniger Sekunden, manchmal dauert es mehrere Minuten — in der Konsole kommen zuverlässig ca. alle 15s neue Werte.

schlimmchen, 2024-03-09:
Recherche bei mir ergibt, dass ich wohl den Buffer für das JSON zu klein gewählt habe in https://github.com/helgeerbe/OpenDTU-OnBattery/blob/c42d68812c57aa5153d8e44507d032a51ce0c97c/src/WebApi_ws_live.cpp#L102

Verdopple das doch mal testweise und sag uns, ob das Problem dann bei dir verschwindet.

> Kann man eigentlich irgendwo in den Quellen (leicht) erkennen, ob vor dem Compilieren auch ein Neubau der Webapp notwendig ist?

Leider weiß ich das nicht. Wenn es möglich wäre, würde ich vorschlagen platformio vor dem Bau der Firmware auch die Webapp bauen zu lassen, wenn es denn nötig ist. Das aber immer zu machen ist lästig. Das wird der Grund sein, warum es tbnobody so gelöst hat, stelle ich mir vor.

cerise21, 2024-03-09:
> Verdopple das doch mal

Ja, so ist es deutlich besser: Alle Powermeter-Werte, die in der Konsole erscheinen, werden auch in die 'Live Data'-Ansicht übernommen. Allerdings tauchen zwischen diesen Werten ab und zu noch andere Powermeter-Werte im Live-View auf, die nichts mit dem aktuellen Verbrauch zu tun haben (also bevor vom Powermeter überhaupt neue Werte eingetroffen sind).

schlimmchen, 2024-03-09:
Okay, damit ist das sicher, dass dieser Buffer größer werden muss. Pflege ich ein.

Warum du Phantasiewerte sieht, kann ich dir nicht erklären. Bist du dir sicher, dass das bis zuletzt nie so war? Würde schon helfen zu wissen, falls das ein älterer Fehler ist.

Wie funktioniert dieser Power-Meter? Iteriert der die Phasen nacheinander und könnte es passieren, dass ein Power-Meter Wert an die WebApp geschickt wird, während die einzelnen Phasen abgeholt werden, sodass der Wert inkonsistent ist?

cerise21, 2024-03-09:
> Warum du Phantasiewerte sieht, kann ich dir nicht erklären. Bist du dir sicher, dass das bis zuletzt nie so war? Würde schon helfen zu wissen, falls das ein älterer Fehler ist.

Du hast Recht, ich habe testweise Releases aus dem Januar und Februar installiert und es kamen auch dort, asynchron zu den Konsolen-Werten, andere Grid Power-Werte im Live View an. Und zwar nur selten genau die gleichen Werte — wohingegen mit der aktuellen Version alle Werte, die in der Konsole mit 'PowerMeterClass: TotalPower: xxx' erscheinen auch in den Live View übernommen werden (zwischen diesen kommen dann halt ab  und zu noch andere).

> Wie funktioniert dieser Power-Meter?

Puh, das kann ich nicht sagen… Könnte durchaus sein, was Du sagst, aber ich durchschaue den Code da nicht…

schlimmchen, 2024-03-10:
> Du hast Recht, ich habe testweise Releases aus dem Januar und Februar installiert und es kamen auch dort, asynchron zu den Konsolen-Werten, andere Grid Power-Werte im Live View an.

Danke fürs Testen. Dann konzentrieren wir uns in diesem Issue nun darauf, dass dir unplausible Werte präsentiert werden unter Verwendung deines Power-Meters. Der Rest ist geklärt.

kloppy1984, 2024-03-10:
Seit dem letzten Uodate muss ich gefühlt 3x Täglich den Wechselrichter neustarten, damit die DTU mit Einspeisen anfängt.
Obwohl der Akku über 50% geladen ist, startet dir DRU nicht die Einspeisung.
Die DTU schaltet mir zu oft den Wechselrichter ab. Auch wenn  keiner zuhause ist, bei einer Grundlast von 120W. 

Komischerweise blinkt der Wechselrichter nicht wie im Standby Rot, sondern langsam Grün.

cerise21, 2024-03-11:
> Der Rest ist geklärt.

Vielen Dank schon mal!

Es könnte durchaus sein, dass die 'Phantasiewerte' Verbrauchswerte sind, die zwischen zwei in der Konsole geloggten Powermeter-Werten gemessen werden (?), denn meistens liegen die zusätzlichen Werte in der Nähe der Konsolen-Werte. Nur selten sind die Werte sehr weit von den Konsolen-Werten entfernt — das könnten aber auch kurzzeitige Überschwinger sein.

Allerdings war ich bisher der Meinung, dass das SDM ca. alle 15s abgefragt wird; wo kommen dann aber zusätzliche Messwerte her?
(Ich habe versucht, diese 15s im Code zu finden, bin aber nicht fündig geworden…)

Kann ich noch irgendwie helfen bzw. etwas zum Thema beitragen?

schlimmchen, 2024-03-11:
Ach herrje, dieser SDM code ist... fragwürdig. Das wundert mich sehr, dass sich Benutzter dieser Schnittstelle nicht schon häufiger über diverse Probleme beschwert haben. Also das wird jede Phase einzeln abgefragt, und jede Abfrage ist eine *synchrone* serielle Kommunikation, die die Haupt-Loop blockiert. Sehr spannend...

Jedenfalls ist es absolut kein Wunder, dass du dieses Verhalten siehst. Alle zehn Sekunden werden alle aktuellen Werte für die Live-Ansicht übertragen, und es ist absolut plausibel, dass es häufig passiert, dass diese Abfrage des Power Meters dann einen Moment trifft, in dem mindestens eine aber noch nicht alle drei Phasen abgefragt wurden. Die Ausgabe auf der Konsole erscheint aber immer erst nach einer vollständigen Runde an Abfragen, weshalb die einigermaßen konsistent ist. Nur einigermaßen, weil zwischen den Abfragen auch signifikante Zeit vergeht und die Leistungswerte daher nie so recht zum gleichen Zeitpunkt gemessen wurden.

Da bleibt nichts anderes übrig, als die ganzen Abfragen möglichst zügig hintereinander zu erledigen und die Ergebnisse aber zwischenzuspeichern und dann (am besten geschützt durch einen Mutex) gesammelt in die jeweiligen Variablen zu schreiben, sodass en Zugriff auf `getPowerTotal()` konsistente Werte liefert.

Ich bin mir nur unschlüssig, ob ich das wirklich noch mit einem Pflaster versehen soll, wo der PowerMeter eigentlich dringend ein Refactoring bräuchte...

Ach komm, lass ein Pflaster basteln.

Schau mal in #749 und spiele die dort in der Build-Pipeline gebaute Firmware ein (Abschnitt Artefakte in https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8240226059?pr=749) und sag uns, ob dein PowerMeter weiterhin funktioniert und ob er jetzt konsistente Werte liefert. Jedenfalls sollte im LiveView nun nichts mehr auftauchen, was nicht auch in der Konsole steht. Dass die Leistungswerte der einzelnen Phasen nie so recht zum gleichen Zeitpunkt bestimmt sind, kann ich nicht ändern, das ist dem Protokoll geschuldet -- oder wenigstens der Implementierung in der SDM Lib (die ich nicht anfassen möchte).

cerise21, 2024-03-12:
Oh, super, dass Du direkt hier weiter machst!
Kann ich die #749 -Firmware auch selbst bauen, indem ich ausgehend von Deinem letzten Commit (2ba57e25f5a6c41e55c3baa5173854c5aa75b76d) compiliere?
Ich benötige nämlich ein anderes als das [default-Pinmapping für SDM_RX/TX_PIN](https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/547).

schlimmchen, 2024-03-12:
Klar! Vergiss nicht die Web App zu bauen^^

> Ich benötige nämlich ein anderes als das https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/547.

Ja, das müssen wir auch noch angehen...

cerise21, 2024-03-12:
Hmm, habe den branch 2ba57e25f5a6c41e55c3baa5173854c5aa75b76d ausgecheckt, dort die Web App  und dann die Firmware gebaut: 
Nach Ctrl+F5 bleibt die Live Data -Seite minutenlang leer, dann erscheinen Werte. Allerdings scheinen die Daten von Laderegler, Akku, Huawei und Wechselrichter nur einmalig gelesen worden zu sein; die Data Age -Zähler werden nicht mehr zurückgesetzt.
Der Powermeter-Wert bleibt auf 0W.
In der Konsole kommen gar keine Meldungen mehr an.
Testweise habe ich den branch 784e3694828df6adc533d5e14b9661968257bbd4 gebaut: diese Firmware funktioniert gefühlt wie c46980d6af918fa90e7ec3cbc015c7bfcd490e6d.

schlimmchen, 2024-03-12:
Ohje... Also es ist möglich, dass ich deine Zeit verschwende. Falls dem so ist, tut mir das leid. Lass mich morgen mal schauen, nicht dass ich einen doofen Fehler gemacht habe...

cerise21, 2024-03-12:
Kein Problem, so ein Test tut nicht weh —  bin froh, dass Du Dich überhaupt diesem Thema angenommen hast…

schlimmchen, 2024-03-13:
So. Ich dachte mir ja schon, dass ich versuchte das Mutex rekursiv zu sperren, aber ich musste jetzt erst einmal genau hinsehen, bevor es offensichtlich wurde, wo das passiert.

Bau doch mal b101ce4434979cf574efe89b4457bed75163ea04 mit den Anpassungen für die Pins und versuch es nochmal.

Jedenfalls ist dieser grobe Schnitzer jetzt raus. Die vorherige Version hat bei mir ehrlich gesagt auch nicht funktioniert, ich war mir meiner Sache vorgestern aber so sicher, dass ich nur das Kompilieren geprüft habe... Sorry.

cerise21, 2024-03-14:
Mit b101ce4434979cf574efe89b4457bed75163ea04 sind die Live-Daten wieder da und die Konsole spricht wieder.
Die Powermeter-Werte der Konsole erscheinen auch in den Live-Daten.
Allerdings erscheinen zwischen zwei Konsolen-Werten meist noch zwei zusätzliche Leistungswerte im Live-View, selten nur ein zusätzlicher Wert oder gar keiner.

schlimmchen, 2024-03-14:
Danke fürs Testen! :heart: 

> Allerdings erscheinen zwischen zwei Konsolen-Werten meist noch zwei zusätzliche Leistungswerte im Live-View, selten nur ein zusätzlicher Wert oder gar keiner.

Aber es ist der zuletzt auf der Konsole ausgegebene Wert? Und wie hast du das festgestellt?

Die Wwebsocket-Implementierung für die Live-Daten schickt alle 10 Sekunden alle Daten. Wenn dein PowerMeter seltener abgefragt wird, dann kommt es vor, dass im Websocket der gleiche Power-Meter-Wert zweimal landet, bevor dein PowerMeter tatsächlich einen neuen Wert abholt.

cerise21, 2024-03-14:
Von der Möglichkeit, sich per Websocket mit OpenDTU zu verbinden, bzw. von websocat habe ich [hier erfahren](https://github.com/tbnobody/OpenDTU/discussions/1775). 

Sorry, aber mir ist nicht klar, was genau zu untersuchen ist, bzw. ob die zusätzlichen Powermeter-Werte, die ich  in den Live-Daten sehe nun ok sind, oder nicht auftauchen sollen.
> Das ist richtig so, das soll so sein.

Ich war der Meinung, dass der Websocket `ws://admin%xxx_@opendtu_ip/console` exakt die Ausgabe der Konsole der Webapp (Menü Info|Console) liefert; mit dem Vorteil, dass live gefiltert werden kann. (?)

Ich habe also die in der Konsole auftauchenden Powermeter-Werte (via websocat) mit denen in den Live-Daten verglichen und in den Live-Daten tauchen zusätzliche, sich von den Konsolen-Werten unterscheidende Werte auf.

> Die einzige verbleibende spannende Frage wäre, ob alle Werte im Live-View bzw. im websocat (das muss dasselbe sein) die gleichen sind seit der letzten Ausgabe in der Konsole. 

Ist das 
> (das muss dasselbe sein) 

eine Forderung, die wir beweisen wollen, oder ist das eine Tatsache? (Live-View = websocat)

Hier bin ich abgehängt:
> die gleichen sind seit der letzten Ausgabe in der Konsole.
> Wenn das nicht so ist, ist das ein Fehler. Aber falls im websocat so lange dieselben Werte kommen, bis in der Konsole ein neuer Wert auftaucht, dann ist alles wie erwartet.

Meinst Du mit 'Konsole' auch 'Info|Console' der Webapp?
Das ist doch dann zwangsläufig identisch mit der websocat-Ausgabe?

Ich sehe z. B. Folgendes:

`websocat/Konsole   |   Live-Daten`

`PowerMeterClass: TotalPower: 290.31`
`   |   290.3`
`   |   301.5`
`   |   298.2`
`PowerMeterClass: TotalPower: 292.09`
`   |   292.1`
`   |   289.4`
`   |   315.0`
`PowerMeterClass: TotalPower: 311.11`
`   |   311.1`
`PowerMeterClass: TotalPower: 314.45`
`   |   314.5`
`   |   301.9`
`PowerMeterClass: TotalPower: 298.55`
`  ...`

Darf das so sein?

schlimmchen, 2024-03-14:
Ne, da stimmt dann was nicht.

Aber du hast ja offenbar einen langen Atem, vielen Dank dafür! Dann lass uns mal weiter schauen.

Ich verstehe, was du mit dem websocat gemacht hast. Du siehst da genau den Inhalt, der auch auf der Seite Info -> Console ausgegeben wird. Und das ist wiederum genau das, was auch auf der seriellen Schnittstelle über USB ausgespuckt wird. Da landen alle "printfs". Und wenn du es wie angedeutet mit grep filterst, siehst du natürlich nur die Zeilen, die deinem Filter entsprechen.

Ich hatte gedacht du würdest mit dem websocat auf dem Websocket für die Live-Daten horchen und die Konsole live lesen im Browser oder am seriellen Port. Die Live-Daten siehst du an `websocat ws://<ip>/livedata` (Authentifizierung (admin%...@) brauchst du nicht, es sei denn du hast unter Einstellungen -> Sicherheit "Nur-Lese-Zugriff auf die Weboberfläche ohne Passwort zulassen" tatsächlich ausgeschaltet.

Im Livedata Websocket siehst du dann auch schön, dass da immer mal wieder Teil-Daten kommen, und alle zehn Sekunden kommt "alles", oder jedenfalls "mehr". Und ich sehe, dass es noch einen Fehler gibt, der aber nichts hiermit zu tun hat, dennoch danke dass ich den jetzt sehe, weil ich websocat einmal auf das livedata angesetzt habe :+1: 

So. Jedenfalls sollte im Live-View nicht öfter als in der Konsole ein neuer Wert auftauchen. Das ist nicht richtig. Und ich hab dafür leider keine vernünftige Erklärung.

Schalte mal Verbose Logging möglichst überall aus und beobachte dann nochmal. Vielleicht gehen Nachrichten den PowerMeters verloren weil irgendein Puffer voll ist.

Kommen die Ausgaben vom PowerMeter in einem sehr regelmäßigen Abstand? Das sollten sie nämlich.

Kannst du einen Weg finden, die Angaben mit Zeitstempel zu versehen, sodass man vielleicht aus der zeitlichen Abfolge etwas ableiten kann?

cerise21, 2024-03-15:
Danke für die ausführliche Erklärung — denke, jetzt hab’ ich’s kapiert, wo das Missverständnis lag.
Ah, ok, die Live-Daten lassen sich auch per websocket anzapfen — gut.
Also, mit 
`websocat -t - timestamp:ws://opendtu_ip/livedata |grep --line-buffered power_meter | (while read -d ' ' a b; do echo -n "$b  grid:  "; echo "$a" |jq '.["power_meter"] | .["Power"] | .["v"]'; done) | tee livedata.log`
und 
`websocat -t - timestamp:ws://opendtu_ip/console |grep --line-buffered -i powermeter |tee console.log`
habe ich Live-Daten und Konsole mitgeschnitten und mit
`cat livedata.log console.log | sort >livedata_console_sorted.log`
beide Logs zusammengefügt und sortiert:
[livedata_console_sorted.log](https://github.com/helgeerbe/OpenDTU-OnBattery/files/14619078/livedata_console_sorted.log)

Zunächst dachte ich, mich laust der Affe, keine zusätzlichen Powermeter-Werte mehr in den Live-Daten.
Einziger Unterschied zum letzten Test: Inverter aus, da keine Sonne und Akku leer.
Nach Einschalten des Inverters (timestamp: 1710522400) waren sie tatsächlich wieder da, meine Freunde…
Nach erneutem Deaktivieren des Inverters (timestamp: 1710522701) ist dann wieder alles paletti.

Jetzt frage ich mich: Stört der Wechselrichter irgendwie physikalisch die Datenübertragung vom Powermeter oder macht die Software etwas unerwartetes?
Liegt’s möglicherweise an fehlerhafter Compilierung der Webapp (durch mich)?

schlimmchen, 2024-03-15:
Sehr gute Arbeit.

> Nach erneutem Deaktivieren des Inverters (timestamp: 1710522701) ist dann wieder alles paletti.

```
1710522785.475199  grid:  2443.789795
1710522799.447571 PowerMeterClass: TotalPower: 294.85
1710522799.5756972  grid:  2443.789795
1710522799.6765492  grid:  294.8464355
```

Warum wird laut Konsole frisch der Wert 294W gelesen, aber dann 2443W in die Live-Daten geschrieben? Hm, ist vielleicht nur ein Timing-Problem, scheinbar wird nach jeder Ausgabe auf der Konsole anschließend noch einmal der alte Power-Meter-Wert im Live-Data ausgespuckt, aber dann kurz darauf der neue.

```
1710522655.2060072 PowerMeterClass: TotalPower: 229.77
1710522655.2112586  grid:  229.11203
1710522655.2822735  grid:  229.772934
1710522660.3897078  grid:  229.772934
1710522665.6428945  grid:  1216.917603
1710522669.9821553  grid:  2373.177002
1710522670.0359972 PowerMeterClass: TotalPower: 2373.85
```
Das hier sieht mir ehrlich gesagt auch nicht sooo falsch aus. Der Abstand zwischen den Ausgaben auf der Konsole ist weiterhin 14 Sekunden. Aber dass dazwischen weitere, neue Werte kommen ist absolut merkwürdig.

Warte mal... Ich such mal einen anderen Aufrufer von `getPowerTotal()`.

Äh... Sowohl der PowerLimiter, als auch der HuaweiCan Controller, als auch die PowerMeter Implementierung selbst rufen `getPowerTotal()` ohne Parameter auf, also mit `forceUpdate = true`, teilweise mehrfach, und sorgen so dafür, dass in deinem Fall neue Werte übers SDM Protokoll gelesen werden...

Puh. Also das ist super unschön. Aber es ist kein Fehler. Es sollte natürlich so sein, dass jeder PowerMeter (außer der über MQTT, der ja Event-Driven ist) ein einstellbares Polling-Intervall bekommt, und dass der PowerMeter genau in diesem Zeitraster Daten abholt. Und jeder, der dann nach der aktuellen Leistung am PowerMeter fragt, erhält dann die zwischengespeicherten Werte. Und dann würdest du auch die Daten so vorfinden, wie du es zurecht erwartest, nämlich in den Live-Daten wird bestenfalls ein PowerMeter Wert wiederholt, aber es wird kein neuer abgeholt.

Ich änder den Titel dieses Issue nochmal, um wiederzugeben, was tatsächlich zu tun ist.

Ich danke die nochmal für diese wunderbare Datenaufbereitung!

Ist das Thema wird dich hinreichend erklärt und du kannst OpenDTU-OnBattery uneingeschränkt nutzen, oder führt das zu häufige Abfragen der PowerMeter Werte auch zu einem Problem?

cerise21, 2024-03-15:
Danke sehr für Analyse und Erklärungen!

Nein, bisher habe ich keine Probleme im Zusammenhang mit den Powermeter-Werten gefunden.

Bzgl. Polling-Frequenz: ist die momentan, aus Sicht des DPL, identisch mit den genannten 1/10s?
Ich hatte schon mal nach einer Möglichkeit gesucht, diese zu erhöhen, um häufiger Leistungswerte zu bekommen. Für die SDM-Geräte scheint dies aber in der Webapp nicht vorgesehen zu sein. (?)
Oder könnte der DPL mit häufiger vorliegenden Leistungswerten gar nichts anfangen, im Sinne von besserer Ausregelung des Zielwerts?
Weil der Wechselrichter häufigeren Stell-Befehlen gar nicht nachkommen könnte?

spcqike, 2024-03-15:
> Oder könnte der DPL mit häufiger vorliegenden Leistungswerten gar nichts anfangen, im Sinne von besserer Ausregelung des Zielwerts?
Weil der Wechselrichter häufigeren Stell-Befehlen gar nicht nachkommen könnte?

du willst den WR nicht häufig stellen. Dafür gibt es die hystwrese, dass er eben nicht jeden Watt unterschied nachregeln muss. Du willst auf größere leistungsänderungen nur zeitnah reagieren. Und eben nicht erst nach 5 bis 10 Sekunden, wenn denn der neue Wert von powermerrrr da ist. 

Ich befürworte ein gezieltes Abfragen des powermeters in jedem powerlimiter Durchgang (so es denn möglich ist), um schnell reagieren zu können

cerise21, 2024-03-16:
Ah, ok, auf möglichst schnelles Reagieren auf größere Änderungen kommt es an - danke.

Mir ist aufgefallen, dass seit ca. zwei Tagen (?) (seit der Installation von b101ce4434979cf574efe89b4457bed75163ea04 (?)) der Zähler 'Inverter Total Yield Day' in den Live -Daten nicht mehr nachts zurückgesetzt wird.
Kann das sein?

schlimmchen, 2024-03-16:
Also mit dem genannten Commit hat es sicherlich nichts zu tun. Kann ich mir jedenfalls gerade nicht vorstellen.

Ich nehme an du hast ein System mit Batterie, sonst würde der Inverter nachts in jedem Fall neu starten müssen, weil keine Energie da ist für die Logik im Inverter. Hast du den nächtlichen Inverter-Neustart aktiviert? Meiner wurde um 1 Uhr wie geplant neu gestartet.

cerise21, 2024-03-16:
> Also mit dem genannten Commit hat es sicherlich nichts zu tun.

Stimmt. Natürlich habe ich zeitgleich noch an anderen Stellen geändert: Bei leerem Akku deaktiviere ich jetzt den DPL, um dann den Inverter stoppen zu können (jeweils per API), weil mir die '0.0W' bei 'Inverter Total Power' besser gefallen als die ca. 20W, die dort ohne gestoppten Inverter angezeigt werden (Vorher hatte ich nur das 'upper power limit' des DPL auf '0' gesetzt. Was macht der DPL eigentlich, wenn 'upper power limit' < 'lower power limit' ist? Kommen dann 'upper' oder 'lower' raus?).

Da der Inverter-Restart wohl vom DPL getriggert wird, kann das nur funktionieren, wenn letzterer aktiviert ist (?).

Ist es aber möglicherweise bzgl. des Stromverbrauchs des Inverters irrelevant, ob der gestoppt ist, oder vom DPL als Zielwert '0W' vorgegeben bekommt? Mit dem kosmetischen Unterschied, dass im ersten Fall halt '0W' angezeigt wird, weil ausgeschaltet (aber die DC-Seite ist ja nach wie vor verbunden)?

Anscheinend ist es aktuell so (mit früheren Firmware-Ständen schien mir das nicht so (?)), dass ein 'Disable' des DPL schon ausreicht, um die in den Live-Daten angezeigte 'Inverter Total Power' auf '0W' zu bringen (und ein Stop des Inverters nicht mehr notwendig ist)?

(Sollte ich diese Fragen ggf. in ein anderes Thema umziehen, da dies mit dem Powermeter-Thema, bzw. Live-Data-Aktualisierung nichts mehr zu tun hat?)

cerise21, 2024-03-16:
> Ist das Thema wird dich hinreichend erklärt und du kannst OpenDTU-OnBattery uneingeschränkt nutzen, oder führt das zu häufige Abfragen der PowerMeter Werte auch zu einem Problem?

Hierzu fällt mir noch etwas ein:
Vielleicht könnte man, zumindest für das SDM630, den TotalPower-Wert etwas näher an den tatsächlichen aktuellen Wert bringen (auch ohne Wartezeiten auf das Einsammeln (hoffentlich) zusammengehöriger Phasenwerte), wenn getTotalPower() statt der Addition der drei Phasenwerte direkt den (beim SDM636 vergfügbaren) Wert `SDM_TOTAL_SYSTEM_POWER` benutzte (vgl.` SDM.h`)?
Soweit ich das verstehe, müsste dieser dann zusätzlich in `readPowerMeter()` per `sdm.readVal()` ausgelesen werden(?).

Hintergrund ist:
Das SDM-Powermeter macht keine richtige (nach unseren europ. Vorstellungen) Saldierung von Import und Export, d. h. die zur Verfügung gestellten Import- und Export-Werte (`SDM_IMPORT_ACTIVE_ENERGY` und `SDM_EXPORT_ACTIVE_ENERGY`) addieren nur stumpf jeweils alle positiven und alle negativen Werte über die Phasen.
Deshalb integriere ich (in openHAB) numerisch die per MQTT angelieferten Leistungsdaten, um, je nach Vorzeichen in der aktuellen Integrationsperiode, den Energiehappen dem Import- oder Exportzähler zuzzuschlagen.

schlimmchen, 2024-03-16:
> Bei leerem Akku deaktiviere ich jetzt den DPL, um dann den Inverter stoppen zu können

Dann wird der Inverter auch nicht neu gestartet.

> weil mir die '0.0W' bei 'Inverter Total Power' besser gefallen als die ca. 20W, die dort ohne gestoppten Inverter angezeigt werden

Check ich nicht. Der DPL macht doch den Inverter aus, wenn die Batterie leer ist, das ist doch eine seine Hauptaufgaben?! Und wenn der Inverter im Standby ist, ist auch Inverter Total Power 0W (bei mir).

> statt der Addition der drei Phasenwerte direkt den (beim SDM636 vergfügbaren) Wert SDM_TOTAL_SYSTEM_POWER benutzte

Vorstellbar, aber ich frage mich, ob wir das einfach umstellen können. Kann den jeder SDM PowerMeter liefern, diesen Wert? Falls ja, ist es natürlich Unfug, dass der SDM-PowerMeter die Phasen einzeln abholt und addiert.

Das ließe sich einigermaßen leicht umsetzen, **nachdem** die ganze PowerMeter-Implementierung einem Refactor unterzogen wurde. Das ist zwar nötig, aber gerade nicht hoch priorisiert, jedenfalls nicht bei mir.

spcqike, 2024-03-16:
> Falls ja, ist es natürlich Unfug, dass der SDM-PowerMeter die Phasen einzeln abholt und addiert.

ist es nicht in manchen Ländern Pflicht, eine echte Nulleinspeisung zu machen? Und haben wir nicht auch daher die Möglichkeit, einzelne Phasen abzufragen? 

Sollte man den SDM Code überarbeiten, vielleicht kann man dann direkt die Option vorhalten, die Summe abzufragen oder aber auch die Phasen einzeln. 

Bisher kann der DPL zwar nicht damit umgehen (außer, wie manche User es machen, man nutzt 3 opendtu mit je einem hoymiles auf einer Phase), aber künftig wäre es ja vorstellbar. 3 hoymiles und jeder bedient eine Phase.

schlimmchen, 2024-03-16:
In dem von dir skizzierten Szenario nutzt man "SDM1 Phase" und da gibt es das Problem ja nicht (dass die Summe der Phasen nicht wirklich den aktuellen Bezug darstellt, weil man sie nacheinander abfragt).

closed 2024-03-17 by schlimmchen

cerise21, 2024-03-17:
> Check ich nicht. Der DPL macht doch den Inverter aus, wenn die Batterie leer ist, das ist doch eine seine Hauptaufgaben?! Und wenn der Inverter im Standby ist, ist auch Inverter Total Power 0W (bei mir).

Ja, genau, bei mir auch.

Mein Anwendungsfall ist halt folgender:
Die tagsüber im Akku gesammelte Energie möchte ich, um Einspeisung ins öffentliche Netz beim Ausschalten großer Lasten zu verhindern, gleichmäßig verteilt über die Nachtstunden verbrauchen.
Dafür berechne ich abends einen kleineren Wert für das 'upper power limit' des DPL als die tagsüber gesetzten 800W.
Wenn die Berechnung einen kleineren Wert als z. B. 40W für das 'upper power limit' ergibt, verzichte ich, wegen niedrigem Wirkungsgrad des Inverters (?), auf Einspeisung aus dem Akku.
Hierfür hatte ich zunächst das 'upper power limit' auf 0 gesetzt, mich aber dann an den 'Inverter Total Power'-Werten gestört, die trotzdem zwischen 2W und 30W betrugen.
Deshalb deaktiviere ich in diesem Fall jetzt zusätzlich den Inverter; 'Inverter Total Power' ist dann 0W (lediglich die DC-Leistungsaufnahme bleibt bei ca. 2W, da nicht physikalisch getrennt).

> Anscheinend ist es aktuell so (mit früheren Firmware-Ständen schien mir das nicht so (?)), dass ein 'Disable' des DPL schon ausreicht, um die in den Live-Daten angezeigte 'Inverter Total Power' auf '0W' zu bringen (und ein Stop des Inverters nicht mehr notwendig ist)?

Hier habe ich mich getäuscht, ein 'Disable' des DPL hat wohl schon immer gereicht, um 'Inverter Total Power' auf 0W zu bringen.

> Vorstellbar, aber ich frage mich, ob wir das einfach umstellen können. Kann den jeder SDM PowerMeter liefern, diesen Wert? Falls ja, ist es natürlich Unfug, dass der SDM-PowerMeter die Phasen einzeln abholt und addiert.

In `SDM.h` ist eine Tabelle, die die bei den verschiedenen Modellen verfügbaren Werte zeigt.
`SDM_TOTAL_SYSTEM_POWER` ist bei einigen Modellen verfügbar, nicht bei allen.
Falls für ein Modell nicht verfügbar, könnte dann ja ein fallback auf die aktuelle Lösung mit Addition der Phasenwerte erfolgen.

> Das ist zwar nötig, aber gerade nicht hoch priorisiert, jedenfalls nicht bei mir.

Gar kein Problem. Ich wollte diese Möglichkeit für das SDM630 nur mal festhalten…

schlimmchen, 2024-03-17:
> In SDM.h ist eine Tabelle, die die bei den verschiedenen Modellen verfügbaren Werte zeigt.
SDM_TOTAL_SYSTEM_POWER ist bei einigen Modellen verfügbar, nicht bei allen.
Falls für ein Modell nicht verfügbar, könnte dann ja ein fallback auf die aktuelle Lösung mit Addition der Phasenwerte erfolgen.

Mach dafür mal einen Feature-Request auf.

Ach, ich sehe ich habe dieses Ticket geschlossen als ich #749 ge-merge't habe. Meh. Das kommt davon, wenn man drei Themen in einem Issue abarbeitet. Wir hätten neue aufmachen sollen, als wir jeweils den Fokus gewechselt haben.

Ich lass dieses hier geschlossen und wir machen ein neues für das verbliebene Thema.

commit 2ba57e25 (2024-03-11): fix: prevent getTotalPower() reading intermediate results

the SDM power meter (among others) writes the power consumption of three
phases in multiple steps. this change helps to prevent getTotalPower()
reading intermediate values, i.e., reading a new value for phase 1 but
old values for phase 2 and 3 since phase 2 is currently read.

cache the values, and write them all at once, protected by a mutex,
later.

closes #732.
commit b101ce44 (2024-03-13): fix: prevent getTotalPower() reading intermediate results

the SDM power meter (among others) writes the power consumption of three
phases in multiple steps. this change helps to prevent getTotalPower()
reading intermediate values, e.g., reading a new value for phase 1 but
old values for phase 2 and 3 since phase 2 is currently read.

cache the values, and write them all at once, protected by a mutex,
later.

closes #732.
PR #749 (2024-03-17): fix: prevent getTotalPower() reading intermediate results
the SDM power meter (among others) writes the power consumption of three phases in multiple steps. this change helps to prevent getTotalPower() reading intermediate values, i.e., reading a new value for phase 1 but old values for phase 2 and 3 since phase 2 is currently read.

cache the values, and write them all at once, protected by a mutex, later.

closes #732.
commit 90032674 (2024-03-17): fix: prevent getTotalPower() reading intermediate results

the SDM power meter (among others) writes the power consumption of three
phases in multiple steps. this change helps to prevent getTotalPower()
reading intermediate values, e.g., reading a new value for phase 1 but
old values for phase 2 and 3 since phase 2 is currently read.

cache the values, and write them all at once, protected by a mutex,
later.

closes #732.