hoylabs/OpenDTU-OnBattery #735: [Request] Support seperate import/export values from power meter

jladefoged, 2024-03-10:
### Is your feature request related to a problem? Please describe.

Some power meters (like AMS reader) have separate import/export values for power.
It does not show a negative grid import value when electricity is exported.
This separate export value is not supported by OpenDTU-OnBattery.

### Describe the solution you'd like

Add a value for export in power meter settings.
Calculate the total from import and export values to show current grid power.
Export value from AMS reader is in my case a positive number, but maybe some meters show it as a negative number?

### Describe alternatives you've considered

I suggested adding a combined import/export value feature in AMS reader.
Both options seems like a good idea for both projects so I'm also suggesting adding support for an export value in OpenDTU-OnBattery.

### Additional context

Link to AMS reader:
https://github.com/UtilitechAS/amsreader-firmware

spcqike, 2024-03-10:
As Alternative solution it would be possible to add basic calculation to the current power meters. 

It’s possible to query up to three different devices (as we can make one query per phase). This can have the same or different url, and they have different json path. 

If it would be possible to query the json and to simple calculation like multiply by -1, one could query AMS grid Import for phase 1, grid export for phase 2. grid export must than be negated, so multiplied by -1. 

I think, even to its more or less just a workaround, it would do the trick. And, who knows, maybe simple calculation may be needed for other powermeter, too, as they could report values as mW or kW or something.

schlimmchen, 2024-03-24:
@jladefoged Are you using MQTT or HTTP+JSON to retrieve the power meter reading from your "AMS reader"?

@spcqike Using the existing settings to query the export is a good idea. The only thing that is needed is a switch to tell the respective PowerMeter that a value denotes exported power even though it is positive, so it will be multiplied by -1. And we need to loosen up the labels in the HTTP+JSON power meter, which says "Phase 1" through "Phase 3". Let's make them "Value 1" through "Value 3". And it should be reflected in the code as well. So.... this is some work.

jladefoged, 2024-03-24:
> @jladefoged Are you using MQTT or HTTP+JSON to retrieve the power meter reading from your "AMS reader"?
> 

I use HTTP+JSON to read values from AMS reader.
Available JSON values can be seen under the "http://<ams_hostname>/data.json" section here:
https://github.com/UtilitechAS/amsreader-firmware/wiki/Message-formats

schlimmchen, 2024-03-24:
Understood.

Not that it solves your problem, but consider switching reading the power meter values from MQTT, which is much less resource-intensive.

The fix for your issue will be made for both MQTT and HTTP+JSON power meter implementations.

jladefoged, 2024-03-24:
> Understood.
> 
> Not that it solves your problem, but consider switching reading the power meter values from MQTT, which is much less resource-intensive.
> 
> The fix for your issue will be made for both MQTT and HTTP+JSON power meter implementations.

Sounds great. Looking forward to getting the new feature.

Thanks for the tip regarding MQTT instead of HTTP+JSON.
AMS reader already publishes values to my MQTT server that I use in Home Assistant.
I've tried setting it up before in OpenDTU, but newer got it working. I will try again.

closed 2024-04-18 by schlimmchen