hoylabs/OpenDTU-OnBattery #748: Emergency Inverter reset

MalteSchm, 2024-03-11:
### Is your feature request related to a problem? Please describe.

I had two cases now were the inverter operated at full power with communication between openDTU and the inverter being broken. Soft and hard restarts of the ESP are unsuccessful. I got the system back by issuing an inverter reset command.

### Describe the solution you'd like

When the inverter communication is offline for a long time or when the battery SoC drops under a certain limit perform an Inverter reset. If this does not help perform an ESP restart

In addition it would be good to establish MQTT topics for alerts (potentially also info messages) that can carry basic log info. I would like to display the last alerts messages in Node Red to get some history

schlimmchen, 2024-03-19:
> or when the battery SoC drops under a certain limit

I don't support this, as the battery being discharged to a particular level does not justify restarting the inverter. There can be plenty other reasons for why the battery SoC dropped.

> If this does not help perform an ESP restart

Meh. Okay.

I propose the following. The DPL (conceptually) has a small loop inside the `loop()` which makes sure that the inverter is set to the requested state (power production enabled/disabled and the limit). This can time out. Let's introduce a timeout counter. If it reaches 5, the inverter receives a restart command. If the counter still increases, it receives another restart command. If the counter increases further, the ESP is reset. The counter is reset when the inverter reaches the requested state.

> In addition it would be good to establish MQTT topics for alerts 

What do you need? Please specify. How about the value of the aforementioned timeout counter? It would indicate a problem as soon as it is non-zero, and the value indicates the level of escalation :wink:

MalteSchm, 2024-03-19:
I meant to trigger the inverter restart when the SoC drops below a GUI configured limit. 
But overall I'm fine with any solution that works. Your approach has the benefit of catching this issue early. 

The MQTT idea is meant to complement logging / monitoring. An inverter restart / ESP restart goes unnoticed. When this is rare I don't mind this behavior. When this happens often I would want to take some action. 
My idea is to publish something to an `/alert` topic that contains the current time and a message. Many run some external system (Node Red in my case) that would be able to monitor and display this info on a dashboard.

closed 2024-04-24 by MalteSchm