hoylabs/OpenDTU-OnBattery #751: Implementation of disable PIN for DPL and inverter

hoschiking, 2024-03-13:
### Is your feature request related to a problem? Please describe.

For many scenarios, it would be practical if you could deactivate the DPL externally. At the same time, the inverter should also be deactivated.

### Describe the solution you'd like

I would suggest a disable pin for this. 
Possible applications:
External watchdog
Activation of the feed-in during the day while the battery is being charged.

### Describe alternatives you've considered

disable the DPL with the mqtt command.
In this case the DPL can be controlled by software only.

spcqike, 2024-03-14:
Can you describe such a scenario where you need an external watchdog?

hoschiking, 2024-03-14:
for example:
for monitoring the battery voltage.
if the battery voltage falls below a threshold value, the inverter and the DPL will be switched off.
for this can be used f.e. the load output of any victron MPPT, this ouput is programmable.

spcqike, 2024-03-14:
Why couldn’t one just use the inverter DC input voltage for this? Do you see an advantage in an external device which than controls the opendtu?

hoschiking, 2024-03-14:
One of the most important requirements of a monitoring system (watchdog) is that it runs on another hardware.
If the DPL feeds in uncontrollably, as has been reported several times by users, the battery will not be discharged uncontrollably. 
do you need more advantages?

schlimmchen, 2024-03-14:
> do you need more advantages?

Yes, because if the DPL software goes rogue, there is no use in adding more software on the same firmware/microcontroller that monitors a particular GPIO pin and then tries to tell the rogue software to behave.

I am sorry, @hoschiking, that I made you open this request. There was a misunderstanding: I though you were also interested in a feature that allows to disable charging and/or discharging of the battery by instructing the JK BMS respecively.

You are asking for something very different. And as is @spcqike, I am not convinced that this is a valuable feature. I would understand if you were interested in a kill-switch that runs on a different hardware, but that kill-switch then also needs to control something outside the OpenDTU-OnBattery ESP32 such that its countermeasure is effective regardless of the OpenDTU-OnBattery software or hardware. That, however, is then out of scope for this project, obviously.

spcqike, 2024-03-14:
I had similar thoughts before. But also came to the conclusion, that it’s not really useful or clever to read inputs to control something, which could be controlled directly by openDTU. 

One thing one could do is a forced reset of your openDTU. This only needs a short external signal to your ESPs reset input. So an opendtu which doesn’t respond anymore would reset and hopefully behave normal again. 

But this is all external. And the decision whether or not to reset must be made very carefully.

hoschiking, 2024-03-14:
@spcqike 
I suggested nothing other than what was described in your last post.
And as long as this reset is present, the OpenDTU does not restart, but remains off until the reset signal has been reset.

@schlimmchen 
I didn't suggest what you describe either and it doesn't really make much sense. It is explicitly not about controlling a piece of software.

schlimmchen, 2024-03-14:
Dann sollten wir vielleicht auf Deutsch weitermachen, wenn wir aneinander vorbei reden?

Also worum geht es dir? Was @spcqike beschreibt ist "lediglich" das steuern des Reset-Pins (oder BOOT heißt der glaube ich oft) des ESP32. Ich bin mir nicht sicher, wie einfach man da rankommt, kommt ja auch auf das Board an, aber da muss dein externer Watchdog dann ran und diesen Pin steuern. Wenn man den festhält, wird auch die Software halten. Wenn man den Pin wieder loslässt, dann wird der ESP32 neu starten. Festhalten und loslassen meint hier eine Spannung anlegen bzw. wegnehmen bzw. an GND kurzschließen oder was auch immer nötig ist, ich weiß es auswendig nicht :wink:

Das ist dann eine individuelle Sache. OpenDTU-OnBattery muss hier nichts tun.

hoschiking, 2024-03-14:
Ja, soweit so gut, das was du beschreibst ist ja bekannt, aber mit diesem Reset wird der Inverter das Einspeisen nicht unterbrechen, da dieser in dem Fall in seinem letzten Zustand verbleibt und unkontrolliert weiter einspeist. Leider ist dieser nur per Funk steuerbar, daher die Idee mit dem Abschalten des Inverters. Wenn dies aber so nicht realisierbar ist, dann wird es wohl nicht gehen. Ich habe eine gewünschte Funktion beschrieben, nicht wie es realisiert werden kann oder soll.

schlimmchen, 2024-03-14:
Korrekt. Da muss dann die externe Hardware versuchen, per Funk mit dem Inverter zu sprechen und ihn abstellen. Oder du vertraust darauf, dass der ESP32 mit OpenDTU-OnBattery sich nach dem Reset schon wieder einkriegen wird.

Oder du nimmst an, dass OpenDTU-OnBattery sehr stabil läuft, was es meiner Meinung nach tut, und dass im Notfall den BMS zur Stelle ist :wink:

skippermeister, 2024-03-14:
Es wird ein DC-seitige Abschaltung der Verbindung zwischen der Batterie und dem Inverter benötigt.

In meiner Schaltung sind Optokoppler für die Ansteuerung von Hochstrom MosFETs zur Aktivierung der DC Verbindung zwischen Batterie und Inverter Low-active (das ist genau der Grund meines Designs gewesen).
Wenn die EP32 CPU resettet wird, dann werden alle GPIOs hochohmig und Eingänge. 
Damit wird die Verbindung des Inverters DC-seitig sofort unterbrochen und der Inverter speist nicht mehr ein.

@schlimmchen : Schau Dir mal meine Software Erweiterung manageBatteryDCpowerSwitch() im DPL Modul zur DC-seitigen Ein-/Aus Steuerung des Inverters dazu an. 
Mein Setup ist zwar etwas anders, aber die Funktion welche ich im DPL in meinem Fork implementiert habe realisiert den automatischen DC-seitigen Softstart in der Nacht und die automatische Abschaltung der DC-Verbindung am Tage.
Und die Hardware Lösung stellt im Störungsfall oder Reboot oder Software Update und sogar beim Reset ausgelöst durch die Watch Dog der ESP32 CPU eine sofortige Unterbrechnung der DC-Verbindung sicher.

spcqike, 2024-03-14:
> zwischen Batterie und Inverter Low-active

wie stellst du sicher, dass sich nicht ein FET verabschiedet und die restlichen DC- Leitungen überlastet werden? Oder gibt es nur einen für alle DC Eingänge? 

Wäre eine AC seitige abschaltung des inverters nicht einfacher und unkritischer? (Abgesehen vom weiterhin vorhanden Eigenverbrauch des wechselrichters)