hoylabs/OpenDTU-OnBattery #753: [Request] target power from grid during charging of battery

gitisgreat2023, 2024-03-14:
### Is your feature request related to a problem? Please describe.

Similarly like the feature "Target power consumption from grid" it would be ideal to have it for charging as well. Currently during charging the algorithm tries to achieve 0 (?) grid load and accordingly on less sunny days energy is fed into the grid, due to the fluctuations in power (of pv or usage). Ideally one can choose the target oneself.

### Describe the solution you'd like

The feature "Target power consumption" as it is implemented in the Dynamic Power limiter.

### Describe alternatives you've considered

None.

### Additional context

Adding a variable target for charging helps to reduce feeding energy into the grid.

schlimmchen, 2024-03-15:
Is this about the Huawei charger?

gitisgreat2023, 2024-03-15:
Yes.

oldman4U, 2024-03-15:
So you keep the DC/AC inverter feeding power from the battery into the grid while you charge the battery with power from the grid using an AC/DC converter?

gitisgreat2023, 2024-03-16:
@oldman4U 
As long as the target value to grid while charging is less then the absolute lowest base load, say 90W, you don't charge the battery from the grid.

I have only an AC charger (Huawei), no direct DC loading of the battery. Currently, as the algorithm aims for 0 W grid load, thus during fluctuations the grid load goes negative.

Now the base load is always present, say 100 W. If I set as charging target say 20 W, it means 20 W is drawn from the grid for the baseload, the rest is covered by the photovoltaic. The surplus of the photovoltaic after covering the load over 20W is completely fed in to battery.

The logic is exactly the same as for the "Target power consumption from grid" feature. There its for discharging, but you have the same situation while charging. In both cases I want to avoid to feed into the grid.

During less sunny periods like these I would put the value to say 20W or so, clearly less than the base load but away from 0W. During summer, where I have too much energy to use & store all day, I would set it to say -20W.

oldman4U, 2024-03-16:
Hmmm. It is easily possible that I am wrong. But for me it makes no sense to have the inverter on while charging the battery from grid. Therefore I stop the inverter while charging the battery from the grid. This allows the battery to charge faster and guarantees me that I never export any power to the grid during this. And there is no loss because of the conversions.

gitisgreat2023, 2024-03-17:
I'm not sure we're talking about the same. I have an AC charger and HM with battery, no Victron which loads directly dc into the battery.

We're talking about the situation that we have surplus power that we want to load into the battery. The inverter is off, the charger charges. The goal is to avoid exporting power to the grid. With the target of 0 for the charger that happens about 50% of the time. A small offset of say 20 W also improves quite a lot.

This is both relevant for time periods where there is not enough photovoltaic energy in total over the day: then it makes sense to set the AC charger target to say 20 or 30 W.

When there's too much over the day in total, then it makes sense to set it to say -20 or -30 W, because exactly the same argument goes the other way: 50% of the time it needs grid power if it targets for 0 W.

oldman4U, 2024-03-17:
OK

schlimmchen, 2024-04-15:
@MalteSchm Is this as trivial as it seems?

@gitisgreat2023 Reusing as setting of the power limiter will be very confusing. Adding a new setting in the context of the AC charger is fine.

MalteSchm, 2024-04-15:
> @MalteSchm Is this as trivial as it seems?

Yes I think so

MalteSchm, 2024-04-24:
Addressed in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/918

commit 3e208219 (2023-04-05): Fix #753: Only apply offset if data is in the buffer to prevent negative numbers
commit d099bc4f (2023-04-06): Fix #753: Only apply offset if data is in the buffer to prevent negative numbers
closed 2024-04-24 by schlimmchen