hoylabs/OpenDTU-OnBattery #754: [Request] DC/Battery stop SoC

gitisgreat2023, 2024-03-14:
### Is your feature request related to a problem? Please describe.

Currently stopping of the charging is defined by voltage. As both the start and stop threshold conditions are defined in SoC %s, adding a target to stop charging would make sense.

### Describe the solution you'd like

Adding Battery SoC - Stop charging.

To avoid confusion: best rename 
a) Battery SoC Stop threshold  to Battery SoC - Stop discharging 
b) Battery SoC - Start charging to Battery SoC - Allow discharging

Manos1966, 2024-03-14:
Quick question for my understanding:
"Adding Battery SoC - Stop charging."

What are you using for charging?
Because as I understand, the VICTRON MPPTs are not receiving data from OpenDTU (in order to stop charging)

schlimmchen, 2024-03-15:
There is no "stop charging". There is only "start discharging", or in other words "start using the battery for energy production".

I don't understand what you are asking for.

What should happen if the newly proposed "Battery SoC - Stop charging" threshold is reached?

schlimmchen, 2024-03-15:
Is this about the Huawei charger?

gitisgreat2023, 2024-03-15:
Yes, after reaching the "Battery SoC - Stop charging" threshold it should stop charging...

Currently there are two SoC values: stop discharge, start again discharging. For example 15% (stop) and after reaching 20% allow discharging again.

The third value would be the upper level to stop charging, like 90% or 95%. 

Yes my setup is a HM, Huawei, Pylontech setup.

schlimmchen, 2024-04-11:
I think this is completed with #848, which will be included in the next release.

[image]

Please re-open this issue if I got it wrong.

closed 2024-04-11 by schlimmchen