hoylabs/OpenDTU-OnBattery #760: Powermeter, HTTP+JSON, Umstellung auf asynchrone Abfrage der Werte

SW-Niko, 2024-03-16:
### Is your feature request related to a problem? Please describe.

Die synchrone Abfrage der Powermeter Werte über HTTP+JSON (Z.B. Shelly 3EM) überschreitet mehrmals am Tag die konfigurierten zeitlichen Grenzwerte. (Default 1sec)
Dadurch werden weitere DTU Komponenten negativ beeinflusst.
- Receive Puffer Überlauf der Seriellen Schnittstellen (MPPT VE.Direct) Siehe auch: #752 
- Vermutlich auch das kurzzeitige Abschalten des Inverters. Siehe auch #669
- Weitere ???

### Describe the solution you'd like

Zeitliche Probleme (Timeout) beim Abfragen der Powermeterwerte (WLAN) sollten andere DTU Komponenten, die nicht direkt von den Powermeterwerten abhängen, nicht negativ beeinflussen.

### Describe alternatives you've considered

Umstellen der Abfrage der Powermeterwerte von Synchron auf Asynchron.

### Additional context

Auszug aus meinem Statistik logging:

12:52:35.344 > TST: Data valid, Duration: 0 days, 5 hour, 5 min, 30 sec
12:52:35.344 > --------------------------------------------------------------
12:52:35.344 > TST:  HttpPo...updateValues():    OK-Time: Max:952ms, Min=72ms, Avg=109ms, Ok=11T, Value=0.00
12:52:35.344 > TST:                         : NotOk-Time: Max=1357ms, Min=1017ms, Avg=1177ms, Fault=5/0.04%, Value=0.00
12:52:35.344 > TST:                         :  Back-Time: Max=10021ms, Min=3ms, Avg=1887ms

Es wurde 5 Stunden, die Funktion HttpPowerMeterClass::updateValues(): geloggt.
In dieser Zeit wurde die Funktion 11000mal aufgerufen. Rückgabe wert = true:
Im Durchschnitt dauert es 109ms bis die Werte da sind. Die maximal Zeit war 952ms
Es wurden 5 Fehler (Timeout) registriert. Rückgabe wert = false:
Interessant ist auch noch die minimale Back-Time von 3ms. Das ist die Zeit vom Verlassen der Funktion bis zum erneuten Aufruf.

Ich interpretiere das so:
Wenn es zu einem Timeout (ca. 1Sek) kommt dann versuche man es gleich nochmal. Dann sind wir bei 2 Sekunden und das deckt sich mit meinen Messungen am receive Puffer. #

schlimmchen, 2024-03-22:
Ja, das macht Sinn, und das habe ich schon kommentiert in deinem Issue mit dem SDM power meter, dass der SDM power meter Treiber furchtbar ist und synchron Daten anfordert und dann auf Antworten hört. Hier wird was anderes, aber ganz ähnliches bemängelt, nämlich dass die Anfrage und Verarbeitung von HTTP power metern ebenfalls synchron läuft und andere Aufgaben in der Firmware aufhält.

ottelo9, 2024-03-26:
Ich habe auch gerade gesehen, dass die Werte von meinem Stromzähler (Tasmota) nicht im eingestelltem Intervall abgefragt werden, sondern fast schon jede Sekunde. Wenn ich DPL ausschalte, dann kommt die Abfrage nur im eingestelltem PowerMeter Intervall (bei mir auf 27s zum Test eingestellt).

Aktiviere ich DPL kommt die Abfrage zuerst alle 27s, dann plötzlich alle 1-3s.
[image]

Hier mal die Logs:
opendtu
https://pastebin.com/RWpvsBQw

tasmota (StatusSNS)
https://pastebin.com/19yahXcq

schlimmchen, 2024-03-26:
Ja, das gehört in #579 und hat mit diesem Issue nichts zu tun.

schlimmchen, 2024-07-01:
Umgesetzt in #1077.

closed 2024-07-10 by schlimmchen