hoylabs/OpenDTU-OnBattery #763: Ungewollt häufiges Abfragen des PowerMeter

schlimmchen, 2024-03-17:
### What happened?

Im Verlauf von #732 haben wir festgestellt, dass der PowerMeter ungewollt häufig abgefragt wird, und zwar mit dem Zwang, ein Update zu holen.

### To Reproduce Bug

Gut beschrieben in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/732#issuecomment-2000179569. Mit websocat lässt sich bestätigen, dass aus scheinbar unklaren Gründen der PowerMeter Wert im Live View häufiger wechselt/aktualisiert, als die Konsole zeigt. Das hat damit zu tun, dass mehrere Komponenten den PowerMeter befragen und dabei ein Update erzwingen. Beispielsweise der DPL, der nur für eine Debug-Ausgabe riskiert, eine langwierige serielle Transaktion oder drei HTTP Requests zu machen.

Siehe auch https://github.com/helgeerbe/OpenDTU-OnBattery/issues/732#issuecomment-2000262707

### Expected Behavior

Der PowerMeter sollte sich vor allem selbst verwalten und entscheiden, wie oft er neue Daten holt. Benutzer sollten lediglich den zuletzt abgeholten Wert erhalten (können).

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

foo

schlimmchen, 2024-03-18:
Im Grunde ein Duplikat zu #579.

closed 2024-03-18 by schlimmchen

SW-Niko, 2024-03-18:
Das erklärt auch meine Zeitmessungen. Siehe #760 
Sehr gut ....

SW-Niko, 2024-03-18:
Ja genau. Ich bin da zufällig darüber gestolpert als ich mich auf die suche nach dem Störenfried für den Puffer Überlauf gemacht habe. Ich kann dir aber nicht sagen wo im Code das passiert. Aber anscheinend nachdem eine Powermeterabfrage wegen Zeitüberschreitung angebrochen wurde.