hoylabs/OpenDTU-OnBattery #765: [Request] SDM-PowerMeter: TotalPower-Wert direkt auslesen, statt drei Phasenwerte addieren

cerise21, 2024-03-17:
### Is your feature request related to a problem? Please describe.

In https://github.com/helgeerbe/OpenDTU-OnBattery/issues/732 ergab sich, dass getTotalPower() sehr häufig, von verschiedenen Programmteilen, aufgerufen wird und die zum TotalPower-Wert addierten Werte der drei Phasen nicht unbedingt aus dem selben Abfragezyklus stammen, was zu falschen (Zwischen-)Werten führte.

### Describe the solution you'd like

Einige SDM-Powermeter, wie z. B. das SDM630, bieten zusätzlich zu den Leistungswerten der einzelnen Phasen auch schon einen Summenwert (SDM_TOTAL_SYSTEM_POWER) an. (Vgl. Tabelle in SDM.h.)
Dieser Wert müsste dann zusätzlich in readPowerMeter() per sdm.readVal() ausgelesen werden(?)
Falls für ein Modell nicht verfügbar, könnte dann ein fallback auf die aktuelle Lösung mit Addition der Phasenwerte erfolgen.

### Additional context

Ggf. mit der Bearbeitung von https://github.com/helgeerbe/OpenDTU-OnBattery/issues/763 kompatibel?

AndreasBoehm, 2025-02-23:
Hallo @cerise21 ich habe deinen wunsch in https://github.com/hoylabs/OpenDTU-OnBattery/pull/1651 umgesetzt. Eine Test-Version ist dort in den Beschreibung verlinkt. Wäre super wenn du testen könntest ob weiterhin alles funktioniert, Danke.

closed 2025-03-19 by schlimmchen

cerise21, 2025-03-19:
> Hallo [@cerise21](https://github.com/cerise21) ich habe deinen wunsch in [#1651](https://github.com/hoylabs/OpenDTU-OnBattery/pull/1651) umgesetzt. Eine Test-Version ist dort in den Beschreibung verlinkt. Wäre super wenn du testen könntest ob weiterhin alles funktioniert, Danke.

Oh, vielen Dank für's Einbauen!
Hab' das leider gerade erst mitbekommen.
Ich gucke, dass ich das die nächsten Tage testen kann.

schlimmchen, 2025-03-20:
@cerise21 Bitte nimm Firmware aus [diesem Action Run vom development Branch](https://github.com/hoylabs/OpenDTU-OnBattery/actions/runs/13955639609).

commit ce1e8554 (2025-02-18): Feature: 3-phase SDM: read SDM_TOTAL_POWER

Closes #765
commit 74a0fda3 (2025-02-18): Feature: 3-phase SDM: read SDM_TOTAL_POWER

Closes #765
commit 08a58862 (2025-03-19): Feature: 3-phase SDM: read SDM_TOTAL_POWER

Closes #765
commit ea0d989a (2025-03-19): Feature: 3-phase SDM: read SDM_TOTAL_POWER

Closes #765