hoylabs/OpenDTU-OnBattery #766: Optionaler Fallback auf konfigurierbare WR-Leistung (Grundlast) statt Abschalten im Falle alter oder fehlender PowerMeter Daten

der-peterh, 2024-03-17:
### Is your feature request related to a problem? Please describe.

Einer der häufigeren Gründe, warum der Wechselrichter abschaltet, ist die Meldung
```
19:00:40.728 > [  10225.822] DPL: power meter readings are outdated
19:00:40.730 > [DPL::updateInverter] Stopping inverter...
```
Nun kann das WLAN oder - in meinem Falle - die MQTT Infrastruktur - durchaus mal Schwierigkeiten verursachen. 
Für diese Fälle wäre es sinnvoll, wenn zumindest eine konfigurierbare Grundlast durch den Wechselrichter bereitgestellt wird. 

### Describe the solution you'd like

Bei der Konfiguration des Dynamic Power Limiter könnte es eine Möglichkeit geben, einen Wert für die Grundlast einzugeben. Dieser wird dann genutzt, wenn kein Power Meter vorhanden bzw. konfiguriert ist oder die Daten nicht problemlos abgerufen werden können bzw. 'outdated' sind. 
Dabei wird die Batteriespannung weiterhin überwacht, dass sie innerhalb der vorgegebenen Grenzen bleibt.

### Describe alternatives you've considered

Alternativ könnte bei der Konfiguration des Power Meter auch die Grundlast als Fallback oder default Wert hinterlegt werden. Falls kein Power Meter vorhanden ist, wäre diese Art der Konfiguration aber irgendwie irreführend.

### Additional context

Dies erlaubt eine sinnvolle Nutzung des Projekts als Grundlasteinspeisung in absoluter Minimalkonfiguration mit lediglich einem HM an einer Batterie ohne irgendwelche weitere zu überwachende Geräte.
Zusammen mit OpenDTU-OnBattery, einem HM-600 für 80 Euro + Akkus für 200-600 Euro + 50 Euro MPPT Controller günstiger Machart, gespeist von wenigen 90 Euro Solarmodulen sowie etwas Kabelei würde man so bereits eine valide Startkonfiguration zusammenstellen können, die die Grundlast solide abdeckt.

spcqike, 2024-03-17:
Bin ich auch für :) wurde gestern erst in #755 angesprochen.

schlimmchen, 2024-03-29:
Hast du ein System mit Batterie?

der-peterh, 2024-03-29:
Hier kurz meine Konfiguration:
8 Zellen LiFePo 24Volt. 
Einfachen China MPTT Laderegler. 
DC Anbindung HM-600 an Batterie (über einen zweistufigen Schalter 1: Widerstand zum Laden der DC seitigen Kondensatoren, 2: direkter Anschluss). 
Der HM-600 ist als Balkonkraftwerk angemeldet.
Zusätzlich liefert ein Shelly 3 über MQTT die aktuellen Stromwerte (Verbrauch/ Einspeisung).
Die Steuerung mit OpenDTU on Battery soll also zunächst sicherstellen, dass Batterie weder in die Tiefentladung geht (klappt), noch überladen wird (klappt ohne Victron MPTT leider aktuell auch noch nicht - ist aber auch noch durch BMS abgesichert).
In diesem Leistungsbereich soll nun so wenig Energie wie sinnvoll regelbar ins Netz eingespeist werden.
Sofern irgendwelche Komponenten wie WLAN, Shelly oder MQTT mal nicht verfügbar sind, soll halt stattdessen eine konfigurierbare Leistung eingespeist werden, die zum Beispiel meiner Grundlast entspricht. 
Schöne Ostertage 👋

schlimmchen, 2024-04-07:
> Dabei wird die Batteriespannung weiterhin überwacht, dass sie innerhalb der vorgegebenen Grenzen bleibt.

Genau das ist der lästige Teil.

~~Also wenn der PowerMeter gar nicht konfiguriert ist, dann soll der DPL weiter seinen Dienst verweigern (und den Inverter abschalten), meiner Meinung nach. Gibt es dazu Gegenmeinungen?~~ Siehe https://github.com/helgeerbe/OpenDTU-OnBattery/issues/809#issuecomment-2041587094.

Ohje, noch lästiger wird die Berücksichtigung von Solar-Passthrough. Was soll den passieren, wenn vom Laderegler 100W kommen, aber das untere Leistungslimit auf 300W steht (wäre ein stimmiger Wert für meinen Haushalt). Bei diesem Beispiel würde bisher der WR ausgeschaltet bleiben. Das ist schade. 100W würde ich beispielsweise durchaus direkt nutzen wollen, statt sie in die Batterie zu schieben. Das tue ich zur Zeit, weil mein unteres Leistungslimit auf 30W steht.

Vielleicht braucht es eine neue Interpretation des Lower Power Limit (wird im Grunde ja auch so gefordert).

Bisher wird die Einstellung "Unteres Leistungslimit" verwendet, um
1. beim Shutdown des Inverters ein bekanntes Limit einzustellen und
2. in `setNewPowerLimit()` zu prüfen, ob das berechnete Limit kleiner ist als das untere Leistungslimit.

Punkt 1 können wir wegdiskutieren: Das bekannte Limit beim Shutdown ist eigentlich nicht wichtig. Wichtig ist nur, dass er ausgeschaltet wird. Da wir inzwischen `updateInverter()` haben, wird sichergestellt, dass beim Wiedereinschalten des Inverters das dazu gewünschte Limit schon eingestellt ist, bevor der Inverter beginnt zu produzieren.

Punkt 2 ist weiterhin relevant. Es macht eben keinen Sinn den Inverter mit 20W zu betreiben, wenn gerade 25W vom Lageregler erzeugt werden, weil dann die Effizienz sehr schlecht ist (und wie wir neuerdings wissen, schalten die WR dann häufig auch einfach ab, weil sie schwingen). Im Grunde "siehe #809". Dieses Konzept muss bleiben.

Ich schlage vor, dass wir dieses Konzept neu implementieren, ohne Einstellmöglichkeit. Wenn das errechnete Limit des Ziel-Inverters unter einen Inverter-spezifischen Wert liegt, dann wird er weiterhin abgeschaltet. Die Schwelle wird errechnet (TODO) und ist nicht konfigurierbar. In der Praxis spielt sie nur bei Solar-Passthrough eine Rolle, weil wir annehmen, dass die Leute (zumindest zukünftig) die Einstellung "Unteres Leistungslimit" als "Grundlast" verstehen und entsprechend hoch ansetzen (100W+).

Ohne Solar-Passthrough gibt's nur noch
1. Inverter aus, weil Batterie leer.
2. Inverter an mit mindestens dem unteren Leistungslimit

Zu Punkt 1 gibt es wohl keine abweichenden Gedanken.

Zur Punkt 2: Falls es keinen PowerMeter gibt oder dessen Wert veraltet ist, werden die auf dem PowerMeter basierenden Rechnungen nicht angestellt in `calcPowerLimit()`, sondern als Limit wird die Einstellung "Unteres Leistungslimit" (oder neu: "Grundlast") angenommen. Danach wird allerdings trotzdem auf die solare Leistung begrenzt, wenn die Batterie gar nicht entladen werden darf. Und im Falle von Full Solar-Passthrough wird dann trotzdem aufgedreht. Und falls weder die Batterie entladen werden darf, noch es solare Leistung gibt vom Laderegler, dann wird weiterhin mit `Status::NoEnergy` der WR ausgeschaltet.

Das klingt ehrlich gesagt sehr gut, finde ich!

Meinungen? @spcqike @PhilJaro @der-peterh

PhilJaro, 2024-04-07:
> Das klingt ehrlich gesagt sehr gut, finde ich!

Ja!
Gute Lösung die zumindest aus meiner Sicht der Dinge die Probleme und Missverständnisse lösen würde. 

Danke dir! (Zumindest schonmal für die gedankliche Umsetzung 😉)

spcqike, 2024-04-08:
Ich finde die Gedanken zur Umsetzung auch sehr stimmig und gut. Das bringt dem Projekt viel Mehrwert und bringt Möglichkeiten der Anwendung, die out-of-the-Box fertigsysteme bereits mitbringen. 
(Die openDTU-ob kann bisher zwar weitreichendere Funktionen, ein simples dauerhaftes einspeisen und abschalten, wenn die Batterie leer geht, aber nicht :) ) 

der ein oder andere User würde seine Grundlast ggf zeitbasiert festlegen / ändern. So wie Anker und co es anbieten. Das geht hier dann im erst Schritt per MQTT, später vielleicht ebenfalls direkt aus der openDTU (es gibt im upstream wünsche eines schedulers um zeitgesteuert das Limit vorgeben zu können) 

> Ich schlage vor, dass wir dieses Konzept neu implementieren, ohne Einstellmöglichkeit. Wenn das errechnete Limit des Ziel-Inverters unter einen Inverter-spezifischen Wert liegt, dann wird er weiterhin abgeschaltet. Die Schwelle wird errechnet (TODO) und ist nicht konfigurierbar.

das würde ich tatsächlich nicht machen, wenn es nicht zwingend nötig ist. Meiner Meinung nach bleibt es sinnvoll, das Limit selbst vorgeben zu können. Die wechselrichter und deren Bauteile unterliegen Fertigungstoleranzen. 30W mögen bei einem Benutzer gehen, beim zweiten nicht. Also muss eine feste Grenze konservativ sehr hoch angesetzt werden? Und Benutzer mit Hardware, wo alles „genau passt“, die kleinere Limits nutzen könnten, werden an feste hohe Limits gebunden. (Vergleich overclocking bei PCs) 
Ich würde das Limit wenigstens in der config als „expert Einstellung“ beibehalten, so wie aktuell das powermeter interval (oder ist das inzwischen in der webUI? Weiß ich grade nicht :) ) 

> Es macht eben keinen Sinn den Inverter mit 20W zu betreiben, wenn gerade 25W vom Lageregler erzeugt werden, weil dann die Effizienz sehr schlecht ist 

da hast du recht. Das trifft aber nur zu, wenn die Batterie bereits voll ist, oder? Andernfalls macht es Sinn seine 5W Überschuss zu speichern. Oder habe ich dich falsch verstanden?

schlimmchen, 2024-04-08:
> da hast du recht. Das trifft aber nur zu, wenn die Batterie bereits voll ist, oder? Andernfalls macht es Sinn seine 5W Überschuss zu speichern. Oder habe ich dich falsch verstanden?

Das trifft dann zu, wenn die Batterie **leer** ist, also wenn nur der solare Output genutzt werden darf. Die Zahlen habe ich gewürffelt. 25W Output abzgl. Verluste, also etwas weniger (ssagen wir 20W) AC Limit am Inverter.

> das würde ich tatsächlich nicht machen

Können solche Experten nicht ihre eigene Firmware bauen bzw. tun das sowieso? Die Firmware soll (das ist jedenfalls meine Motivation) für möglichst viele Leute funktionieren und *zu verstehen* sein. Jeder Schalter versteilert die Lernkurve, vor allem jene mit "Freitext", wo man sich also echt Gedanken machen muss.

Es geht um einen kurzen Zeitraum während der Dämmerung... Dafür einen neuen Schalter einführen (und dokumentieren und pflegen)... :frowning:

Was ich mir vorstellen könnte ist es einen Schalter "Experteneinstellungen" einzuführen im DPL. Wenn der aktiv ist, werden eine Reihe von Schaltern sichtbar und deren Werte vom DPL benutzt, die ansonsten sinnvolle Default-Werte nutzen. Beispiel: Target Consumption wird in der Konfig auf 10W gesetzt, wenn "Experteneinstellung" aus ist und die DPL Einstellungen gespeichert werden. Und die ganzen Schwellwerte werden reduziert auf "Welche Nennspannung hat die Batterie" (Dropdown). Fänd ich ziemlich geil, ist aber ein anderes Feature.

Zurück zum "Mindestlimit" im Sinne von "drunter könnte der Inverter instabil sein": Ich schlage 10W pro Eingang vor plus 25% Marge. Beim HM-1500 sind das 50W. Und wie gesagt: Diese Logik ist nur relevant für Solar-Passthrough während die Batterie sich in einem Ladezyklus befindet und zeitlich nur in der Dämmerung relevant. Wir können in der Konsole eine freundliche, erklärende Nachricht dazu ausgeben, dass der Inverter nicht angeschaltet wird, obwohl es solare Leistung gibt, weil die Software einschätzt, dass sich das nicht lohnt (Gefahr des Schwingens, niedrige Effizienz). Das finde ich "fair", und dann möge sich jemand beschweren, dass das für sein/ihr Setup nicht passt und nicht ertragen werden kann, dass für zehn Minuten am Morgen 20W Leistung in die Batterie gespeichert wird, die auch sofort hätte eingespeist werden können. Zwischen den Zeilen ließt du schon, dass ich das als Genörgel abtun wollen würde, um ehrlich zu sein.

PhilJaro, 2024-04-08:
> Ich schlage 10W pro Eingang vor plus 25% Marge

Hierzu noch eine Verständnisfrage, sprechen wir von pro Eingang oder pro MPPT, da ich das auch schon ein paar mal gelesen hatte…

schlimmchen, 2024-04-08:
Das ist eine gute Frage, und mir schwebt 10W pro Eingang vor, aus pragmatischen Gründen: Die Anzahl Eingänge lässt sich deutlich leichter bestimmen als die Anzahl MPPTs. Ob 12,5W für einen Hoymiles mit einem Eingang wirklich genug sind, weiß ich nicht. Ich stell meinen HM-400 mal auf 12W ein und schaue, was passiert (allerdings hängt der an einem Solarmodul, nicht an der Batterie).

der-peterh, 2024-04-08:
Ich freue mich auf eine Umsetzung 👍

Bei errechneten unteren Leistungswerten für eine Abschaltung bin ich übrigens ebenfalls etwas skeptisch: die Formel müsste ja irgendwie auch berücksichtigen, wie viele DC Stufen das jeweilige WR Modell verbaut hat (1, 2, oder 4), da ja wohl dort ein Großteil der Verluste entstehen und für stabilen Betrieb wohl auch eine Abhängigkeit zum Batterie System (24/48 V) besteht? Ein HM-300 an 48V wird vielleicht zwischen 50 und 300 Watt sinnvoll funktionieren, ein HM-1500 an 24 Volt vielleicht nur zwischen 160 und 900 Watt oder so...

Entsprechend würde ich eher auf geeigneten erklärenden Text bzw Namensgebung der Parameter setzen, wie etwa: 
- Leistungsgrenzen WR für stabilen Betrieb (Ober und Untergrenze)
- Grundlast Leistung WR (auch bei fehlenden Daten der Verbrauchsmessung)

Die Grundlast Leistung darf dann halt irgendwo im stabilen Betriebsfenster liegen.
Das ist zwar ein Wert mehr als wirklich nötig, dafür sollte aber klarer werden, was die Anwender jeweils ermitteln und konfigurieren müssen... 

Ich bin ja auch nur 'Entwickler' weil mir ein Feature fehlte ;)

der-peterh, 2024-04-08:
Und kurz meine 50 Cent zum Thema Laderegler und Bypass:
Das ist doch nach meinem bisherigen Verständnis nur exakt dann relevant, wenn genau ein einziger Victron Laderegler zum Einsatz kommt, oder?
Bei mehreren Reglern wegen Ost-West Ausrichtung oder sonstigen Konstellationen auch mit anderen Ladereglern kann eine Bypass Kalkulation ja nicht ermittelt werden, oder?
Ich würde tendenziell daher eher den allgemeinen Fall betrachten, dass die Batterie (zu) voll wird und das BMS droht abzuschalten, da nicht alle BMS ohne manuelle Intervention wieder einschalten bzw. wieder Leistung liefern.
Hier könnte also im Bereich der oberen Füllgrenze die Leistung des WR soweit angehoben werden, dass ein oberer Spannungswert möglichst nicht überschritten bzw gehalten wird, selbst wenn zu viel Energie erzeugt würde...

Ich stelle dieses Thema aber lieber als weiteren Feature request ein als es hier mit der unteren Leistungsgrenze bzw. Grundlast und Fallback being fehlenden Daten der Verbrauchsmessung zu vermischen...

spcqike, 2024-04-08:
> Ich würde tendenziell daher eher den allgemeinen Fall betrachten, dass die Batterie (zu) voll wird und das BMS droht abzuschalten, da nicht alle BMS ohne manuelle Intervention wieder einschalten bzw. wieder Leistung liefern.

das ist Aufgabe des Ladereglers. egal ob einer oder zwei, wenn Victron verwendet werden können diese untereinander kommunizieren und müssen irgendwann aufhören zu laden. das ist kein openDTU Thema.

> Hier könnte also im Bereich der oberen Füllgrenze die Leistung des WR soweit angehoben werden, dass ein oberer Spannungswert möglichst nicht überschritten bzw gehalten wird, selbst wenn zu viel Energie erzeugt würde...

Ok, ich sage das dann meinen 3.5kWp und dem 2kVA Wechselrichter , sofern ich mal ne Batterie habe :) 
Nein: das reicht nicht. Wie gesagt, Ladeschluss gehört in den Ladecontroller. Es gibt sicherlich mehr als genug User, die ein Overpanelling betreiben, also mehr Modulleistung auf dem Dach haben, als Wechselrichterleistung im Keller. ein einfach "dreh den Wechselrichter einfach auf" reicht da halt nicht, wenn von oben doch noch mehr kommt.

> Zwischen den Zeilen ließt du schon, dass ich das als Genörgel abtun wollen würde, um ehrlich zu sein.

hast ja irgendwie recht :) wird wohl nicht all zu oft vorkommen und die "Verluste" (sofern es denn welche sind) halten sich in Grenzen :) 

wie siehst du die Implementierung denn letzten Endes? Man kann ja weiterhin zwischen Solar-passthrough ja und nein unterscheiden, oder?

der einfache Fall ist sicherlich S-P nein. also "erzeuge mindestens Grundlast. erzeuge mehr, wenn der PowerMeter mehr meldet. so lange die Batterie es erlaubt"

komplexer wird es mit S-P. mit aktivem S-P kann die "Grundlast" dennoch unterschritten werden? So zumindest hab ich dich verstanden. 
In meinem Kopf war das bisher kein Thema. Ich dachte "Batterie_Start_Spannung erreicht -> erzeuge Grundlast" und "S-P Startspannung erreicht -> erzeuge dynamisch auch mehr als Grundlast, je nach PowerMeter". 
Aber das kann daran liegen, dass ich bisher keine Batterie nutze und die vorhandenen Funktionen und Funktionsweisen noch nicht durchspielen und testen konnte.

schlimmchen, 2024-04-08:
> Bei mehreren Reglern wegen Ost-West Ausrichtung oder sonstigen Konstellationen auch mit anderen Ladereglern kann eine Bypass Kalkulation ja nicht ermittelt werden, oder?

Doch. Ist im letzten Release auch schon drin: https://github.com/helgeerbe/OpenDTU-OnBattery/commit/75541be2482e3bf0042c98649b77ee8d5648d5cf (zweiter Laderegler). Im nächsten Release wird auch die VE.Direct Netzwerk-Gesamtleistung zur Verfügung stehenn (wenn ein Laderegler angesprochen werden kann, also TX verbunden ist).

> Ich stelle dieses Thema aber lieber als weiteren Feature request ein als es hier mit der unteren Leistungsgrenze bzw. Grundlast und Fallback being fehlenden Daten der Verbrauchsmessung zu vermischen...

Naja... Ist meiner Meinung nach obsolet. Und wie spcqike schon erwähnt hat, ist dafür der Laderegler zuständig. Der muss halt entsprechend eingestellt sein.

> mit aktivem S-P kann die "Grundlast" dennoch unterschritten werden?

Ja. Der Bereich zwischen "Mindestleitung für stabilen Betrieb" und "Einstellung Grundlast" ist der verbleibende lästige Fall.

Können wir es nicht etwas einfacher machen? Es bleibt bei einer Einstellung, also "unteres Leistungslimit". Und wenn solar-passthrough aktiv ist aber weniger liefert als "unteres Leistungslimit" während die Batterie "leer" ist, dann ist (bleibt) der WR aus und die Leistung geht stattdessen in die Batterie? Für alle anderen Fälle (kein PowerMeter oder Werte veraltet, aber Batterie kann entladen werden) nehmen wir dann das untere Leistungslimit an und lassen den WR an? Ist auch nicht geil, vor allem im Winter, wenn die solare Leistung vielleicht wirklich über einen langen Zeitraum gar nicht ausreicht, um die Grundlast zu decken?

FYI: Mein HM-400 funktioniert mit 12W stabil (an einem Solarmodul).

Mir wäre immer noch ein konservatives, automatisch bestimmtest "stabiles Limit" lieber, allerdings hat @der-peterh schon recht, dass dieser Wert für verschiedene Systeme durchaus verschieden sein kann. Da halte ich dann aber nochmal dagegen, was ich schon geschrieben habe: Es betrifft nur einen Fall, der auch noch zeitlich begrenzt ist, und wer da jammert dass er zehn Minuten mehr Leistung hätte direkt umsetzen können statt in der Batterie zu speichern, der möge das dann erstmal rechtfertigen, dass ihm etwas entgangen ist.

Ein kleines System, z.b. nur ein Panel und ein HM-400, wird ja auch ein sehr niedriges stabiles Limit "zugewiesen bekommen", 12W wie gesagt. Ein HM-600 25W. Aber da sind dann sinnvoller Weise auch mind. zwei Module im Spiel, also 13W pro Modul. Die kommen an einem "normalen" Tag auf jeden Fall auch raus, oder? D.h. da würde dann auch irgendwann der WR eingeschaltet werden und das Limit richtet sich nach Solar-Passthrough?

spcqike, 2024-04-08:
> Und wenn solar-passthrough aktiv ist aber weniger liefert als "unteres Leistungslimit" während die Batterie "leer" ist, dann ist (bleibt) der WR aus und die Leistung geht stattdessen in die Batterie?

genau so hab ich es am Anfang auch verstanden und gedacht. 

Dann bleibt dort nur die Frage, wie weiter verfahren würde.
Wenn es mehr liefert, startet der Wechselrichter. Mit der "Grundlast"? mit dem min(PowerMeter / Nulleinspeisung ;Solar_Input), was halt zutrifft? Wenn Solar_Input wieder fällt, geht der WR wieder aus? Oder bleibt er an, nachdem er einmal gestartet wurde, auch wenn Batterie_Start_Treshold noch nicht erreicht wurde?

> Ein HM-600 25W. Aber da sind dann sinnvoller Weise auch mind. zwei Module im Spiel, also 13W pro Modul. Die kommen an einem "normalen" Tag auf jeden Fall auch raus, oder?

bei meinem HM-600 an 2 425W Modulen (144 Halbzellen, also ~42V 10A am MPP) sieht der bewölkte Start so aus

[image]

Es geht also ab ~2.5W pro Eingang los. Allerdings mit einem vom DPL gesetzten Limit von ~300W. Ob die Hoymiles interne Regelung anders reagiert, wenn sie einfach nur 5W liefern kann bzw. wenn sie per Limit nur 5W liefern soll (und sich dabei "aufhängt") weiß ich nicht.

schlimmchen, 2024-04-08:
> Dann wäre das letzte verbleibende Argument für eine WR-Leistungserhöhung bei maximal voller Batterie lediglich, erzeugbaren Strom lieber so weit wie möglich ans Netz zu 'verschenken', statt mittels Laderegler abzuregeln ... Reicht sicher nicht für hohe Priorität ;)

Ich bin verwirrt. Das ist entweder Full Solar-Passthrough, oder ich hab deine Aussage nicht verstanden. Full-Solar-Passthrough beginnst du kurz vorm Erreichen der Ladeschlussspannung. Wenn dann immer noch mehr Leistung aus dem Laderegler kommt, als die WR wegschaffen können, dann erreichst du trotzdem die Ladeschlusspannung und der Laderegler regelt die Leistung runter.

Also verstehe ich es richtig, dass ihr auch der Meinung seid, der einfache Ansatz könnte ausreichen? Das ist jener, wo es weder eine zweite Einstellung gibt, noch ein errechnetes "Stabilitätslimit", sondern der WR bleibt aus im Falle dass die Batterie "leer" ist, aber solare Leistung aus dem Laderegler kommt, diese allerdings nicht ausreicht für das "Untere Leistungslimit" bzw. "Grundlast"? In diesem Fall gibts eine Ausgabe im Log (auch ohne Verbose Logging).

Wenn die Batterie aber Leistung abgeben kann, dann mindestens "Grundlast", weil niemand diesen Wert so hoch einstellen sollte, dass jemals ein signifikanter Teil dieser Leistung ins Netz geht.

Dann wird die UI passend gemacht mit Hinweistext und die Doku wird fit entsprechend fit gemacht.

Zweites Feature ist dann der Betrieb ganz ohne PowerMeter.

PhilJaro, 2024-04-08:
> Also verstehe ich es richtig, dass ihr auch der Meinung seid, der einfache Ansatz könnte ausreichen? Das ist jener, wo es weder eine zweite Einstellung gibt, noch ein errechnetes "Stabilitätslimit", [...]

Ja, finde ich auch gut so!

spcqike, 2024-04-08:
> Dann wäre das letzte verbleibende Argument für eine WR-Leistungserhöhung bei maximal voller Batterie lediglich, erzeugbaren Strom lieber so weit wie möglich ans Netz zu 'verschenken', statt mittels Laderegler abzuregeln ... Reicht sicher nicht für hohe Priorität ;)

naja, verschenken geht nur, solange der Strom auch "kostenlos" ist. bei negativen Börsenpreisen ist "verschenken" nicht so cool.

[image]

gestern wurde letzten endes jede "gekaufte" kWh mit geplanten 2-3Cent/kWh vergütet (Day Ahead), warum auch immer war der Interday Preis aber bei bis zu -15Cent/kWh. eine private Einzelanlage fällt da sicher nicht so ins Gewicht, dennoch ist das bspw. ein Grund, warum ab einer gewissen kWp größe eine Direktvermarktung (inkl. der daraus resultierenden Drosselung bei Überkapazität) Pflicht ist. 

Würdest du deinen Strom auch "verschenken", wenn du pro "verschenkter" kWh 15Cent bezahlen müsstest? :) 

Die netzdienliche Berücksichtigung vom Börsenpreis (also dem Netzbetreiber tatsächlich etwas schenken oder doch zu seinen Kosten einfach einspeisen) wäre noch n cooles Feature :D auch für AC gestütztes Aufladen der Batterie. Aber das ist n ganz anderes Thema :)

> Also verstehe ich es richtig, dass ihr auch der Meinung seid, der einfache Ansatz könnte ausreichen?

Ja das denke ich. Wobei für mich als Frage nur über bleibt, wie damit umgegangen werden soll, wenn die PV Leistung wieder < Grundlast fällt, aber die Batterie die Startspannung noch nicht erreicht hat. WR wieder ausschalten, oder weiter entladen, bis Batterie Stoppspannung erreicht wird?

schlimmchen, 2024-04-08:
> Wobei für mich als Frage nur über bleibt, wie damit umgegangen werden soll, wenn die PV Leistung wieder < Grundlast fällt, aber die Batterie die Startspannung noch nicht erreicht hat. WR wieder ausschalten, oder weiter entladen, bis Batterie Stoppspannung erreicht wird?

Besonders dämlich ist es natürlich, wenn der hier an ist:

[image]

aber der WR ausgeschaltet wird, sobald die solare Leistung unter Grundlast sinkt, und dann wieder eingeschaltet wird, wenn aufgrund noch weiter gesunkener solarer Leistung (unter 20W) dann doch ein Entladezyklus beginnt...

Kurz am Rande: Ich hab das Problem nicht, weil meine Start-Spannung mit 52V eher niedrig ist. Sobald die Batterie einigermaßen Energie hat, nutze ich diese (Stop-Spannung 51V).

Aber andere wünschen sich ja mit einer hohen Start-Spannung, dass der Akku möglichst stark geladen wird, und stellen dann mit "Batterie zur Nacht entladen" ein, dass sie die dann "nachts" doch entleeren wollen. Und die sehen dann an einem gewöhnlichen Abend, nachdem die Start-Spannung nicht erreicht wurde, dass der WR ausgechaltet wird, und dann zur "Nacht" wieder eingeschaltet.

Ich tendiere dazu, dass das dann so ist. Mit der Begründung, dass sich der Betrieb des WR in diesem Leistungs-Band sowieso nicht lohnt und die Energie dann halt einen Umweg über die Batterie macht. :man_shrugging:

In #671 habe ich jedenfalls gelernt, dass es ein Anliegen ist, die Batterie nicht "versehentlich" untertags zu entladen wegen einer dunklen Wolke. Nach diese Logik sollte dann auch der WR stoppen, wenn die solare Leistung < Grundlast wird, und dann im Zweifel später wieder starten wenn "klar" ist, dass wirklich Nacht ist.

spcqike, 2024-04-08:
klingt sinnig.

vielleicht "einfach" einmal probieren und gucken, wie es am Ende reagiert :)

Ich befürworte diese Anpassung, kann sie aber leider selbst nicht testen.

> Kurz am Rande: Ich hab das Problem nicht, weil meine Start-Spannung mit 52V eher niedrig ist. Sobald die Batterie einigermaßen Energie hat, nutze ich diese (Stop-Spannung 51V).

so würde ich es später wohl auch handhaben.

der-peterh, 2024-04-08:
> Ich bin verwirrt. Das ist entweder Full Solar-Passthrough, oder ich hab deine Aussage nicht verstanden. Full-Solar-Passthrough beginnst du kurz vorm Erreichen der Ladeschlussspannung. Wenn dann [...]
So hatte ich es auch verstanden, dachte aber, dass alle Passthrough Einstellungen nur für Victron Laderegler gelten? Ich habe doch nur einfache China Regler ohne Bus...
;)

closed 2024-04-15 by schlimmchen