hoylabs/OpenDTU-OnBattery #768: [Request] Dynamic Power Limiter (DPL) on two HM-1500 in parallel

gitisgreat2023, 2024-03-18:
### Is your feature request related to a problem? Please describe.

With my Huawei charger I'm limited to a HM-1500 as highest power inverter.

To cover peak loads of 3000 W I would love to have a second HM-1500 in parallel.
That would make the typical setup more balanced: the Huawei AC charger is capable of 3kW (the 4875 even 4kW), but the inverter is only 50% of that. Typically the load has much higher peak powers than the photovoltaic source, but the power constraint is currently on the inverter.

### Describe the solution you'd like

My proposal is to give the user the choice which HM is the first to deal with the request. As soon as that reaches its maximum the second is turned on as well, and is turn off if the maximal request power drops again below the rating of the first inverter. Hereby you avoid issues with efficiency at lower power and the logic is simple.

By being able to choose the primary inverter, the user can distribute the usage over time (for example switching each year the primary inverter).

By the cascading principle, the overhead should also be ok, most of the time one inverter is off.

### Describe alternatives you've considered

Setting up another OpenDTU-OnBattery. But I'm not even sure it would work with my battery setup (share can data, etc etc).

### Additional context

An increased polling time during the 2 inverter scenario would be okay. Also limiting to two inverters, that would avoid someone hooks up 5 HM-1500 or something similarly stupid and burns down his house.

schlimmchen, 2024-03-18:
> With my Huawei charger I'm limited to a HM-1500 as highest power inverter.

Why? The battery (current, or capacity and voltage, respectively) should limit the output, if it does limit such an inverter in the first place.

> My proposal is to give the user the choice which HM is the first to deal with the request.

What request?

Is this about the DPL? I really don't understand what is asked here... Support for more than one inverter in the DPL is a long-standing request, what is new in this issue?

gitisgreat2023, 2024-03-18:
The Huawei can currently only be combined with the HM series, not the HMT/HMS (for example the HMT-2250). So the highest power inverter I can use with my Huawei setup is the HM-1500.

The request for the inverter power: say one inverters are parallel connected, inverter 1 and inverter 2, both HM-1500.
With a requested inverter power of say 1300W only inverter 1 is active, with request inverter power of 2800W inverter 1 delivers 1500W and inverter 2 the remaining 1300W.

"Support for more than one inverter in the DPL is a long-standing request, what is new in this issue?"
-> okay, which issue is that? I'll write there then and close this one with referencing to that request.

schlimmchen, 2024-03-18:
I just noticed that this is a duplicate of your own issue #601. Support of multiple inverters will not come sooner if more issues are opened. I assume that opening this issue was an honest mistake :monocle_face:

closed 2024-03-18 by schlimmchen

gitisgreat2023, 2024-03-18:
argh... yeah... forgot about that one... that's then a duplicate yes.... sorry.