hoylabs/OpenDTU-OnBattery #774: Victron SmartShunt Charged/Discharged wrong Values

PhilJaro, 2024-03-19:
### What happened?

Every once in a while the SmartShunt values for Charged and Discharged energy are set to 0. This resolves in wrong Energy data in HomeAssistant.
[image]

### To Reproduce Bug

Maybe the SmartShunt delivers incomplete data and OpenDtu isn’t proving if it is valid. But I am not 100% sure about what triggers the wrong Values. 
Rebooting of the OpenDtu or disconnecting of the VeDirect doesn’t trigger them. 

### Expected Behavior

For Victron MPPT  I can’t see this behavior. When I interpret the code right, there is a check if the Data is Valid and this isn’t the case for the SmartShunt. 

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

g9003267

schlimmchen, 2024-03-29:
I think I know what happens: As per @philippsandhaus comments in `VeDirectShuntController.cpp`, the SmartShunt sends its VE.Direct TEXT infos in two frames. (Only) If the PID was decoded, the temporary frame is handed to the upper software layers. This achieves that a full set of info is propagated, but only if the communication is flawless. In your case, if the frame without the PID in it is broken, its invalid data is kept in the temporary data structure and is later used once the PID was decoded from another frame.

I am currently working to fix #689 properly and this issue is closely related to it since the fragmentation is handled incorrectly in both cases.

schlimmchen, 2024-03-29:
@PhilJaro Could you please test the firmware built in the [PR run](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8485520347?pr=814) of #814? Does the problem occur so often that you can be sure that it is gone after a reasonable amount of time running the new firmware? I am optimistic that the changes in #814 should have a positive impact on this issue.

PhilJaro, 2024-03-30:
Sure, I will test it. I have tried many things to reproduce the error but I couldn’t. Once it runs stable for a week and then it fails 10 times a day. 

The only thing I found out is, if I remove the Antenna completely and set the refresh rate very high(1 or 2sec), the signal to the Hoymiles gets worse and the whole system becomes unstable and laggy. And now the error occurred much more often.

PhilJaro, 2024-04-02:
So after a few days I can tell, i seems to work. I could not recreate the false values. 
But for now my Console semms to be unreadable, because of the amount of logs produced by the VE-Direct without verbose logging.

`[VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:07.228 > Middle missing
12:10:07.237 > Request retransmit: 3
12:10:07.374 > Middle missing
12:10:07.385 > Request retransmit: 3
12:10:07.393 > [DPL::announceStatus] the system is stable, the last power limit is still valid
12:10:07.432 > Websocket: [/livedata][9] disconnect
12:10:07.447 > Websocket: [/vedirectlivedata][9] disconnect
12:10:07.461 > Websocket: [/batterylivedata][9] disconnect
12:10:07.488 > Success
12:10:08.036 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:08.057 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:08.084 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:08.327 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:08.488 > Fetch inverter: 116183073478
12:10:09.124 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:09.126 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:09.126 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:09.127 > Success
12:10:09.160 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:10.083 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:10.084 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:10.093 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:10.185 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:10.596 > Fetch inverter: 116183073478
12:10:11.107 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:11.109 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:11.110 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:11.110 > Success
12:10:11.209 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:12.131 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:12.133 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:12.134 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:12.174 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:12.540 > Fetch inverter: 116183073478
12:10:13.054 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:13.056 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:13.057 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:13.057 > Success
12:10:13.360 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:14.013 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:14.028 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:14.037 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:14.187 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:14.588 > Fetch inverter: 116183073478
12:10:14.795 > PowerMeterClass: TotalPower: 11.00
12:10:15.103 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:15.105 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:15.105 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:15.106 > Success
12:10:15.185 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:15.407 > [DPL::updateInverter] sending limit of 11.6 % (174 W respectively), max output is 1500 W
12:10:15.430 > [DPL::announceStatus] waiting for a power limit command to complete
12:10:16.002 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:16.010 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:16.023 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:16.190 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:17.045 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:17.047 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:17.048 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:17.195 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:17.482 > All missing
12:10:17.484 > Nothing received, resend whole request
12:10:18.071 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:18.073 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:18.073 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:18.274 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:18.991 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:18.993 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:19.000 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:19.299 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
12:10:19.550 > All missing
12:10:19.559 > Nothing received, resend whole request
12:10:20.011 > [VE.Direct SmartShunt 13/-1] Unknown text data 'AR' (value '0')
12:10:20.013 > [VE.Direct SmartShunt 13/-1] Unknown text data 'BMV' (value 'SMARTSHUNT 500A/50MV')
12:10:20.014 > [VE.Direct SmartShunt 13/-1] Unknown text data 'MON' (value '0')
12:10:20.205 > [VE.Direct MPPT 14/-1] Unknown text data 'IL' (value '0')
`

PR #814 (2024-04-02): Properly handle VE.Direct fragmentation
* add unique prefix to VE.Direct messages. this makes it easier to map debug messages to one-another.
* properly handle fragmented VE.Direct messages. queue every text event until the frame was checked by it checksum. then process the data directly into the buffer struct. do not clear the buffer struct, so it will always include the most recent value of a particular data point.
* use float rather than double. double precision floating point numbers are not needed to handle VE.Direct values. handling double is implemented in software and hence *much* more [... cut]
closed 2024-04-02 by schlimmchen