hoylabs/OpenDTU-OnBattery #776: [Request] Cut-Off für Solar-Passthrough via sunset / sunrise (anstatt 20 W Schwelle)

ottelo9, 2024-03-19:
### Describe the solution you'd like

Cut-Off für Solar-Passthrough bzw. die Nutzung der Batterie an die Zeit (sunrise/sunset) knüpfen. Somit wird gewährleistet, dass der Cut-Off für Solar-Passthrough nur fix nur einmal stattfindet. 

[
[OpenDTU-OnBattery/src/PowerLimiter.cpp](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/7d4a30dde43667dbf91ba988df499ee917717486/src/PowerLimiter.cpp#L254-L260)

Lines 254 to 260 in [7d4a30d](https://github.com/helgeerbe/OpenDTU-OnBattery/commit/7d4a30dde43667dbf91ba988df499ee917717486)

 // TODO(schlimmchen): should be supported by sunrise and sunset, such 
 // that a thunderstorm or other events that drastically lower the solar 
 // power do not cause the start of a discharge cycle during the day. 
 if (config.PowerLimiter.SolarPassThroughEnabled && 
         config.PowerLimiter.BatteryAlwaysUseAtNight) { 
     return getSolarPower() == 0; 
 }](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/7d4a30dde43667dbf91ba988df499ee917717486/src/PowerLimiter.cpp#L254-L260)

### Additional context

https://github.com/helgeerbe/OpenDTU-OnBattery/issues/773#issuecomment-2007952869

schlimmchen, 2024-03-19:
Siehe auch #671 (habe ich gerade wiederentdeckt...) :speak_no_evil:

ottelo9, 2024-03-19:
Stimmt hab ich letztens auch erst gelesen. Gehirn …

AndreasBoehm, 2024-08-08:
Sollte sich erledigt haben wenn https://github.com/helgeerbe/OpenDTU-OnBattery/pull/1126 gemerged wurde

closed 2024-08-15 by schlimmchen