hoylabs/OpenDTU-OnBattery #784: Solar-Passthrough mit "Entleeren zur Nacht" stoppt Entladen der Batterie zu früh

schlimmchen, 2024-03-22:
### What happened?

Neuerdings habe ich die Battery Drain Strategy auf "Empty at Night" eingestellt, bzw. den neuen Schalter "Batterie nachts sogar teilweise geladen nutzen" eingeschaltet. Jetzt stelle ich fest, dass das "morgens" ungewollte Folgen hat: Sobald wieder Sonne da ist, gilt die Solar-Passthrough-Beschränkung, d.h. die Batterie wird nicht weiter entladen und mein Netzbezug wird nicht gedeckt, weil die solare Leistung die Wechselrichterleistung begrenzt.

### To Reproduce Bug

"Batterie nachts sogar teilweise geladen nutzen" einschalten und bei Sonnenaufgang den DPL beobachten während die Batterie noch reichlich Energie hat.

### Expected Behavior

Sinnvoll fände ich es, wenn entweder
* nur die Stop-Threshold den Entladezyklus der Batterie beendet
* oder *genügend* solare Leistung den Entladezyklus der Batterie beendet (und natürlich die Stop-Threshold).

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

1234

### Relevant log/trace output

```Shell
08:33:52.083 > [DPL::loop] ******************* ENTER **********************
08:33:52.088 > [DPL::loop] PowerMeter: 207 W, target consumption: -10 W, solar power: 184 W
08:33:52.093 > [DPL::loop] battery interface enabled, SoC: 63 %, StartTH: 45 %, StopTH: 40 %, SoC age: 1 s, ignore: yes
08:33:52.095 > [DPL::getBatteryVoltage] BMS: 52.68 V, MPPT: 52.84 V, inverter: 52.80 V, returning: 52.68V
08:33:52.099 > [DPL::loop] dcVoltage: 52.68 V, loadCorrectedVoltage: 52.73 V, StartTH: 53.00 V, StopTH: 51.00 V
08:33:52.100 > [DPL::loop] StartTH reached: no, StopTH reached: no, inverter is producing
08:33:52.101 > [DPL::loop] battery discharging prevented, SolarPT enabled, use at night: 1
08:33:52.103 > [DPL::loop] Consuming Solar Power Only -> solarPowerAC: 172 W, newPowerLimit: 380 W
08:33:52.105 > [DPL::setNewPowerLimit] calculated: 172 W, requesting: 172 W, reported: 165 W, diff: 7 W, hysteresis: 30 W
```

### Anything else?

Weil das eine signifikante Änderung im Verhalten wäre, möchte ich das nicht einfach "korrigieren", sondern eine Weile lang Meinungen hierzu einsammeln.

spcqike, 2024-03-22:
Ich kann es zwar nicht testen :D aber meine Gedanken dazu:

wo machst du dann tagsüber den Schnitt? Ich kann verstehen, dass du beim Sonnenaufgang gern weiterhin den Akku nutzen möchtest. Wenn das nun aber anhand der verfügbaren Solarleistung festgemacht wird, was passiert, wenn tagsüber dein Verbrauch auf  ``> PV Leistung`` steigt? 

wäre ein ``Batterie nachts sogar teilweise geladen nutzen`` dann nicht eher ein ``Solar-Passthrough mit Batterieunterstützung``? oder bräuchte es ein zusätzliches Limit, bis wohin die Batterie unterstützen darf? (bspw. das bereits diskutierte untere Limit, aber nicht als Startbedingung sondern als "Fallback" bzw Mindesteinspeiseleistung?)

schlimmchen, 2024-03-22:
> wäre ein Batterie nachts sogar teilweise geladen nutzen dann nicht eher ein Solar-Passthrough mit Batterieunterstützung?

Das ist aber sehr viel weniger konkret als die bisher gewählte Beschriftung^^ Ich weiß auch nicht, was du damit meinst...

Dass das Problem auch tagsüber/abends existiert, ist mir bewusst. Und das ist nicht trivial zu entscheiden. Die Frage lautet ja: "Lohnt es sich noch zu laden oder wars das eh für den Tag und ich kann genausogut anfangen, die Batterie zu entladen?".

spcqike, 2024-03-22:
> Lohnt es sich noch zu laden oder wars das eh für den Tag und ich kann genausogut anfangen, die Batterie zu entladen?

diese Entscheidung kann man wohl nur sinnvoll treffen, wenn man einen solar forecast einbaut und online die Prognose zieht. Nicht, dass eine Wolke um 12 Uhr mittags den tag bereits beendet. (Zusätzlich zur Uhrzeit / den bekannten Dämmerungen natürlich)

Man könnte früh morgens vielleicht unterstützen, bis die Grundlast erstmalig rein von der PV gedeckt wird. Und damit die Nacht für beendet erklären. 

> Das ist aber sehr viel weniger konkret als die bisher gewählte Beschriftung^^ Ich weiß auch nicht, was du damit meinst...

Meine Gedanken sind nicht immer gut nachvollziehbar und ich kann sie meist noch schlechter in Worte fassen :D 
ich meinte, dass es für eine rein leistungsbasierte Entscheidung (bspw unterstütze bis zur grundlast mit Batterie) dann ja nicht mehr auf die Nacht begrenzt ist. Auch in deinem Fall kommt ja bereits etwas von der PV, die Nacht ist damit ja bereits beendet. Möchtest du die Batterie also in den Morgenstunden zur Unterstützung weiter nutzen, passt die aktuelle Beschreibung nicht mehr :) 

ich sehe aktuell drei unterschiedliche, für mich aber plausible Möglichkeiten 

1. unterstützen aus Batterie bis Solarleistung erstmalig > bedarf. Dann die Batterie „abschalten“ und umstellen auf den aktuellen Tag 
2. Generelle Unterstützung bis zu einem gewissen Leistungswert. Bspw bis zur bekannten Grundlast. 
3. generelle Unterstützung ohne Limit. So wie es die fertig Lösungen auch machen (das sollte ja das normale Verhalten ohne passthrough sein, oder?)

helgeerbe, 2024-03-22:
Hallo @schlimmchen das Verhalten ist mir auch schon aufgefallen. Habe es bisher aber nicht wirklich als "Problem" gesehen.

Wenn ich das richtig verstanden habe, sprechen wir hier über die Zeitspanne bis zum Erreichen des Start-Schwellwertes. Damit besteht die Möglichkeit, dass ich morgens noch Energie in der Batterie hätte, die ich nutzen könnte. Problematisch wird es dann, wenn die Sonne so scheint, dass die Batterie voll wird. Da wäre es ggf. schön, wenn ich vorher die Batterie wirklich entladen hätte.

Andererseits könnte ich auch den Start-Schwellwert absenken und sofort die Batterie nutzen. Oder habe ich da einen gedanklichen Fehler?

Der Start-Schwellwert ist eher ein psychologisches Ding. Aus irgendwelchen Gründen wollten viele Nutzer einen Batteriepuffer aufbauen, um in der Nacht dann mit Sonnenstrom Fernzusehen. Spannend wäre das, wenn ich dynamische Stromtarife habe. Da machen aus meiner Sicht, entsprechende Ladestrategien auch Sinn.

Persönlich sehe ich die Batterie nur als Überlaufpuffer. Wenn ich ein Überangebot habe, dann in die Batterie. Wenn ich zu wenig Sonnenenergie habe, sofort mit der Batterie auffüllen, wenn es geht.

cerise21, 2024-03-22:
Meistens ist der Akku morgens leer, aber kürzlich habe ich das oben geschilderte frühe 'Abschschalten' der Akkuunterstützung auch bemerkt, und darauf in den DPL-Settings den 
> Schalter "Batterie nachts sogar teilweise geladen nutzen"

ausgemacht und den Wert für 'Start Threshold for Battery Discharging' heruntergesetzt, damit der Akku den morgendlichen Verbrauch noch unterstützen konnte.

Sowas wie
> unterstützen aus Batterie bis Solarleistung erstmalig > bedarf. Dann die Batterie „abschalten“ und umstellen auf den aktuellen Tag

fände ich auch gut, statt bei >20W MPPT-Power (das ist doch aktuell die Bedingung?) die Nacht für beendet zu erklären.
Wobei '>Bedarf' bedeutet, dass erstmalig die 'Target Grid Consumption' erreicht wird?

Was aber tun, wenn die Solarleistung dann wieder kurzzeitig sinkt, so dass die '>Bedarf'-Bedingung verletzt wird? 
Jetzt hätte ich die Akkuunterstützung wieder aktiviert, bis eine einstellbare Solarleistung überschritten wird. 
Hier braucht man einen neuen Schwellwert, weil sonst mit jedem Verbrauchspeak, dem der DPL träge hinterherläuft, die Akkuunterstützung wieder aktiviert und dann gar kein Unterschied zwischen Tag und Nacht mehr gemacht würde.

Hierfür vielleicht einen eingerückten 'Unterschalter': "Batterie morgens weiternutzen, bis 'MPPT Total Power' > xxx W" unter den Schalter "Batterie nachts sogar teilweise geladen nutzen". Zusätzlich ein Eingabefeld für das 'xxx W'-Limit; beide nur aktiv, wenn der 'Nachts sogar teilweise…'-Schalter EIN ist.

LukasVFL99, 2024-03-28:
Kann mir jemand sagen was ich hier für Werte eintragen muss?
[image]

CyberDem0n, 2024-05-05:
My solar panels are looking to south-west and besides that there is a shade from a tree until about 9 morning.
Therefore until 9 panels sometimes can't even cover base demands of the household and it would make total sense to continue using the battery. 

The tricky part is how to define up until which time battery should be allowed to be used. We can try to build something on top of standard opendtu feature that defines night hours, but the problem with this approach is that night hours may vary a lot depending on external environment.

The second option is to take some base household needs and apply some margin on top of it. For example, in my case the baseline is 150W, and until panels produce 200W we can effectively count it as night (if current time is before noon) and should start charging the battery only when panels produce more than 200W. Before 200W battery shouldn't be charged and all power sent to inverter (Full Solar-Passthrough) despite the fact we will be losing some of it.

Of course, if battery SOC is below configured thresholds it shouldn't be discharged.

And a few more thoughts outloudly. It seems that the current implementation takes input parameters from different sources and relying on the fact that panels produce enough power during the day to cover household need and charge the battery. That is, the current state (battery is charging on discharging) is not persistent anywhere. As a result we may experience a situation that after a short period of charging the battery we might start spending its energy, because it suddenly became dark (for example heavy clouds).
It could be avoided by using a persistent state of battery:
- if current state is discharging and panel output exceeds a threshold (200W in my case) - switch state to charging
- once in charging state don't let it discharge until other conditions are meet.

It is all very rough and maybe not very well thought, just my 2ct.

AndreasBoehm, 2024-08-16:
Da wir nun Tag und Nacht anhand des Sonnenauf- und Sonnenuntergangs ermitteln hat sich dieses Problem anscheinend auch gelöst.

Hier ein Screenshot in dem zu sehen ist das heute früh die Batterie noch benutzt wurde bis sie den Stop Threshold erreicht hat, obwohl es bereits `Tag` war und Solarleistung vorhanden war.

[image]

schlimmchen, 2024-08-16:
> das heute früh die Batterie noch benutzt wurde bis sie den Stop Threshold erreicht hat, obwohl es bereits Tag war und Solarleistung vorhanden war.

Der Code behauptet das Gegenteil. Oder zumindest ist es nicht so einfach: Wenn wir "nighttime discharging" machen und es wird Tag, dann wird die Batterie genau dann weiter entladen, wenn die **start** threshold (noch) erreicht ist.

https://github.com/helgeerbe/OpenDTU-OnBattery/blob/5d1d071c8a3a0820e96e1e31835faa7a5dbf1b9e/src/PowerLimiter.cpp#L245-L248

Es kann also weiterhin passieren, dass
* ein Entladezyklus gestartet wurde, weil es Nacht wurde (obwohl der Ladezyklus nicht vollständig war, wie gewünscht)
* es Tag wird und die solare Leistung aber gar nicht ausreicht, um den Haushaltsbedarf zu decken
* der Entladezyklus beendet wird, obwohl die Stop-Schwelle noch nicht erreicht wurde

Was nicht geht: `return !isStopThresholdReached();`? Dann würden wir zwar dieses Issue erschlagen, aber die Batterie würde anschließend potentiell entladen werden, obwohl der Benutzer wünscht, dass sie erstmal bis auf die Start-Schwelle geladen wird (Ladezyklus vervollständigen).

Die Abbruchbedingung für "nighttime discharging" sollten sein:
* Stop-Schwelle erreicht **oder**
* *genügend* solare Leistung vorhanden, um den Haushaltsbedarf zu decken.

Das ist die Idee dieses Issues.

CyberDem0n, 2024-08-22:
@schlimmchen yesterday was a sunny day with some accidental clouds and I was observing strange behavior with new firmware, which I think is a bug.

Time: ~14:00
Battery SOC: 60%
StartThreshold: 80%
StopThreshold: 15%
MaximumPowerLimit: 800W
Solar-Passthrough: enabled
Use battery at night even if only partially charged: on
Solar-Passthrough Losses: 7%

Panel Power was floating between 300W (clouds) and 1500W (sunny)
Avg household consumption 150W-200W.

I switched on the kettle and DPL set power limit to the maximum (800W), despite the next cloud floating around (panels were producing ~350W). And the pictogram was showing that is "on battery".

In fact, even now it was showing "on battery", until I restarted device.

Lets check what will happen tomorrow morning.

schlimmchen, 2024-08-22:
This reads like "nighttime discharging is not reset as expected once a new day begins", is that correct? If this happens again, please don't hesistate to open a new issue. Also, if possible, add DPL logs (verbose logging enabled) from the web console while the issue is occuring (while the kettle is on).

schlimmchen, 2024-12-02:
Im August habe ich geschrieben:

Die Abbruchbedingung für "nighttime discharging" sollten sein:
* Stop-Schwelle erreicht **oder**
* *genügend* solare Leistung vorhanden, um den Haushaltsbedarf zu decken.

Macht eigentlich weiterhin Sinn, allerdings funktioniert das dann nur für Nutzer mit VE.Direct Laderegler (oder von mir aus auch zukünftig für solche mit Laderegler-Daten über MQTT). Was machen wir mit Nutzern, die ihre solare Leistung nicht kennen?

Nutzer ohne Schnittstelle zum Laderegler aber mit Schnittstelle zur Batterie könnten stattdessen die Abbruchbedingung "Wenn das BMS bestätigt, dass die Batterie geladen wird" (braucht es dafür eine konfigurierbare Schwelle?).

Das wird dann schnell kompliziert. Kann man zwar in einer Funktion kapseln, aber trotzdem...

Ich schlage drei neue Optionen vor:
1. Dokumentiere das Verhalten und gut is. Ernsthaft beschwert hat sich keiner.
2. Entferne das Feature. Erziehe die Nutzer dahingehend, dass dieses Feature keinen Sinn macht, und dass es immer effizienter ist, den Strom sofort zu verbrauchen, statt ihn bis abends aufzuheben. Das trifft aber wiederum für Nutzer mit Solar-Passthrough gar nicht zu, weil eh nur überschüssige Energie gespeichert wird.
3. Stoppe das nächtliche Entladen nicht zum Zeitpunkt des Sonnenaufgangs, sondern ein paar Stunden später. Ich denke dabei an drei oder vier Stunden (nicht konfigurierbar, aber dokumentiert).

Option 3 macht das Feature, das ich mit Option 2 diskreditiert habe, nur noch komplizierter zu warten, als es eh schon ist. Option 2 ist ne Regression und Nutzer werden sich beschweren... Daher tendiere ich zu Option 1.

Option 1 habe ich jetzt pauschal umgesetzt.
[image]

Parallel dazu lese ich gerne Eure Meinungen. Hast du eine Meinung, @AndreasBoehm? Wenn dazu nichts kommt, schließe ich das Issue spätestens das nächste Mal, wenn ich Issues aufräume.

closed 2024-12-04 by schlimmchen