hoylabs/OpenDTU-OnBattery #799: Second MPPT, periodical Update Issue

SW-Niko, 2024-03-24:
### What happened?

On my system, the second MPPT periodically does not update for about 11 seconds. 
The error occurs about once a minute. 
[image]

### To Reproduce Bug

Configure second MPPT and watch the live view for 2-3 minutes

### Expected Behavior

Update every 2 seconds, More or less identical  to first MPPT

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

gf147d83

### Anything else?

However, my statistical logging shows that the data from the two MPPT arrive after 2.5 seconds at the latest. That's why I assume the problem must be in the higher levels.

14:50:47.348 > TST: Data valid, Duration: 0 days, 1 hour, 10 min, 59 sec
14:50:47.400 > --------------------------------------------------------------
14:50:47.513 > TST:                    MPPT1:    OK-Time: Max:0ms, Min=0ms, Avg=0ms, Ok=0, Value=0.00
14:50:47.718 > TST:                         : NotOk-Time: Max=0ms, Min=0ms, Avg=0ms, Fault=0/0.00%, Value=0.00
14:50:47.922 > TST:                         :  Back-Time: **Max=2453ms,** Min=1ms, Avg=605ms
14:50:48.126 > TST:                    MPPT2:    OK-Time: Max:0ms, Min=0ms, Avg=0ms, Ok=0, Value=0.00
14:50:48.331 > TST:                         : NotOk-Time: Max=0ms, Min=0ms, Avg=0ms, Fault=0/0.00%, Value=0.00
14:50:48.536 > TST:                         :  Back-Time: **Max=2663ms**, Min=1ms, Avg=713ms

schlimmchen, 2024-03-24:
Mh, weird... I certainly did not see this while I was tweaking and testing the feature.

Please tell us more about how your statistics work, i.e., how do you determine the highlighted max value and what does "Back-Time" mean? Are you suggesting that there is valid data decoded by the VictronMpptController class at least every ~2.5s? If so, how do you determine this?

You should know that the websocket is only supplied with the MPPT data if it is valid and if it has newer data than the last time the websocket was supplied with data from that MPPT. After 10s, however, all data is always sent (unless the MPPT controller says it is invalid). So there is a chance that you are only getting updates for the second MPPT after ten seconds because something is wrong with the update timestamp in the MPPT controller class.

You are writing that you "periodically" see this behavior. What does that mean? Is the data of the second MPPT **always** only updated after 10 or 11 seconds? Or does it sometimes get updated in a shorter amount of time?

schlimmchen, 2024-03-24:
Did you have one of the two hooked up to your OpenDTU-OnBattery before the update and how did that behave?

And does the self-compiled firmware you use contain changes, like your VE.Direct HEX support? Commit f147d83 you mentioned sure smells like it. Can you please double-check that the problem still persists if you use the released firmware?

SW-Niko, 2024-03-24:
> Please tell us more about how your statistics work, i.e., how do you determine the highlighted max value and what does "Back-Time" mean? Are you suggesting that there is valid data decoded by the VictronMpptController class at least every ~2.5s? If so, how do you determine this?

I'm logging function "VeDirectMpptController::frameValidEvent()"
The Back-Time is the time between 2 calls of the function. And the Back-Time Max is the maximum logged time during 1 hour and 10 Minutes.
Inside of function VeDirectMpptController::frameValidEvent()" we set _lastUpdate = millis();
So .. I think we set _lastupdate in worst case every 2.5 sec. The average is about  700ms.

> You are writing that you "periodically" see this behavior. What does that mean? Is the data of the second MPPT **always** only updated after 10 or 11 seconds? Or does it sometimes get updated in a shorter amount of time?

Most of the time it takes 1sec. After 1 minute it is growing 4 sec, next 6 sec, next 9 sec, next 11 sec, next 11 sec, next 1 sec.

> Did you have one of the two hooked up to your OpenDTU-OnBattery before the update and how did that behave?

Yes, I did some hardware cross checks, MPPTs, cable. I'm 100% sure it is not a hw fault. It is always the 2nd MPPT.

> And does the self-compiled firmware you use contain changes, like your VE.Direct HEX support? Commit [...]

I started with released fw and switched to self compiled to do some checks.

schlimmchen, 2024-03-24:
Do you see any (newly added) "DynamicJsonDocument overflowed" messages on the console/serial output?

SW-Niko, 2024-03-24:
> Do you see any (newly added) "DynamicJsonDocument overflowed" messages on the console/serial output?

No

It is not critical. System is still running. I would recommend not to waste your time. 
Let us wait if someone else recognize the same issue and in the meantime I will try to find the root cause.

meisterlampe, 2024-03-25:
I can confirm the problem. On my system it is also periodically increasing.

Also it seems, that the DPL is only setting the power limit to about half of the value of what is needed to achieve the target consumption (even if only one victron is configured, going back to previous version solves the problem. But that is maybe a separate issue.. need to investigate first)

spcqike, 2024-03-25:
> Also it seems, that the DPL is only setting the power limit to about half of the value of what is needed to achieve the target consumption

that’s odd. What inverter do you have and how is it connected? Do you use all available DC inputs?

SW-Niko, 2024-03-25:
Sorry I have to switch to german..

Hallo @schlimmchen ,
Ich habe den bug noch nicht gefixt aber ich kann dir sagen wie du ihn reproduzieren kannst.

Ersetze in der Funktion VictronMpptClass::getDataAgeMillis(size_t idx)
die Zeile 
return millis() - _controllers[idx]->getLastUpdate();
mit
return millis() - _controllers[**0**]->getLastUpdate();

dann solltest du bei deinem system ebenfalls das Problem reproduzieren können. Mit dem einem Unterschied dass der Fehler jetzt permanent auftaucht.

**Kurze Beschreibung:**
Die Web Anzeige hat ein Problem wenn beide MPPT Daten Zeitstempel gleich sind.

**Lange Beschreibung:**
In der Tontechnik spricht man von "Schwebung". Eine Schwebung tritt dann auf wenn 2 Quellen mit fast gleicher Frequenz aufeinander treffen. Die 2 Quellen sind die 2 MPPT und die Frequenz macht der Text Frame. Und jetzt passiert folgendes...
Wenn die Frequenzen (Frame) leicht unterschiedlich sind dann kommt es periodisch vor das beide zur gleichen Zeit einen Frame senden und mit dem gleichen Zeitstempel versehen werden. Und wenn der Zeitstempel gleich ist dann hat die Web-Ausgabe ein Problem damit. Mit der Code Änderung erzeuge ich praktisch eine "scheinbare Synchronisierung".
Das erklärt auch warum nur einige das Problem feststellen. Ist die Frequenz etwas **mehr** unterschiedlicher als bei mir, dann taucht der Fehler nicht auf.  Ist die Frequenz etwas **weniger** unterschiedlicher als bei mir dann taucht der Fehler vielleicht nur einmal in der Stunde oder am Tag auf. 

Ich mach da heute Abend weiter. Momentan hab ich aber mit diesem Teil der FW noch keine Erfahrung. Es kann noch etwas dauern bis ich das vernünftig fixen kann.

LukasVFL99, 2024-03-25:
@SW-Niko hey, wie sieht dein Pin mapping aus für den zweiten Victron? Würde meinen zweiten auch gerne anschließen. Danke :)

schlimmchen, 2024-03-25:
@LukasVFL99 Siehe https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Device-Profiles (habs gerade nachgezogen).

> Die Web Anzeige hat ein Problem wenn beide MPPT Daten Zeitstempel gleich sind.

Wundert mich, aber das ist durchaus möglich. Kann sowohl in der WebApp Probleme verursachen, als auch in der Websocket-Implementierung in der Firmware.

Und es passt auch zu meinen Beobachtungen, denn auf dem Schreibtisch hab ich zur Zeit einen echten MPPT und einen simulierten, und letzterer hat definitiv eine andere Sendefrequenz.

Danke, @SW-Niko dass du so dran bist :muscle: 

@meisterlampe Please open a new issue and provide more details (setup, DPL verbose logging, etc.).

SW-Niko, 2024-03-25:
Hallo @schlimmchen,
Wenn meine Annahme stimmt, dann must du mit der  "scheinbaren Synchronisierung".
return millis() - _controllers[**0**]->getLastUpdate();
auch den Fehler reproduzieren können.  Ganz egal ob du einen echten MPPT oder einen simulierten MPPT verwendest.
Vielleicht kannst du das bei Gelegenheit mal testen. Nicht das ich auf dem Holzweg bin. Danke.

SW-Niko, 2024-03-25:
Hallo @schlimmchen ,
ich habe den "root cause" gefunden.
in der Funktion WebApiWsVedirectLiveClass::generateJsonResponse() wird _lastPublish = millis() beim Update für den 1ten MPPT gesetzt und die Funktion WebApiWsVedirectLiveClass::hasUpdate() liefert beim 2ten Schleifendurchlauf false zurück, wenn der Zeitindex des 2ten MPPT gleich oder älter als der des 1ten ist.

Ich habe mal folgende Code Änderung gemacht:
Zeile 27: uint32_t _lastPublish[2] = {};
Zeile 50: auto publishAgeMillis = millis() - _lastPublish[idx];
Zeile 141: _lastPublish[idx] = millis();
Also für jeden MPPT eine eigene Variable eingeführt. Und jetzt ist der bug bei mir weg.

Das kann man so aber nicht lassen. Die MPPT Klasse und weitere Funktionen sind ja schon für weitere MPPT. vorbereitet.
Stichwort dynamisches Array / Vector. Da bin ich noch nicht C++ sattelfest. 
Ich denke aber damit hast du genügend Informationen, um das Problem zukunftssicher zu fixen.

schlimmchen, 2024-03-26:
> die Funktion WebApiWsVedirectLiveClass::hasUpdate() liefert beim 2ten Schleifendurchlauf false zurück

Du hast absolut recht, danke fürs Debuggen!

Es braucht aber keine separaten `_lastPublish` Werte. Meine Idee war: "Wenn ein Controller Daten hat, die jünger sind als das letzte mal, als ich **alle** Controller abgeklappert habe, dann schicke ich die Daten des Controllers in den Websocket." Diese Logik passt auch weiterhin, aber `_lastPublish` muss natürlich *nach* der Schleife aktualisiert werden.

Teste doch mal die Anpassung in 91a09929 bzw. rebase auf diesem commit bzw. branch `development`. Ich bin optimistisch, dass das Problem dann verschwindet.

SW-Niko, 2024-03-26:
Jepp, das ist die elegantere Methode. Denke auch das das passen sollte.
Heute komm ich nicht mehr zum testen aber Morgen solle es klappen.

SW-Niko, 2024-03-27:
Hallo @schlimmchen ,
der Bug ist weg.  👍 
Ich hab den Fix aufgespielt und ein paar Minuten Beobachtet.

closed 2024-03-27 by SW-Niko