hoylabs/OpenDTU-OnBattery #809: Wechselrichter wechselt in Standby und verharrt dort

schlimmchen, 2024-03-28:
[image]

Per Hand neu gestartet:
[image]

```
RX Period End
07:48:22.546 > Middle missing
07:48:22.588 > Request retransmit: 3
07:48:22.638 > TX RequestFrame 865.00 MHz --> 15 91 43 32 52 80 10 26 80 83 12 
07:48:22.691 > Interrupt received
07:48:22.737 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 03 00 00 00 12 00 03 00 00 A1 A9 00 01 11 DE 00 1D D8 | -35 dBm
07:48:22.792 > RX Period End
07:48:22.839 > Success
07:48:22.886 > [DPL::updateInverter] Starting inverter...
07:48:22.927 > TX PowerControl 865.00 MHz --> 51 91 43 32 52 80 10 26 80 81 00 00 B0 01 E5 
07:48:22.966 > Interrupt received
07:48:23.006 > RX 865.00 MHz --> D1 91 43 32 52 80 10 26 80 81 00 00 00 00 24 00 F0 | -35 dBm
07:48:24.665 > RX Period End
07:48:24.718 > Success
07:48:24.767 > Fetch inverter: 114483761097
07:48:24.806 > TX RealTimeRunData 865.00 MHz --> 15 83 76 10 97 80 10 26 80 80 0B 00 66 05 12 B8 00 00 00 00 00 00 00 00 D6 CA 0F 
07:48:24.846 > Interrupt received
07:48:24.891 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 01 00 01 01 3C 01 3F 03 1A 03 09 09 CE 09 B5 00 01 BB | -46 dBm
07:48:24.937 > Interrupt received
07:48:24.974 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 02 01 26 00 01 2F BD 00 AD 00 94 08 FE 13 88 12 7B 5A | -46 dBm
07:48:25.025 > Interrupt received
07:48:25.066 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 83 00 04 00 CD 03 E7 00 DA 00 01 5D 7A 83 | -46 dBm
07:48:25.242 > RX Period End
07:48:25.276 > Success
07:48:25.673 > Fetch inverter: 116491433252
07:48:25.727 > TX RealTimeRunData 865.00 MHz --> 15 91 43 32 52 80 10 26 80 80 0B 00 66 05 12 B9 00 00 00 00 00 00 00 00 46 C7 53 
07:48:25.779 > Interrupt received
07:48:25.824 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 01 00 01 01 6E 01 6D 00 03 00 01 00 0E 00 04 00 00 1A | -35 dBm
07:48:25.868 > Interrupt received
07:48:25.917 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 02 D4 D7 00 00 A2 C5 00 20 00 1E 01 6B 01 67 00 05 40 | -35 dBm
07:48:25.959 > Interrupt received
07:48:26.009 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 03 00 00 00 12 00 03 00 00 A1 A9 00 01 11 DE 00 1D D8 | -35 dBm
07:48:26.069 > Interrupt received
07:48:26.118 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 04 00 18 09 01 13 88 00 00 00 00 00 00 00 00 00 73 ED | -35 dBm
07:48:26.157 > Interrupt received
07:48:26.199 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 85 00 01 B2 33 14 | -35 dBm
07:48:26.249 > RX Period End
07:48:26.297 > Success
07:48:26.334 > [DPL::updateInverter] Starting inverter...
07:48:26.378 > TX PowerControl 865.00 MHz --> 51 91 43 32 52 80 10 26 80 81 00 00 B0 01 E5 
07:48:26.427 > [   4705.573] DPL: waiting for a start/stop/restart command to complete
07:48:26.478 > Interrupt received
07:48:26.525 > RX 865.00 MHz --> D1 91 43 32 52 80 10 26 80 81 00 00 00 00 24 00 F0 | -35 dBm
07:48:28.127 > PowerMeterClass: TotalPower: 1601.66
07:48:28.300 > RX Period End
07:48:28.345 > Success
07:48:28.393 > Fetch inverter: 114483761097
07:48:28.444 > TX RealTimeRunData 865.00 MHz --> 15 83 76 10 97 80 10 26 80 80 0B 00 66 05 12 BC 00 00 00 00 00 00 00 00 16 F8 F9 
07:48:28.489 > Interrupt received
07:48:28.536 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 01 00 01 01 3C 01 3F 03 1C 03 0A 09 D3 09 BA 00 01 AC | -46 dBm
07:48:28.578 > Interrupt received
07:48:28.618 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 02 01 26 00 01 2F BE 00 AD 00 95 09 01 13 89 12 84 58 | -46 dBm
07:48:28.664 > Interrupt received
07:48:28.710 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 83 00 04 00 CD 03 E7 00 DA 00 01 BC 1D 05 | -45 dBm
07:48:28.868 > RX Period End
07:48:28.913 > Success
07:48:29.249 > [DPL::updateInverter] timeout, state transition pending: yes, limit pending: no
07:48:29.288 > [   4708.515] DPL: waiting for sufficiently recent inverter data
07:48:29.334 > Fetch inverter: 116491433252
07:48:29.394 > TX RealTimeRunData 865.00 MHz --> 15 91 43 32 52 80 10 26 80 80 0B 00 66 05 12 BD 00 00 00 00 00 00 00 00 86 F5 A5 
07:48:29.501 > Interrupt received
07:48:29.545 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 01 00 01 01 6E 01 6D 00 03 00 01 00 0E 00 03 00 00 1D | -35 dBm
07:48:29.587 > Interrupt received
07:48:29.636 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 02 D4 D7 00 00 A2 C5 00 20 00 1E 01 6B 01 67 00 05 40 | -35 dBm
[... 747 lines cut ...]
07:49:35.873 > Interrupt received
07:49:35.923 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 02 01 2B 00 01 2F C2 00 B2 00 99 09 03 13 88 12 AD 10 | -45 dBm
07:49:35.950 > Interrupt received
07:49:35.991 > RX 865.00 MHz --> 95 83 76 10 97 80 10 26 80 83 00 04 00 CF 03 E7 00 DD 00 03 73 08 D8 | -45 dBm
07:49:36.258 > RX Period End
07:49:36.305 > Success
07:49:36.531 > [DPL::loop] ******************* ENTER **********************
07:49:36.659 > [DPL::loop] PowerMeter: 1465 W, target consumption: -400 W, solar power: 10000 W
07:49:36.660 > [DPL::loop] Consuming Solar Power Only -> solarPowerAC: 9379 W, newPowerLimit: 1985 W
07:49:36.663 > [DPL::setNewPowerLimit] calculated: 1985 W, requesting: 1700 W, reported: 1700 W, diff: 0 W, hysteresis: 25 W
07:49:36.702 > Fetch inverter: 116491433252
07:49:36.742 > TX RealTimeRunData 865.00 MHz --> 15 91 43 32 52 80 10 26 80 80 0B 00 66 05 13 00 00 00 00 00 00 00 00 00 14 C9 B7 
07:49:36.790 > Interrupt received
07:49:36.836 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 01 00 01 01 4A 01 48 00 73 00 66 01 7D 01 51 00 00 2A | -35 dBm
07:49:36.882 > Interrupt received
07:49:36.918 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 02 D4 D7 00 00 A2 C4 00 00 00 00 01 47 01 37 00 5E 58 | -35 dBm
07:49:36.952 > Interrupt received
07:49:37.007 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 03 00 4D 01 36 00 F2 00 00 A1 A9 00 01 11 DD 00 00 5F | -35 dBm
07:49:37.040 > Interrupt received
07:49:37.082 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 04 00 00 09 07 13 88 04 B8 00 01 00 34 03 E7 00 73 9E | -35 dBm
07:49:37.131 > Interrupt received
07:49:37.163 > RX 865.00 MHz --> 95 91 43 32 52 80 10 26 80 85 00 01 0F 02 98 | -35 dBm
```

_Originally posted by @chrisli01 in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/755#issuecomment-2024519562_

schlimmchen, 2024-03-28:
@chrisli01 Auf welchen Wert hast du das untere Leistungslimit eingestellt?

Dein Inverter wird zwar vom DPL angewiesen zu starten, er verweigert das aber offensichtlich.

Dein Problem wird mit #748 voraussichtlich umschifft werden. Es wäre aber gut zu erfahren, was die Ursache dafür ist, dass der Inverter sich nicht wecken lässt.

chrisli01, 2024-03-28:
[image]
[image]

Soll ich mal 40 einstellen?

Solarteur, 2024-03-28:
Ist sicher einen Versuch wert. Bei mir läuft ein HM-600 mit 50W (ca. 8% der Gesamtleistung) unterem Leistungslimit total stabil.

chrisli01, 2024-03-28:
Dankeschön!

habs mal auf 40 gestellt….

spcqike, 2024-03-28:
Generell sind 10W pro MPPT wahrscheinlich nicht verkehrt. So wie es sich in letzter Zeit abbildet mir nicht richtig reagierenden wechselrichtern.

chrisli01, 2024-04-02:
Hatte wieder nen gelben WR… bin mal auf 50W gegangen.

chrisli01, 2024-04-05:
Mit 50W siehts gut aus🤗🤗

schlimmchen, 2024-04-05:
Super, vielen Dank für die Rückmeldung.

Da müssen wir uns etwas überlegen. Entweder eine fette Warnung im Log falls das untere Leistungslimit "eher klein" ist, oder pauschal verweigern weniger als... (keine Ahnung... welche minimale Leistung ist denn okay?) als Limit einzustellen. Vielleicht sollte das als Hinweis in die noch zu schaffende DPL Karte im Live-View? Puh.

Jedenfalls ist es auch nicht schön, wenn wir mit #748 "ständig" den Inverter neu starten und er sich dann wiederholt selbst abschaltet weil das kleine Limit einfach zu wenig ist und er "schwingt".

spcqike, 2024-04-05:
ich würd es wohl dem User frei lassen, aber bei Werten < 10W pro MPPT (oder 12.5W, wie hier bei in Summe 50W am HMS-2000) eine Warnung zeigen, dass das instabil sein kann.

Kann, einfach weil die Bauteile im Wechselrichter sicherlich auch Toleranzen und unterschiedliche EIgenschaften haben. 30W kann bei einem User laufen, beim zweiten nicht. Bzw. gern auch in Abhängigkeit der Temperatur.

Ist ein bisschen wie Overclocking, wenn man das "Maximale" (hier das Minimale) rausholen will, muss man es einfach probieren, was mit seiner Hardware geht. Zu hoch fest vorgeben würde ich es wohl nicht.

PhilJaro, 2024-04-07:
Ich habe leider das Problem wenn ich das Minimum auf 10W pro Eingang (HM1500 4*10W) setzte, liegt dies zu nah an meinem Grundverbrauch in der Nacht ca. 50W und der Wechselrichter schaltet sich gut alle 20-30 Minuten aus… selbst wenn ich den Wert auf 10 setze, schaltet er sich alle 2-3 Stunden aus. 

Schön wäre hier wenn das Ausschalten nicht sofort sondern erst nach gewisser Zeit unter Minimum, oder gar nicht passieren würde.

schlimmchen, 2024-04-07:
> Schön wäre hier wenn das Ausschalten nicht sofort sondern erst nach gewisser Zeit unter Minimum

Gibt es da schon ein Issue mit Request? Ich glaube nein. Es gab eins für Systeme ohne Batterie, aber mit Batterie war das bisher nur irgendwo anders diskutiert. Ich hatte dazu vermerkt, dass das Ausschalten bei Batteriesystemen verzögert sein sollte, weil sonst beim Schwanken um das untere Leistungslimit in der Dämmerung, der Inverter täglich häufig aus- und eingeschaltet wird. Eine Verzögerung fänd ich auch gut.

spcqike, 2024-04-07:
@schlimmchen 
Es gibt glaube keins für „schalte erst nach X Minuten ab“ aber eins für „schalte gar nicht ab, produziere stattdessen das untere Limit“

https://github.com/helgeerbe/OpenDTU-OnBattery/issues/766

so wie du es vor kurzem für uns PV only User eingeführt hast (:

schlimmchen, 2024-04-07:
Das ist, was ich nicht möchte. Daher muss das konfigurierbar sein. Was ich natürlich auch nicht möchte, das weißt du, aber ist dann so.

Das Verzögern sehe ich als separate Funktion. Die wäre für alle und per default an und gleich, jedenfalls bis ich jemand beschwert...

PhilJaro, 2024-04-07:
> Das ist, was ich nicht möchte. Daher muss das konfigurierbar sein. Was ich natürlich auch nicht möchte, das weißt du, aber ist dann so.

Vielleicht blöde Frage, aber warum? 

Der WR schaltet ja auch nicht ab wenn der Netzverbrauch über dem eingestellten Maximum liegt.
Somit hatte ich anfangs auch erwartet, das Minimum verhält sich wie das Maximum und der WR begrenzt nur auf den Wert falls der Netzverbrauch darunter fallen sollte, aber schaltet nicht ab.

schlimmchen, 2024-04-07:
Wenn ich in die Situation komme, dass ich weniger Strom erzeugen muss, als das untere Leistungslimit angibt, um meinen Bedarf zu decken, dann möchte ich doch nicht mit niedriger Effizienz meinen Akku umsonst ins Netz entleeren?

Vielleicht habe ich aber echt einen Denkfehler, weil ich zu viel auf den Code geschaut habe.

In die Situation kommt man ja "eh nicht". Der Normalfall ist ja jener, dass man bei hinreichend geladener Batterie seinen Bedarf so lange deckt (Ausnahme: der Inverter kann weniger erzeugen als benötigt wird), bis die Batterie leer ist. Dann wird der Inverter aus gutem Grund abgeschaltet (da sind wir uns vermutlich alle einig).

Wenn ich also weggehe von der Frage "Was soll denn das untere Limit sein, wenn der Wechselrichter wegen leerer Batterie abgeschaltet werden muss", und einsehe, dass es die Situation "Es wird weniger Leistung benötigt, als das untere Leistungslimit angibt" bei vernünftiger DPL-Konfig gar nicht gibt, dann macht der Wunsch in #766 (und der wurde ja schon häufig geäußert) plötzlich total Sinn.

Vermutlich weil mir jetzt klar ist, dass nichts verschenkt wird.

Hm... :monocle_face:

spcqike, 2024-04-07:
> macht […] total Sinn

macht es meiner Meinung nach auch :)

mit dieser Funktion, also das nicht abschalten sondern eben das „Mindesterzeugen“ wäre wohl auch hier geholfen 

https://github.com/helgeerbe/OpenDTU-OnBattery/issues/850#issuecomment-2041159102

meine Frage, wie die Batterie dann geschützt wird, wenn der DPL einfach deaktiviert wird, hast du da wohl überlesen (oder doch ignoriert? ;) ) 

mit einem anders interpretierten unteren Limit, könnte man eben auch den DPL nutzen, auch wenn das powermeter ausfällt oder man gar keins hat, und eben mindestens (bzw. ohne power Meter halt genau) X Watt erzeugen. Und trotzdem auf solar passthorugh und die Abschaltung des wechselrichters vertrauen. 

Ich sehe nur Vorteile. (Aber bei mir läge das untere Limit eben auch bei 200W, meiner Grundlast. Weit außerhalb der Effizienz Probleme. Und die mindestens 200W will ich eben auch immer erzeugen, sofern die Batterie es zulässt)

PhilJaro, 2024-04-07:
> Wenn ich in die Situation komme, dass ich weniger Strom erzeugen muss, als das untere Leistungslimit angibt, um meinen Bedarf zu decken, dann möchte ich doch nicht mit niedriger Effizienz meinen Akku umsonst ins Netz entleeren?

Das Argument kann ich nachvollziehen. 

In meinem Fall wäre es allerdings „effizienter“, falls man den Begriff dafür hernehmen kann, für ein paar Sekunden oder auch Minuten 5-10 Watt ins Netz einzuspeisen als den WR auszuschalten und währenddessen 30 Watt oder mehr zu beziehen bis der WR wieder „hochgefahren“ ist.

Eine Hysterese bis zur Abschaltung würde natürlich auch dazu führen das dieses Verhalten vermutlich seltener bis gar nicht mehr auftreten würde, nur ist dann die Frage wie komplex man das ganze Thema machen möchte bzgl. Konfiguration…

Für mein Verständnis wäre es am sinnvollsten das Minimum wirklich als Minimum zu nutzen und nicht als Abschaltschwelle. Kann aber natürlich auch sein das ich hiermit nicht für die Mehrheit spreche und das „nur“ meine Sicht der Dinge ist…

> Dann wird der Inverter aus gutem Grund abgeschaltet (da sind wir uns vermutlich alle einig).

Das denke ich doch auch ;)

schlimmchen, 2024-04-07:
> oder doch ignoriert?

Klar hab ich das, und das weißt du. Und ich hab darauf vertraut, dass du es mir nicht übel nimmst. Ich wollte das Issue loswerden :stuck_out_tongue_closed_eyes: Mit diesem Kommentar hat du es beinahe verhindert :grin: 

> wenn das powermeter ausfällt oder man gar keins hat

Man hat keinen Power Meter, aber will den DPL nutzen? Was tut der denn dann?

> Und trotzdem auf solar passthorugh und die Abschaltung des wechselrichters vertrauen.

Also solar-passthrough macht keinen Sinn. Erklär mir, was der DPL Sinnvolles macht, wenn er zwar die solare Leistung kennt, aber keinen power meter hat. Dann macht er unconditional full solar-passthrough? Warum? Bei Systemen ohne Batterie aber mit Victron Laderegler?!

Abschaltung des WR wenn die Batterie leer ist, und sonst nur die "Grundlast" einspeisen. Jo, das kann ich nicht wegdiskutieren. Würde dieses Projekt anwendbar machen für mehr Nutzer. Ich hab dieses Setup bisher nicht wirklich ernst nehmen wollen. Aber es gibt eben auch sicherlich Bastler in Mietwohnungen oder Miethäusern, die einfach keinen Power Meter einbinden können, sehr wohl aber einen kleinen Akku nutzen könnten um wenigstens die Grundlast abzudecken.

Weiter in #766, da kommt gleich ein länglicher Kommentar von mir.

# Fokus

In diesem Issue geht es darum, dass wir den Nutzern erläutern müssen, dass ein zu geringes unteres Limit zu Problemen führen kann (Schwingen).

der-peterh, 2024-04-21:
Ich konnte das Problem mit dem erfolglosen Startversuch nun mehrfach in Kombination mit >= 15 Einträgen in der Ereignisanzeige beobachten und dann durch Neustarten des WR 'heilen'. -> dann nur noch ein Eintrag in der Ereignisanzeige.

Purer Zufall, dass ich dies so beobachten konnte, weil ich gerade durch eine Testinstallation eines ungesteuerten Balkonkraftwerkes und stark wechselnder Bewölkung alle paar Minuten in die Abschaltung des HM-600 meiner OpenDTU-onBattery komme....

schlimmchen, 2024-04-21:
Häufiges Ausschalten steht allerdings bisher nicht im Verdacht auszulösen, dass der WR sich dann nicht mehr wecken lässt. Bisher haben wir nur ein zu kleines Limit im Verdacht. Mein WR wird in der Dämmerung auch häufig aus- und eingeschaltet, mein Ereignis-Log ist häufig auch voll, aber das Verklemmen kann ich nur mit einem zu kleinen Limit erzwingen.

Mit #901 wird der Wechselrichter automatisch neu gestartet werden, wenn er sich nicht aufwecken lässt durch Start-Kommandos. Bist du ein guter Kandidat, diese Änderung zu testen? Falls ja, freue ich mich, wenn du die [entsprechende Firmware aus dem PR-Build-Run](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8773186932?pr=901) installierst und Rückmeldung gibst.

der-peterh, 2024-04-23:
Und heute habe ich dann auch schon den ersten eigenständigen Neustart des HM-600 durch OpenDTU-onBattery beobachtet, wobei OpenDTU-onBattery selbst durchlief.
Auf logging hatte ich verzichtet, weil ich eh nicht wüsste, wonach ich suchen soll. Mir hat als Indikator für den Neustart genügt, dass der Tagesertrag wieder von Anfang an gezählt hat und ein Start um 18:05 als erster Eintrag in der Ereignisanzeige stand.
Ich bleibe also mal bei meinem lokalen Merge und werde nach dem nächsten Release vielleicht versuchen, meine Änderung zum invertieren des PowerMeter Messwertes zu promoten 😉 Vielleicht braucht den ja mal jemand, der die Strom Mess Klemmen ebenfalls verkehrt herum installiert hat 🫣

spcqike, 2024-04-23:
Hilft dir das hier? 
https://github.com/helgeerbe/OpenDTU-OnBattery/pull/887

schlimmchen, 2024-04-23:
> Vielleicht braucht den ja mal jemand, der die Strom Mess Klemmen ebenfalls verkehrt herum installiert hat 🫣

Aus #887:

> Vor kurzem habe ich von jemanden gelesen, der eine angepasste Firmware baut, weil er wohl seine Strommessklemmen falsch herum angeklemmt hat. Leider finde ich den gerade nicht mehr...

Ach du warst das? :wink:

der-peterh, 2024-04-23:
Prima 👍Probiere ich mit meinem Shelly 3EM direkt mal als Alternative zu MQTT aus. Meinen 'Invertiere Messwert' Toggle habe ich deutlich prominenter direkt unter 'Aktiviere Stromzähler' positioniert 🫣

schlimmchen, 2024-04-23:
Hm, ne, lass das lieber. HTTP+JSON ist leider deutlich weniger robust als MQTT.

Magst du das Feature für MQTT anpassen und einen PR erstellen? Ich dachte man müsste das auch für MQTT anbieten, aber da das Code Duplication oder viel Aufwand bedeutete, hatte ich keine Lust drauf 🙊

schlimmchen, 2024-12-02:
> In diesem Issue geht es darum, dass wir den Nutzern erläutern müssen, dass ein zu geringes unteres Limit zu Problemen führen kann (Schwingen).

Erledigt in 2c71dba6.

[image]

closed 2024-12-02 by schlimmchen