hoylabs/OpenDTU-OnBattery #824: Ermittlung der produzierenden Kanäle mittels variabler Mindestleistungsgrenze (keine feste Schwelle von 2W)

RoFranz, 2024-03-31:
### Is your feature request related to a problem? Please describe.

Ich betreibe zwei Akkus (PB2500) jeweils an einem HMS-1600T. Die integrierte Steuerung schaltet unter bestimmten Bedingungen (z.Bsp.: bei einer Eingangsleistung der Eingänge über 500W von möglichen 800W) in den sogenannten Passthrough-Modus (direktes Durchschalten des zweiten Einganges). Jedoch wird beim WR nicht die komplette Leistung nur an einem Eingang angezeigt, sondern an beiden mit falscher Summe.
Im Bild sieht man dies für 2 Akkus mit jeweiligen Passthrough-Modus 
[image]

### Describe the solution you'd like

Variable Mindestleistungsgrenze (Min+Max)/2 wie hier beschrieben 
https://github.com/helgeerbe/OpenDTU-OnBattery/issues/697#issuecomment-1964350827
oder über die Spannungswerte abfragen:
Mindestspannung aus den DPL -> Batterienutzung Start-Schwellwert 

### Describe alternatives you've considered

händisches Setzen der Schwelle auf 50 W -> verlangt aber Mindestleistung des Netzes von 200W (falls nicht gegeben, schaltet DPL den WR aus).

schlimmchen, 2024-03-31:
> oder über die Spannungswerte abfragen: Mindestspannung aus den DPL -> Batterienutzung Start-Schwellwert

Hiermit wäre ich einverstanden (auch wenn ich diese Skalierung nicht leiden kann). Zusätzlich auch noch gegen 16V testen, weil alle Hoymiles Wechselrichter (sowie ich weiß) nicht unter 16V arbeiten.

> Variable Mindestleistungsgrenze (Min+Max)/2 wie hier beschrieben

Das ist auch gut.

Sollte man alles zusammennehmen, denke ich. Ein Kanal produziert nicht, wenn

* Die Spannung bei unter 16V liegt
* Die Spannung unter der Batterie-**Stop**-Schwellspannung liegt (nur bei Batterie-Systemen)
* Die Leistung bei unter 2W liegt
* Die Leistung unter (min+max)/2 der Leistungen aller Kanäle liegt

Keine Skalierung, wenn der Inverter aus ist (das sollte allerdings schon implementiert sein).

Macht das Sinn?

spcqike, 2024-03-31:
Ich denke auch, und hab das bereits anderswo einmal angesprochen, dass die eingangsspannung des Kanals mit berücksichtigt werden sollte. Woher auch immer, misst der HM bei dir jeweils 9V. Vielleicht legt die Batterie wirklich 9V am Ausgang an, wenn der Ausgang deaktiviert ist. Keine Ahnung. 

Eine Abfrage wie (Min+Max)/2 finde ich falsch. Bzw dann müsste wieder differenziert werden zwischen reinen solar Systemen und Batterie Systemen. Bei reinen solar Systemen kann es ja durchaus passieren, dass ein Eingang durch verschattung wesentlich weniger liefert. Das sollte dort mMn. Anders abgefangen und bewertet werden. 

> Jedoch wird beim WR nicht die komplette Leistung nur an einem Eingang angezeigt, sondern an beiden mit falscher Summe.

wie meinst du das? Wir sehen in deinem Bild 37W und 192W. Was sagt denn deine App in dem Fall? Hatte der WR ein gesetztes Limit? An den Eingängen hat er 9V und 45V. 

ich würd gern nachvollziehen wollen, was der WR da wie anzeigt. Ich vermute, dass einfach ein paar A über die geteilte Masse auf Eingang 1 zurück fließen. Durch die 9V am Eingang und den fließenden Strom kommt er auf 37W. Das wären also knappe 4A. Gleiches bei Eingang 2 (4.2A für 45V und 192W)
Daher die Frage: lieferte die Batterie 375W? Also 8.3A bei 45V? Und nicht nur 230W? Sollte das in etwa so hinkommen, müsste man es nämlich auch bei der Skalierung entsprechend berücksichtigen, oder stimmt am Ende die AC Leistung wieder in etwa? Die berechnete Effizienz wäre mit 80% quasi unterirdisch. (Also mit den Daten aus dem Bild)

RoFranz, 2024-03-31:
Abend, 
vielen Dank für die schnelle Antworten und Lösungsvorschläge.
Ich wäre an sich für "keep it simple", weshalb ich eine Lösung unabhängig der Eingangsseite (also ob Akku oder Solarmodul) für alle Benutzer wahrscheinlich am einfachsten wäre. Damit wären wir wahrscheinlich bei einer "Mindestspannung", welche über ein Eingabefeld im DPL-Menü eingestellt werden kann.
@spcqike: Die zusammengelegten Massen sind da wahrscheinlich wirklich ein Problem. Hinsichtlich der wahren Werte von Ausgangswerte Akku (aus der App bzw. aus Homeassistant) und der Eingangswerte HMS1600T ergibt sich folgendes Bild (kleiner zeitlicher Verzug für die Screenshots). Werden beide Ausgänge aktiviert, stimmen die Werte von AkkuApp und DTU überein.
DTU:
[image]
Akku1:
[image]
Akku2:
[image]

spcqike, 2024-03-31:
So ganz zusammen bekomme ich die Leistungen in meinem Kopf noch nicht. Sieht jemand anders, was genau da passiert? Wie sich das ganze abbildet? 
Laut Akkus werden 487W entnommen. Laut hoymiles kommen 516W an, woraus dann nur 354W AC werden. Mit meiner Vermutung von oben (beide Ströme aus Ausgang 1 und 2 multipliziert mit Spannung von Eingang 2) komme ich am Ende wieder auf knapp 375W pro Batterie, was ebenfalls falsch ist. 

Kannst du ggf von den hoymiles Daten nicht die Hälfte abschneiden? Und den Test wiederholen? Gerne auch mit gesetzten Leistungslimit am hoymiles. Wie viel Watt zieht erzeugt er wirklich, wenn du bspw. 100W vorgibst? Oder 200W? Wenn du wirklich ~die Hälfte auf der AC Seite raus bekommst, könnte uns die krumme DC Anzeige fast egal sein. Wenn nicht, muss das halt irgendwie analysiert und in der Skalierung berücksichtigt werden.

RoFranz, 2024-04-01:
Hallo @spcqike,

ich habe es bereits aufgegeben mit dem Rechnen, da ich im Moment nicht wirklich den tatsächlichen Stromfluss messen. Auch habe ich mal versucht ein Dashboard mit den wichtigsten Infos zusammenzufassen - falls was fehlt, bitte sagen:
1. Durchlauf: Je Akku OUT2 AKTIV, Verbindung von OUT2 <-> HMS1600 (erwartetes Ergebnis bei Akku und DTU)
[image]
2. Durchlauf: Je Akku OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (ganz wild)
[image]
3. Durchlauf: Je Akku OUT1 AKTIV & OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (unerwartet Akku und erwartet DTU)
[image]

Hoffe das erklärt mehr, als das es Fragezeichen hervorruft.
Frohes Schaffen und danke schon einmal.

RoFranz, 2024-04-01:
Nachbrenner (Dashboard mit Strömen und allen Ausgängen erweitert)
4. Durchlauf: Je Akku OUT1 AKTIV & OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (unerwartet Akku und erwartet DTU)
[image]
5. Durchlauf: Je Akku OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (ganz wild)
[image]

spcqike, 2024-04-01:
Du hast die ganzen Tests bei einem festen Limit von 50% am hoymiles durchgeführt? 

Ich finde die Ergebnisse durchaus schlüssig und aufschlussreich. Die Batterie scheint einen high-side Sensor zu haben. Wenn die Batterie eine Entladung von 378W meldet, würde ich der auch trauen. 

Bekommst du die hoymiles AC Leistung noch irgendwo abgebildet? Bzw nicht nur die vom hoymiles, eventuell auch die eines smarten Zwischensteckers oder anderen leistungsmessers? 

So wie es aussieht, wäre eine einfache Skalierung, wie wir sie bisher haben, in deinem Fall falsch. Also wenn du sagst „eingang1 ist nicht aktiv, daher verdoppel mir bitte das Limit“.

RoFranz, 2024-04-01:
4b Durchlauf (Limit 100%)  Je Akku OUT1 AKTIV & OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (unerwartet Akku und erwartet DTU)
 [image]
 5b Durchlauf (Limit 100%) Je Akku OUT2 AKTIV, Verbindung von OUT1 <-> HMS1600 & OUT2 <-> HMS1600 (ganz wild)
 [image]
Großer Unterschied in den Ausgabewerten ist vor allem der Wirkungsgrad

RoFranz, 2024-04-01:
> So wie es aussieht, wäre eine einfache Skalierung, wie wir sie bisher haben, in deinem Fall falsch. Also wenn du sagst „eingang1 ist nicht aktiv, daher verdoppel mir bitte das Limit“.
Warum ist aus deiner Sicht der Ansatz falsch - was wäre deine Lösungsidee? Doch auch auf Y-Kabel ausweichen?

spcqike, 2024-04-01:
> Warum ist aus deiner Sicht der Ansatz falsch - was wäre deine Lösungsidee? Doch auch auf Y-Kabel ausweichen?

weil der hoymiles so sauber regelt.

In deinem Fall 5b siehst du, was ich meine. 

Der wechselrichter misst seine DC Leistung auf der Masse Seite eines jeden Eingangs. Er denkt, er zieht von der einen Batterie 60W und 198W. Die Batterie hingegen meldet 400W. Was auch realistisch ist. Der Strom fließt von der Batterie zum DC2+. 43V und 8.8A. Nur fließt eben nur die Hälfte des Stroms auch durch DC2- zurück. Die zweite Hälfte fließt durch DC1-, wo dann aber mit falschen 14V ne falsche Leistung berechnet wird. 

würdest du den HM auf bspw 200W limitieren, würde er ja pro Eingang 50W erzeugen können. Durch die falschen Ströme könnte er aber denken, auf DC1 hat er bspw 40W, auf DC2 50, in Wirklichkeit fließen aber, wie hier ja auch, beide Ströme (DC1 und DC2) durch DC2. Also wahrscheinlich weit mehr als 50W. 

Ich glaube auch grade nicht, dass deine AC Leistung stimmt. Ich weiß nicht, ob der hoymiles das AC seitig noch einmal richtig misst oder nur anhand der DC Eingänge berechnet. Aber in deinem Bild 5b, bei in Summe 780W aus beiden Akkus, wären 360W AC seitig echt schlecht. Und der Wirkungsgrad unterirdisch. 

Ja ein Y Kabel sollte helfen. Da so beide Eingänge des wechselrichters normal produzieren können. Ist nur die Frage wie die Batterie damit umgeht. 800W aus einem Ausgang wird wahrscheinlich nicht gehen. Ein Y auf die DC+ Ausgänge, sodass beide DC+ Ausgänge der Batterie auf beide DC+ Eingänge des wechselrichters gehen (das wäre dann ja eher 2 Y bzw 1 X) könnten helfen, das legt allerdings die spannung von DC2+ an DC1+. Ob die Batterie das mag? Keine Ahnung. 

Ich werd die Tage nochmal über die Bilder und Leistungsdaten nachdenken, ich sehe allerdings grad in der Konstellation keine sinnvolle Möglichkeit für eine Skalierung durch den DPL. 

(Der Y Adapter sollte die Skalierungsprobleme beheben. Aber wie gesagt, ich habe keine Ahnung ob die Batterie es mag, wenn die Ausgänge parallel geschaltet werden)

RoFranz, 2024-04-07:
Erstmal generell einfach Danke an eure Unterstützung und die Gedanken, welche ihr euch jeweils macht.

Hinsichtlich der Problemstellung der Ermittlung der produzierenden Kanäle gibt es nun eine anderweitige Lösung. Durch ein Firmwareupdate der Akkus, welches auf Nachfrage dankenswerter Weise bereitgestellt wurde, können nun beide Ausgänge dauernd aktiv sein und es kommt nur zu den Schieflasten - das ist aber eine andere Sache.

Eine Frage hinsichtlich der Leistung über 500W je Kanal (was noch nachgemessen werden müsste). Gibt es die Möglichkeit den DPL auf ein relatives Maximum (z.Bsp.: 99%) zu begrenzen? Falls dazu ein neuer Beitrag aufgemacht werden soll (damit nichts vermischt wird), mache ich dies gerne. Ansonsten schon mal danke für die Antworten...

spcqike, 2024-04-07:
Du kannst ein oberes Leistungslimit im DPL setzen. 

Bspw 800W. Beim HMS-1600 würde damit jeder MPPT auf 200W begrenzen. 

Durch die „schieflast“ allerdings wirst du da keine saubere bzw. gleichmäßige Belastung hinbekommen. Allerdings solltest du am Ende dennoch auf die ~800W kommen.

RoFranz, 2024-05-11:
@spcqike: Hinsichtlich des oberen Leistungslimit vermute ich einen Bug (siehe https://github.com/helgeerbe/OpenDTU-OnBattery/issues/974#issue-2290802098). Vllt mache ich aber auch noch irgendetwas falsch.

Hinsichtlich der Schieflast, habe ich diese durch Verwendung von Y-Kabeln unter Kontrolle bekommen-

Danke für alle Hinweise, Ideen und die großartige Arbeit von euch hier im Forum.

schlimmchen, 2024-08-28:
Ich schließe dieses Issue, weil
* RoFranz sein Problem wohl mit einem verbesserten Hardware-Setup in den Griff bekommen
* das Scaling funktioniert inzwischen ganz anders und kennt nur noch verschattete Inputs statt nicht-produzierender Inputs und die hier erwähnt 2W Grenze spielt keine Rolle mehr dabei.

closed 2024-08-28 by schlimmchen