hoylabs/OpenDTU-OnBattery #846: Live-Ansicht: "Huawei AC Leistung" = "Huawei DC Leistung"

Linos88, 2024-04-05:
### What happened?

Hallo,
nur ein kleiner Schönheitsfehler :-)
In der Live-Ansicht unter "Huawei **AC** Leistung" wird die DC-Leistung angezeigt

[image]

### To Reproduce Bug

siehe Live-Ansicht

### Expected Behavior

Umbenennen oder Anzeigewert ändern

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.07

schlimmchen, 2024-04-11:
@Linos88 Kannst du bitte meinen vorgeschlagenen Fix aus #866 testen? Dort gibt es auch fertige [Firmware aus dem PR-Build Run](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8642597461?pr=866).

Linos88, 2024-04-11:
@schlimmchen sieht gut aus

[image]

PR #866 (2024-04-11): Fix AC charger power in live view
makes the value match its description. since most values in the top part of the live view are related to the AC side of the system, it makes sense to use the correct value rather than to change the description.

closes #846.
closed 2024-04-11 by schlimmchen