hoylabs/OpenDTU-OnBattery #850: [Request] Allow Manual Overwrite Of Power Limit

ButterBetzi, 2024-04-06:
### Is your feature request related to a problem? Please describe.

My Inverter does not feed power into my home circuit even though battery is full and solar passtrough is enabled.
Tried changing every parameter that I believe could have been the problem but no success... Now i've wasted half a days worth of solar power...

### Describe the solution you'd like

I want to be able to force the inverters power to a value that I decide myself. So -> overwriting the power limit manually.
So if a user experiences a similair problem like I currently do he has the ability to set a value himself.

### Additional context

If I can provide any information that would also solve my problem, please tell me so. Will do it ASAP!

Firmware: 2024.03.23

spcqike, 2024-04-06:
> If I can provide any information that would also solve my problem, please tell me so.

you could start by telling us your setup. Do you have a Victron MPPT? Why does solar pass through not work for you?

ButterBetzi, 2024-04-06:
as always. Just 5 minutes after I posted this request I found my bug... My power meter permanently reported 0 W.
So my initial problem is fixes, but I still think that my feature request is still valid for similair dumb users as myself ;)

schlimmchen, 2024-04-06:
> my power meter permanently reports 0 W.

If this happens again, it will be prominently visible in the future DPL card of the live view. That's because "power meter invalid" is its own state and the DPL state will be part of the live view for sure.

> but I think that my feature request is still valid

Why can't you disable the DPL and set a power limit? You can even do that using MQTT, if that is what you want, i.e., set DPL mode to "disable" and set a limit for your respective inverter.

spcqike, 2024-04-06:
> Why can't you disable the DPL

what about discharge protection? This doesn’t work for a fixed power limit without DPL, does it?

ButterBetzi, 2024-04-06:
> Why can't you disable the DPL and set a power limit? You can even do that using MQTT, if that is what you want, i.e., set DPL mode to "disable" and set a limit for your respective inverter.
Well, that would've worked I guess. Didn't think/know about this! That would do it for me at this point :D

ButterBetzi, 2024-04-06:
> You can even do that using MQTT, if that is what you want, i.e., set DPL mode to "disable" and set a limit for your respective inverter.

dont know if this fits here, but is there documentation about how I could do this? Would love to do it!

schlimmchen, 2024-04-06:
> dont know if this fits here, but is there documentation about how I could do this? Would love to do it!

The DPL mode is documented in the [Wiki](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT). For the inverter limit, refer to the upstream docs, which is linked on the mentioned Wiki page.

> Well, that would've worked I guess. Didn't think/know about this! That would do it for me at this point :D

Does that mean the issue can be closed?

ButterBetzi, 2024-04-06:
Thanks a lot!
Yes, can be closed :)

closed 2024-04-06 by ButterBetzi