hoylabs/OpenDTU-OnBattery #858: MPPT calculate the efficiency sometimes wrong

SW-Niko, 2024-04-08:
### What happened?

On MPPT with active DC load output the efficiency is wrong.

[image]

### To Reproduce Bug

Use the DC load output at night

### Expected Behavior

On MPPT with active DC load the calculation must be done including the load current.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

g0ed09ae

### Relevant log/trace output

```Shell
Screen shot at night.
```

### Anything else?

Minor bug. I will fix the issue in the next day and start PR.

schlimmchen, 2024-04-11:
The calculated power value also seems wrong, doesn't it?

Can you elaborate what the load output is, what it does, and when it is active? How does current flow when the load output is active?

SW-Niko, 2024-04-11:
Hello @schlimmchen ,
PR #867 is ready.

Background:
Some MPPT do have an load output. (VE.Direct value IL)
Now you can have different situations:
- All power from the panel flows into the battery
- One part from the panel power flows into the battery one part into the load
- All power from the panel flows into the load
- With no power from the panel all power into the load is taken from the battery.

In short ... to calculate the effectiveness of the MPPT you have to take the battery current and the load current into account.

> The calculated power value also seems wrong, doesn't it?

Yes, In the meantime I learned "Do not calculate with a mix of signed and unsigned integer" 😞

closed 2024-04-13 by SW-Niko