hoylabs/OpenDTU-OnBattery #870: [Request] min/max Zellenspannung berücksichtigen beim Laden & Entladen

Linos88, 2024-04-11:
### Is your feature request related to a problem? Please describe.

Mit der Zeit verändert sich jede Zelle individuell und somit auch ihr Verhalten im Kopf- / Fußbereich vom Spannungsverlauf. Eine Fahrweise der Batterie orientiert an der Gesamtspannung oder am SoC können Risiken mit sich bringen, wenn einzelne Zellen schneller Fallen bzw. Steigen.  

Mit dem folgenden möchte ich beispielhaft zeigen, dass Zellen unbeabsichtigt in den Spannungs-Grenzbereich geraten können obwohl laut Gesamtspannung und oder SoC alles gut ist (die gezeigten Genzen sind beispielhaft gewählt).    

[image]

Setup
-	 Huawei
-	 HM-600
-	 JK-BMS
-	 16S LFP  

Entladen
-	 Der SoC vom BMS mittlerweile stärker abgedriftet (Bekanntes Phänomen).
-	 Letztes Balancieren vor 5 Vollzyklen
-	 Batteriestopp SoC: 25%
-	 SoC-IST: 29%  
02:30 Anstieg der Last im unteren Spannungsbereich der Zellen. Verstärkte Spreizung der Zellenspannungen zu sehen. Soc-IST passt nicht mit der Gesamtspannung zusammen. 03:00 Die erste Zelle erreicht die SoC-0% Spannung. Das BMS korrigiert den SoC von ursprünglich 29% auf 0%, weshalb die Entladung stoppt  

Laden 
15:15 Erste Zelle erreicht Ladeschlussspannung und steigt weiter an. Max-Ladestrom wird immer wieder um 2A reduziert, um ein starkes Überschwingen der vollsten Zelle zu verhindern. So erhält die letzte Zelle genügend Zeit um nachzuziehen. 

### Describe the solution you'd like

Eine Berücksichtigung der min/max Zellenspannung bei Laden und Entladen würde ein über-/unterschreiten von Spannungsgrenzen vermeiden. 

Beispiel Laden: 
WENN --> die höchste Zelle einen einstellbaren Wert (zB maxCellV 3,45V) erreicht hat 
DANN --> reduziere Huawei Max-Leistung um einstellbaren Wert (zB 100 Watt), wiederhole wenn höchste Zelle erneut maxZelleV erreicht diesen
WENN --> Ladezyklus beendet
DANN --> Huawei Max-Leistung auf Ursprung wieder zurücksetzen 

Beispiel Entladen: 
WENN -->die niedrigste Zelle einen einstellbarer Wert (zB minCellV 3,00V) erreicht hat 
DANN -->Stop Wechselrichter, egal was SoC oder Gesamtspannung sagt 

### Describe alternatives you've considered

Da das System stand alone laufen soll/muss (Zitat: "Was ist SmartHome") ist das manuelle überwachen und Nachstellen sehr Zeitintensiv, birgt Risiken und vill. auch etwas unmöglich :-).

schlimmchen, 2024-04-11:
Nach allem was ich weiß, ist es normal, dass die Zellspannungen sowohl nahe der Ladeschlussspannung als auch unterhalb der 3.2V auseinanderdriften. Auch deshalb vermeidet man es (Stichwort: DoD) den Akkupack in diesem Bereich zu betreiben. Tut man das, vermeidet man auch, dass das BMS eingreifen muss weil eine Zelle zu weit abdriftet. Das wäre jetzt ehrlich gesagt meine Lösung für dieses Problem, statt je zwei weitere Schwellspannungen im DPL und im Huawei AC Controller zu berücksichtigen.

Linos88, 2024-04-11:
Hallo @schlimmchen, selbstverständlich hast du recht. Die Ursachen dafür liegen hauptsächlich in den Kapazitätunterschieden (resultierent zB aus Absorberchemie/Elektrolyt) und diversesten weiteren physikalischen Rahmenbedingungen. 

Der obere Grenzbereich lässt sich nicht vermeiden, da zur Kompensation des Zellendrifts balanciert werden muss. 
Den unteren Bereich kann man versuchen großzügig zu umgehen, dafür nehmen wir ein Reduzierung der nutzbaren Kapazität in Kauf. 
Wenn wir den unteren Bereich durch SoC oder Gesamtspannung begrenzen kann es passieren…

SoC stimmt mit der Zeit nicht mehr weil die Grenzen zum „kalibrieren“ nicht angefahren werden und oder der Strom nicht genau genug erfasst wird (JK-BMS meine Erfahrung, diverse andere wohl gleich). Folglich SoC im grünen Bereich und einzelne Zellen können bereits zu stark gesunken sein. 

Wenn man sich nach der Gesamtspannung richtet kann es hier auch vorkommen das ein einzelner Zellenwert zu stark abweicht und sich die Summe der Spannungen noch im Betriebsbereich bewegt