hoylabs/OpenDTU-OnBattery #871: [Request] Delay inverter shutdown if calculated limit < min limit

schlimmchen, 2024-04-11:
### Is your feature request related to a problem? Please describe.

* At dusk and dawn, the solar power (Victron MPPT, solar-passthrough) hovers around the inverter minimum limit, causing the inverter to be shut down and restarted possibly multiple times (battery "empty").
* Transient errors, like power meter readings not perfectly matching the reported inverter output power, may cause a calculated power limit to be below the min limit, causing an unwanted shutdown of the inverter.

### Describe the solution you'd like

Delay the actual inverter shutdown if the reason is `Status::CalculatedLimitBelowMinLimit`.

### Describe alternatives you've considered

I don't think there is a good workaround for the dusk/dawn situation.

Transient failures in calculation might stem from non-ideal setups. People should optimize their power meter setup and avoid oscillating inverters (cabling issues, 24V systems, etc.)

SW-Niko, 2024-04-13:
Hello @schlimmchen ,
I think it is an exzellent idea to solve all "Transient errors" and to keep the origin behavior of the function.

Use Case 1:
Switch the inverter off. If the inverter is not stable below a certain minimum power level.
After time x

Use Case 2:
Switch the inverter off. If the the grid power consumption is below the minimum power level. 
Automatic off used for holiday or leaving.
After time x

Use Case 3:
Avoid switch off. In case of transients.
Before time x

Use case 4:
Avoid switch off. Dusk and dawn condition
Before time x

To manage Use Case 1-3 I assume a delay (time x) of 1 minute is sufficient but if we want to include Use case 4 we need a much longer time and that can go into conflict with use case 1-2.

ali-baba-de, 2024-04-17:
Hello guys,
I stumbled over this discussion today and it made me think that the shutdown behaviour which I'm seeing in my system so far is actually not as normal as I thought it was.
I actually have two BKWs, one connected to a Pylontech battery (via Victron MPPT) and one directly to the grid. The consequence of this setup is that during periods of sunshine when the directly connected BKW delivers more power than the household consumes the power meter sends negative power and the inverter connected to battery is shutdown. During cloudy days the shutdown happens quite frequently as you can imagine.
I'm wondering now whether this behaviour is intended or just a side effect of the dusk/dawn use case? And couldn't the dusk/dawn use case not be identified better by taking `SunPosition.isDayPeriod()` into account?

schlimmchen, 2024-04-17:
> I'm wondering now whether this behaviour is intended

This is by design. The inverter is not working properly with low limits (depends on the setup and model). If you are already exporting energy, the inverter shall not be exporting more energy (unless full solar-passthrough is active). The energy shall go into the battery, don't you agree?

> And couldn't the dusk/dawn use case not be identified better by taking SunPosition.isDayPeriod() into account?

See #671.

ali-baba-de, 2024-04-17:
>The energy shall go into the battery, don't you agree?
Certainly I agree with this :-). If I have to judge whether charging the battery with a few Wh (just a guess, I have no good number how much it really is) is better than switching the inverter off and on frequently, I'm a bit lost. Shutdowns for today:

[image]

It's probably hard to recognize from the screenshot but this sums up to almost 100 shutdowns. If I look at the off/on timestamps a one minute "watch time" (meaning the condition has to be met fully during this interval) would already considerably reduce the number of off/on switches, a five minute watch time would be perfect. If you would like to see more details please let me know.

Regarding [#671](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/671) I fully support the dusk/dawn approach.

KlimekDE, 2024-08-09:
I totally agree with @ali-baba-de in this topic. I have the same situation with a 2nd Micro-Inverter which is directly connected to the grid and this is causing the same flapping of the inverter status on a daily basis. I think it would be a ideal solution to meassure for how long the total power consumption is below 0 in minutes for a period of time i.e. 5,10 or 15 minutes (maybe configureable) to make shutdown or turnup decisions.

Squint5471, 2024-09-03:
I also have the problem with a two inverter setup. I "solved" it by setting the power limits accordingly to prevent inverter shut downs. By doing so I loose energy in the grid connected inverter.

schlimmchen, 2025-02-01:
> Delay the actual inverter shutdown if the reason is `Status::CalculatedLimitBelowMinLimit`.

This enum value isn't even a thing any more in the current DPL implementation.

Solar-powered inverters are never shut down anyways.

Also, the hysteresis setting prevents toggling if it has a reasonable value.

closed 2025-02-01 by schlimmchen