hoylabs/OpenDTU-OnBattery #874: DPL Full-Solar-Passthrough seems to work with the uncorrected Voltages

Napizaki, 2024-04-12:
### What happened?

Hello,
it looks to me as if the DPL is switching on the full solar passthrough with the uncorrected battery voltage. My settings are 51V, and when the Victron timed exactly this battery voltage, the full solar passthrough switched on.
Shouldn't this happen with the voltage corrected by the load correction factor?

[image]
[image]

### To Reproduce Bug

Use the DPL with Full Solar Passthrough and obersrve the behavior

### Expected Behavior

It should be switched with the corrected voltages.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

schlimmchen, 2024-04-15:
Hm. Well, the code says that the load corrected voltage is always used when testing to start or stop full solar passthrough. Just one day before you opened this issue I had a similar experience: I was wondering why full solar-passthrough would not stop at the exact voltage I chose in the settings. That's because of the load corrected voltage. And you can see that in your graph as well: Even though the battery voltage drops from 51.0V to close to 50.0V, and your full solar stop threshold is 50.5V, full-solar passthrough does NOT stop, due to the use of the load corrected voltage.

However, looking at `getLoadCorrectedVoltage()` I realize that this function is broken in multiple ways
* It calculates the corrected voltage solely based on inverter AC output power, disregarding the actual load at the battery.
* It always calculates a higher voltage, even though the battery might be charging and the corrected voltage should be lower.

You are asking about the full solar passthrough start. Given that the inverter is producing, the corrected voltage would be higher than the reading of the Victron batery output, and hence it is plausible that full solar passthrough starts when the Victron reads 51.0V. In fact, it should start a little earlier, i.e., a slightly smaller voltage reading should also trigger full solar passthrough, as the voltage is corrected up.

The voltage 50.895V should trigger full solar-passthrough in your case (I assumed 150W to be produced by the inverter when full solar passthrough was engaged). That's close to 51.0V, but your graph clearly shows that the voltage at the Victron was 51.0V.

What battery do you use? If there is a compatible BMS attached, and from your screenshot it seems you do have the battery interface enabled, the BMS voltage is used to base DPL decisions on. I guess the reading at the BMS was different, which makes sense. The charge controller will output a slightly higher voltage than what is measured at the battery terminal when charging.

Napizaki, 2024-04-17:
Hello, thank you for the very detailed answer.
I am using a Pylontech us5000.
I have added the battery voltage to the screenshot.
[image]

schlimmchen, 2024-04-17:
Yeah, so this supports my case that the load corrected voltage is higher than the measured voltage, even if the battery is charged. It should be lower. Full Solar-Passthrough starts too early. The load corrected voltage should be the voltage that the battery has if it was idle, regardless of whether it is currently charging or discharging.

Should be adjust the title of this issue accordingly? Would a respective fix solve answer your question/solve your issue?

Napizaki, 2024-04-18:
If it makes sense to change the title, I'm happy to do so - what would be your suggestion?
Of course a fix would help - any errors in the code should be corrected if possible ;-)

Napizaki, 2024-04-27:
Here another graph that shows that something goes wrong with FSPT:
There I have a question: 
- What exact voltage is used for correction when a Pylontech is connected? And what power value is used for correction?

[image]
[image]

schlimmchen, 2025-02-01:
Superseded by #1399.

closed 2025-02-01 by schlimmchen