hoylabs/OpenDTU-OnBattery #885: Current Limit is updated very late / Aktuelles Limit wird sehr spät aktualisiert

Manos1966, 2024-04-17:
### What happened?

I would not say it is a "Bug" but, I report it just in case.

This is not "life threatening" so you could even consider it as "enhancement" or, discard it completely, if you wish.

### To Reproduce Bug

use DPL to manage the inverter.
when the Current Limit is set at a low level
deactivate DPL
Restart the inverter (in order to bring it to 100% power)

Watch how long the Current Limit stays at the last set limit, although the inverter is alread producing at max power.

### Expected Behavior

The Current Limit should be showing the real condition of the inverter much faster

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

schlimmchen, 2024-04-17:
> Watch how long the Current Limit stays at the last set limit, although the inverter is alread producing at max power.

Is this around 10 seconds by any chance?

Manos1966, 2024-04-17:
Nee, eher 100-130 Sekunden.

Ich habe auch bemerkt: 
- schalte ich den DPL aus
- dann stellt der DPL den Wechselrichter auf die min Leistung (Lower Power Limit) 
- und auf "Gelb" (gestoppt)

Dadurch musste ich ein Neustart des Wechselrichters machen, und war verwirrt weil das "Lower Power Limit" trotz Neustart, weiterhin angezeigt wurde. Bis ich bemerkt habe, dass die Anzeige "Inverter Total Power" volle Leistung zeigte.

Wie gesagt, "kein Drama", aber ich weiss was fuer ein Perfektionist du bist 😛

schlimmchen, 2024-04-17:
Hast du die Ressourcen, das einmal mit dem Upstream-Projekt zu reproduzieren?

* Inverter persistentes Limit auf 100% setzen (falls nicht schon der Fall).
* Inverter Limit auf z.B. 50W setzen (relatives Limit, non-persistent).
* Inverter in Standby versetzen.
* Inverter (a) neu starten ("Restart") oder (b) Einschalten ("Turn on").

(a) und (b) sind verschieden. Welchen Fall meinst du?
[image]

Meine Vermutung (und Hoffnung) ist, dass das kein OpenDTU-OnBattery-spezifisches Problem ist.

Manos1966, 2024-04-17:
Ich habe (a) neu starten ("Restart") benutzt.

Ich werde eine OpenDTU "Light" einspielen und testen, wahrscheinlich aber erst am Wochenende.

Manos1966, 2024-04-19:
Ich schliesse es erstmal, bis ich Zeit habe zum testen.