hoylabs/OpenDTU-OnBattery #890: JK BMS: Überschüssige Zeichen als Teil des BMS Namens

schlimmchen, 2024-04-18:
Ich habe nochmal eine Frage bzgl JK BMS. In der Live Ansicht werden ja auch die Namen der einzelnen Komponenten wie Victron oder Hoymiles angezeigt,  beim JK BMS steht aber dann bei mir "Batterie: Input Userda " und der Name den ich dem BMS per App vergeben habe. In der App sehe ich natürlich nur den Namen ohne den Zusatz. Gibt es eine Chance den Namen in der DTU noch irgendwo zu ändern, so das auch z.b. der mqtt Name auch etwas mehr "user-friendly" ist?

_Originally posted by @Tu4r3g21 in https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/415#discussioncomment-9103293_

schlimmchen, 2024-04-18:
Also das Problem ist jenes, dass dieser Name (nur) im Feld "Product ID" steht, und darin ist leider auch das, was man in der App als "Private User Data" eingeben kann vorne drangewurstelt. Dass man den Namen des BMS derart verändern kann, war mir leider nicht klar. Ich dachte das sei die fest eingebrannte Produktbezeichnung. Jedenfalls wird die Produktbezeichnung ab dem "JK" aus diesem String extrahiert. Du hast "KWB-BAT-01" gewählt, und daher wird als Fallback der gesamten "Product ID" String genommen, was eben auch die "Private User Data" enthält.

Es scheint aber so, als könnten wir die ersten 12 Zeichen als "Private User Data" herausnehmen, und die Zeichen danach als Namen des BMS, denn beispielsweise wird meine Product ID folgendermaßen aufgefüllt:

`[ 533139.441] JK BMS: ProductId: beki        JK_B2A24S15P`

Da werden scheinbar extra Leerzeichen eingefügt.

Daher schlage ich vor, nicht mehr nach "JK" zu suchen, sondern den String ab dem 13. Zeichen als Produktbezeichnung zu interpretieren.

schlimmchen, 2024-04-18:
@Tu4r3g21 Kannst du bitte die Firmware aus dem [Github Build Run zu #891](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/8737530036?pr=891) testen? Ich bin mir nahezu sicher, dass das dein Problem beheben wird. Wenn ich den Namen meines JK BMS auf etwas ändere, das kein "JK" im Namen hat, wird genau dieser eingestellte Name nun verwendet.

Tu4r3g21, 2024-04-18:
@schlimmchen Perfekt! Der Name kommt korrekt rüber!

schlimmchen, 2024-04-18:
Sehr schön, vielen Dank für das turbo-schnelle Feedback!

closed 2024-04-18 by schlimmchen

Tu4r3g21, 2024-04-18:
Ich habe zu danken! Nochmal, echt super diese Entwicklung. Danke für Deinen Einsatz und die vielen Mühen.

PR #891 (2024-04-18): Fix: JK BMS: BMS name is second part of product ID
Closes #890.