hoylabs/OpenDTU-OnBattery #893: DTU web interface hängt sich auf

Marc2805, 2024-04-18:
### What happened?

Das Web Interface ist nicht erreichbar, nur nach einem Neustart für kurze Zeit. Einmal die Woche hängt sich die DTU Komplett auf erkennbar an unverändertem Netz Bezug.  

### To Reproduce Bug

von alleine

### Expected Behavior

Funktionierendes Web Interface

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

andy75andreas, 2024-04-18:
Hallo,
bei mir das gleiche Verhalten.
Habe es am Log gemerkt.
[image]

Marc2805, 2024-04-18:
ist dein Web Interface erreichbar ?

andy75andreas, 2024-04-18:
Hallo,
-Webinterface -> nicht erreichbar
- MQTT -> Meldungen gingen von 10 - 18 Uhr keine raus. (18 Uhr manueller Neustart)
- Display -> eingefroren

Bis vor wenigen Tagen lief alles Probemlos.
Ich habe lediglich eien Powermeter (IR-Kopf über Seriell) angeschlossen und den DPL aktiviert.
Soweit ich das bisher beurteilen kann ging das auch 2-3 Tage bis heute um 10 Uhr.
-HMS-1600 über CMT2300A
-IR-Lesekopf Seriell
-SSD1306 Display
-Olimex ESP32-POE-ISO am Ethernet

Marc2805, 2024-04-18:
Hallo,
ich hatte seit dem update (ca. 3Wochen) keine Probleme.
Bei mir wurde auch nichts geändert.
HM-1500 über nRF24
Shelly EM3
SSD1306 Display
esp-wroom-32
Was steht bei dir in dem Speicherinformationen ?

stanpete78, 2024-04-19:
Ich habe in letzter Zeit auch wieder verstärkt Aussetzer. 
So ein-zwei Mal täglich verabschiedet sich die DTU aus dem (sehr guten) Wlan, läuft aber anscheinend weiter. Auch aus-/einschalten hat keinen Effekt. Nach 1-3 Stunden läuft dann aber wieder alles normal.
Ich hatte überlegt, ob meine irgendwie anfällig für niedrige Temperaturen ist, weil es eher nachts passiert und es ja in letzter Zeit wieder kälter wurde.

Alles sehr seltsam :)

schlimmchen, 2024-04-19:
> Habe sie im unbeheizten Gartenhaus.

Ich auch, aber keine derartigen Probleme.

> -Olimex ESP32-POE-ISO am Ethernet

@andy75andreas Kannst du das Board auch mit einer guten 5V Spannungsquelle betreiben und schauen, ob das Problem dann verschwindet? Eine gute Stromversorgung ist wichtig, und auch wenn mit Olimex was sagt bin ich bei PoE sofort skeptisch.

Es wäre natürlich klasse, wenn einer von euch mit Problemen einen seriellen Log einfangen könnte über den Zeitpunkt hinweg, zu dem das Problem auftritt.

Marc2805, 2024-04-19:
Das Problem mit dem Internet hatte ich davor auch. Die DTU ging immer in den AP Modus.
Hab das ganze System neu aufgesetzt jedoch ohne Erfolg.
Erst nach einem Neustart meiner FritzBox war das Problem behoben

andy75andreas, 2024-04-19:
Mit dem 5V Netzteil kann ich mal versuchen.
Ich mache mal einen USB Tester dazwischen um zu sehen wieviel Leistung die DTU nimmt. 
Ich glaube aber nicht das es daran liegt, wobei das ganze angefangen hatte als ich den IR-Lesekopf angeschlossen hatte.
Auf dem Olimex ist ein 5V/2W DC-DC macht also 0,4A.
Ich denke mal der 3V Regler auf dem Board ist ehr das Problem, denn da hängt alles dran.
ESP32, Display (auf 20% gedimmt), IR-Lesekopf und der CMT2300A (mit Stützelko).

andy75andreas, 2024-04-21:
Ich habe mal ein Netzteil an die DTU gemacht.
Ich habe mit einem USB-Monitor die Spannung gemessen.
-5,2V bei 0,1A mit Spitzen bei 0,4A alle 1s. (Wahrscheinlich vom Funkmodul.
Alle Komponenten hängen an 3,3V. Beim Eingangsstom von max 0,4A bin ich weit weg von den max zulässigen 0,86A.
Spannungsversorgung ist also nicht auffällig.
[image]
Ich werde das jetzt ein paar Tage beobachten .

andy75andreas, 2024-05-25:
Hallo,

ich würde sagen das es besser geworden ist aber nocht nicht weg ist.

- Mittlerweile habe ich die neueste FW auf meinem Olimex ESP32-POE.
- habe ein gutes 5V Netzteil zur Stromversorgung eingesetzt
- habe anstelle des ungeschirmten Kabels ein Cat6 Kabel verlegt und den Schirm an den PE vom Haus angeschlossen.
Somit kann ich jegliche Störungen über das Kabel ausschließen.

Trotzdem landen immer mal wieder falsche Werte in der DTU bzw in der der Datenbank.
Das sieht dann wie folgt aus: (zufällig im Webinterface gesehen)
[image]
In Grafana habe ich ja Dank der Hilfe von hier einen Filter eingebaut.
Ohne Filter würde es dann so aussehen:
[image]
Und mit Filter so:
[image]

An der Art und Weise wie der Stromzähler die Daten über die IR-Schnittstelle ausgibt werden wir nicht ändern können.
Das Kabel und die Stromversorgung kann ich nun als Fehlerquelle ausschließen.

Ich denke mal das das Problem im Parsing des Datenstromes der Seriellen Schnittstelle liegt.
Hier werden die Daten nicht auf Plausibilität geprüft sondern nach einem festen Muster in die einzelnen OBIS Werte zerlegt.
Wenn irgendein Zeichen in diesem Datenstrom nicht korrekt ist kommt es zu dem oben genannten Problemen und der Datenstrom wird mit völlig falschen Werten angezeigt und verarbeitet.
Das führt leider auch zum sporadischen Aufhängen der DTU.

Hier habe ich nochmal einen kleinen Ausschnitt der Konsole falls Ihr das was sehen könnt:
`Fetch inverter: 116494503###
09:19:22.360 > TX RealTimeRunData 865.00 MHz --> 15 94 50 32 63 80 12 85 08 80 0B 00 66 51 90 F9 00 00 00 00 00 00 00 00 96 44 98 
09:19:22.464 > Interrupt received
09:19:22.510 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 01 00 01 01 41 01 3E 01 0C 01 1F 03 60 03 91 00 01 83 | -22 dBm
09:19:22.555 > Interrupt received
09:19:22.601 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 02 2E E7 00 01 40 58 00 D1 00 D0 01 3D 01 3F 01 27 E8 | -22 dBm
09:19:22.646 > Interrupt received
09:19:22.692 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 03 01 17 03 AA 03 7C 00 01 4D A2 00 01 3A 2E 00 D5 F2 | -22 dBm
09:19:22.747 > Interrupt received
09:19:22.800 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 04 00 D6 09 67 13 87 0D 64 00 02 00 8E 03 E7 01 37 00 | -22 dBm
09:19:22.873 > Interrupt received
09:19:22.921 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 85 00 02 31 BC 15 | -22 dBm
09:19:22.967 > RX Period End
09:19:23.013 > Success
09:19:23.059 > Interrupt received
09:19:23.105 > Interrupt received
09:19:23.153 > Frame kaputt
09:19:27.275 > Fetch inverter: 116494503###
09:19:27.339 > TX RealTimeRunData 865.00 MHz --> 15 94 50 32 63 80 12 85 08 80 0B 00 66 51 90 FE 00 00 00 00 00 00 00 00 A6 62 89 
09:19:27.393 > Interrupt received
09:19:27.439 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 01 00 01 01 41 01 3E 01 0A 01 1D 03 5A 03 8B 00 01 A7 | -22 dBm
09:19:27.486 > Interrupt received
09:19:27.531 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 02 2E E7 00 01 40 59 00 D1 00 D1 01 3D 01 3F 01 26 E9 | -22 dBm
09:19:27.577 > Interrupt received
09:19:27.623 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 03 01 15 03 A4 03 77 00 01 4D A2 00 01 3A 2E 00 D5 F5 | -22 dBm
09:19:27.669 > Interrupt received
09:19:27.721 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 04 00 D6 09 64 13 88 0D 4F 00 02 00 8D 03 E7 01 37 24 | -22 dBm
09:19:27.775 > Interrupt received
09:19:27.821 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 85 00 02 A7 92 AD | -22 dBm
09:19:27.866 > RX Period End
09:19:27.913 > Success
09:19:27.989 > Interrupt received
09:19:28.037 > Interrupt received
09:19:28.195 > Frame kaputt
09:19:28.249 > Interrupt received
09:19:28.298 > Frame kaputt
09:19:31.004 > PowerMeterClass: TotalPower: -184.00
09:19:32.285 > Fetch inverter: 116494503263
09:19:32.341 > Request SystemConfigPara
09:19:32.396 > TX RealTimeRunData 865.00 MHz --> 15 94 50 32 63 80 12 85 08 80 0B 00 66 51 91 04 00 00 00 00 00 00 00 00 C7 49 38 
09:19:32.449 > Interrupt received
09:19:32.506 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 01 00 01 01 41 01 3E 01 09 01 1B 03 54 03 86 00 01 A1 | -22 dBm
09:19:32.557 > Interrupt received
[... 29 lines cut ...]
09:19:38.426 > Interrupt received
09:19:38.478 > Interrupt received
09:19:38.527 > Interrupt received
09:19:42.033 > PowerMeterClass: TotalPower: -177.00
09:19:42.291 > Fetch inverter: 116494503###
09:19:42.354 > TX RealTimeRunData 865.00 MHz --> 15 94 50 32 63 80 12 85 08 80 0B 00 66 51 91 0E 00 00 00 00 00 00 00 00 67 37 EC 
09:19:42.408 > Interrupt received
09:19:42.520 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 01 00 01 01 41 01 3E 01 06 01 18 03 4B 03 7C 00 01 48 | -22 dBm
09:19:42.623 > Interrupt received
09:19:42.671 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 02 2E E8 00 01 40 59 00 D2 00 D1 01 3D 01 3F 01 21 E2 | -22 dBm
09:19:42.720 > Interrupt received
09:19:42.779 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 03 01 11 03 96 03 67 00 01 4D A2 00 01 3A 2E 00 D5 D3 | -22 dBm
09:19:42.826 > Interrupt received
09:19:42.872 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 04 00 D6 09 62 13 86 0D 16 00 00 00 8B 03 E8 01 37 7E | -22 dBm
09:19:42.933 > Interrupt received
09:19:43.032 > RX 865.00 MHz --> 95 94 50 32 63 80 12 85 08 85 00 02 41 7A A3 | -22 dBm
09:19:43.085 > RX Period End
09:19:43.236 > Success
09:19:43.282 > Interrupt received
09:19:43.329 > Frame kaputt
09:19:43.375 > Interrupt received
09:19:43.543 > Frame kaputt
`
Was irgendwie komisch ist "Frame kaputt", keine Ahnung auf was sich das bezieht.
Der Abstand zwischen DTU und WR ist ca. 2m.

schlimmchen, 2024-10-06:
> Trotzdem landen immer mal wieder falsche Werte in der DTU bzw in der der Datenbank.

Dein Screenshot zeigt ungültige PowerMeter-Werte. Diese Probleme sind sicherlich im Griff inzwischen (seit dem PowerMeter Refactoring).

Dieses diffuse Issue schließe ich jetzt, weil es hierzu keine weiteren Rückmeldungen hab. Es gibt keinen Grund anzunehmen, dass der web interface sich regelmäßig "aufhängt". Auch im WiFi-Teil wurden Verbesserungen aus dem Upstream übernommen.

Falls dieses oder ein ähnliches Problem tatsächlich weiterhin besteht, dann bitte ein neues Issue erstellen mit konkreter Fehlerbeschreibung.

closed 2024-10-06 by schlimmchen