hoylabs/OpenDTU-OnBattery #897: [Request]Zendure SolarFlow hub support

Meatstake, 2024-04-20:
### Is your feature request related to a problem? Please describe.

I wonder if it would be possible to add a support of the Zendure SolarFlow hubs to get the battery data from them and organise a full solar bypass if the battery is by 100%. 

### Describe the solution you'd like

The esp reads battery level from Zensuren hub. If battery level is by 100% and the hub does not charge it anymore (it is important, because the %-measurement of this system is not reliable), then the Hoyemiles inverter is set to the value, that allows a pass through of all the produced solar energy till the battery starts to discharge itself. When the battery starts to discharge itself, then at first, the solar bypas value sets itself to the lower level, allowing to charge the battery, and if it does help, and the battery does not charge, but gets further discharged (sunset, clouds etc.), then everything just returns to normal mode. Till the battery is 100% and not charged again.

schlimmchen, 2024-05-24:
Please provide information on how the communication with the Zendure SolarFlow Hub works. Serial? CAN? Bluetooth? HTTP? MQTT?

Also explain what that product does. Is it "only" a battery with an internal BMS?

Is there a Victron charge controller with VE.Direct available in this setup to realize solar-passthrough?

apss-fuchs, 2024-05-24:
MQTT can be used for this.
See here: https://github.com/z-master42/solarflow/blob/main/solarflow.md

The Field "electricLevel" is battery level

It can read from the [appKey]/[deviceID]/state

Example:
{
  "outputHomePower": 199,
  "packInputPower": 208,
  "electricLevel": 64,
  "sn": "ABC"
}

danny1605, 2024-05-29:
It would be great if Zendure were available in OpenDTU via MQTT.

apss-fuchs, 2024-06-04:
@schlimmchen Can we extend the JSON field with a JSON variable?

schlimmchen, 2024-06-04:
I'm not sure that I can follow...

You need the MQTT battery provider to parse JSON in the payload published to the respective topic? Similar to the HTTP+JSON power meter implementation?

apss-fuchs, 2024-06-04:
@schlimmchen Yes, but the JSON is in in the MQTT topic. 
We need to configure a JSON path inside the MQTT topic.
[image]

schlimmchen, 2024-06-04:
Alright, that's straight-forward. I did the work required in preparation for this already in the context of the powermeter refactoring, which lands eventually. In there I also added parsing JSON payload to the MQTT power meter implementation. We can do the same here. Please stand by, I'll trigger you to have you test it once the refactor is ready.

danny1605, 2024-06-06:
Hello, is there any news yet? I would offer to test it out

TvT73, 2024-06-28:
Oh das würde mich auch interessieren. Zumal der Zendure bei mir nur an 2 von den 4 MPPT Trackern läuft (HMS-18004T)

switchtheflip, 2024-07-08:
Das wäre ein grandioses feature!

Anscheinend hat sich schon mal jemand die Zendure-Kommunikation angesehen, sogar unter Umgehung von Cloud und App, vielleicht hilf das hier:

https://github.com/reinhard-brandstaedter/solarflow-bt-manager

apss-fuchs, 2024-07-23:
@schlimmchen any updates on this?

schlimmchen, 2024-07-23:
[image]

I'll prioritize this. Expect an invitation to test a pull request build in the near future, maybe tonight.

PR #1127 (2024-07-31): Feature: support JSON payload in MQTT battery provider
this changeset adds support for parsing the MQTT battery provider's SoC and voltage topics' payloads as JSON to extract a numeric value at a configurable path.

closes #897.
closed 2024-07-31 by schlimmchen