hoylabs/OpenDTU-OnBattery #899: [Request] Add support for intermediate meter to measure the outputpower of the inverter

greymda, 2024-04-20:
### Describe the solution you'd like

Basically the idea comes from the python script https://github.com/reserve85/HoymilesZeroExport (in detail about this option https://github.com/reserve85/HoymilesZeroExport/blob/5b226225a6799eab91857a2c18d21d854c568726/HoymilesZeroExport_Config.ini#L133) and consists on having the (optional) support for an intermediate meter that measures the outputpower of the inverter as it is faster than the DTU current_power value (and possibly more reliable?).

In my particular care I have a Shelly EM with 2 clamps and one monitors the Grid input and the second is measuring the Solar input. I would like to have an option to use the values from the second clamp for Dynamic Power Limit functionality.

[image]

Thoughts?

schlimmchen, 2024-04-20:
We were obviously triggered by different things, but I had this exact idea just moments ago, because it would also appease the 24V battery users.

I would say let's just copy the PowerMeter, name it something else, and if enabled and working, use that value to know the inverter(s) output. Do not wait for new statistics in that case. One can lower the poll frequency and gain more airtime to transmit new limits.

I think this is an awesome idea :+1:

Guess I will add a power meter of some sort when upgrading my setup anyways next week :wink:

Please note, however, that when I write "just copy the PowerMeter", I mean "do the urgently needed refactor of the PowerMeter code base first, have this new feature in mind while doing so, then try to implement this new feature". There will be hiccups, I guess.

greymda, 2024-04-20:
looking forward!

greymda, 2024-04-20:
@schlimmchen but you got my interest about the 24v battery users. can you elaborate a bit when have some spare time?

schlimmchen, 2024-04-20:
> spare time

Spare what? :wink:

Start here: https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/24V-Battery-issues-with-Hoymiles

SW-Niko, 2024-04-21:
Hello @greymda,
I am one of the poor 24VDC battery users 😢  and I really like your idea to use an external power meter to measure the inverter output power. 👍 
I see 2 major advantages:
- We get rid of the nasty bad power measurements from the inverter running on 24VDC
- We can speed up regulation, if the external power measurement is faster as the inverter measurement

Squint5471, 2024-09-03:
I am also interested in this feature :)

rradkov, 2025-09-30:
#2200 I think this is a great future that I'm waiting for so long... Just want to add a hint here. There are cases(like mine) that two inverters HMS-2000 and HMS-1000 are connected to the same intermediate meter - Shelly 1 PM so please include this type of case as well.