hoylabs/OpenDTU-OnBattery #900: Setting DPL to mode 2 stops the inverter

saschaabraham, 2024-04-21:
### What happened?

Setting the DPL mode to 2 through Home Assistant stops the inverter, instead of setting to power limit to the configured limit. In my case 800W on a HM1200.

Below you see in the log some error at the moment the DPL mode is switched to 2. 

Configuration is HM1200 restricted to 800W in the config only a ESPHome SmartMeter reader reporting the power meter values via HTTP+JSON. No battery or similar.

### To Reproduce Bug

Set the DPL mode to 2 with any addition components defined.

### Expected Behavior

Setting the DPL mode to 2 increases the limit of the HM1200 to the configured 800W with no further change. As explained in the wiki.

The idea is, while running the washing machine or dryer, setting the limit to the max, because while they are running, the power consumption is very irregular and the regulation of the inverter ouput is not needed.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

### Relevant log/trace output

```Shell
09:52:54.565 > Received MQTT message on topic: BalkonSolar/powerlimiter/cmd/mode
09:52:54.612 > Power limiter unconditional full solar PT
09:52:55.096 > RX Period End
09:52:55.144 > Success
09:52:55.272 > Fetch inverter: 116182803526
09:52:55.325 > [DPL::updateInverter] actual limit is 55.0 % (660 W respectively), effective 2034 ms after update started, requested were 55.2 %
09:52:55.378 > [DPL::announceStatus] VE.Direct disabled, connection broken, or data outdated
09:52:55.428 > [DPL::updateInverter] Stopping inverter...
```

schlimmchen, 2024-04-21:
> As explained in the wiki.

Where?

The documentation for the DPL mode, editable using MQTT, ist documented here: https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT#power-limiter-topics It explains that mode 2 is unconditional full solar-passthrough. Without VE.Direct, there is no solar-passthrough. Why should the DPL set the inverter to the upper limit without a known value for the solar output power?

If you want to disable the DPL using MQTT, use mode 1, then set a limit for the inverter and start the inverter.

saschaabraham, 2024-04-21:
The wiki stated "2 - Unconditional Solar Passthrough: The power limit is set such that all solar power (adjusted for efficiency) is fed into the home. This is done irrespective of grid consumption and without consuming energy from the battery, which is the key difference to the regular Power Limiter Solar Passthrouh feature. Essentially, the inverter mimics the behavior of a traditional, non-smart inverter"

So why do i need the ve.direct for this?

schlimmchen, 2024-04-21:
Yeah, okay, you are right, I should have re-read the explanation.

The implementation of mode 2 is what it is. We shall update the docs (Wiki) accordingly.

Now, what is your goal? I really did not understand that part. You wish to disable the DPL, but the inverter shall be configured with the upper power limit?

saschaabraham, 2024-04-21:
Yes, i think so. at this moment mode 2 is the same as mode 1, if there is no ve.direct defined.

The log message said "VE.Direct disabled, connection broken, or data outdated" before the stop of the inverter. I think the case "VE.Direct disabled" should be ignored, because the disabled is not the same like a broken connection.
So there sould be 3 different behaviors behind the DPL-mode
0 - Runs normally with dynamic limit
1 - Stop the inverter - sets the output to zero
2 - sets the ouput to the defined upper limit. It should use the VE.direct values if it is defined and set the output just to the defined upper limit if there is no VE.direct. If there is an error within the ve.direct, the inverter should stop.

schlimmchen, 2024-04-21:
> No battery or similar.

Oh, okay. In that case, my statement "mode 2 is what it is" is also wrong, we can indeed do better.

> It should use the VE.direct values if it is defined and set the output just to the defined upper limit if there is no VE.direct.

That does not work for users with a battery but without Victron charge controllers, which there are also :wink:

Let's do this (I updated the wiki, and will change the code accordingly):

[image]

What do you think?

saschaabraham, 2024-04-21:
I think this is the right way.😉

PR #903 (2024-04-22): Fix: DPL mode 2 for solar-powered inverters
for solar-powered inverters, the unconditional full solar-passthrough implementation now sets the upper limit configured in the DPL settings.

Closes #900.
closed 2024-04-22 by schlimmchen