hoylabs/OpenDTU-OnBattery #904: Probleme mit DPL Schwellwerten bei Upgrade von 2023.03.07

schlimmchen, 2024-04-21:
Ich hatte heute auch Probleme mit Inverter-Abschaltung. Ursache ist zwar vermutlich eine andere, aber passt dennoch zum Thema.
Mein Setup: HMS-1800-4T, 4 Module, keine Batterie
Im 2024.03.07-Release hatte ich die DTU erstmalig eingerichtet, noch ohne Wechselrichter. Die Parameter hatte ich bereits für Betrieb ohne Batterie gesetzt, die Schwellwerte für Batterienutzung entsprechend auf 0 bzw. 1 gesetzt. In der aktuellen Version gibt es den Schalter "Speisung von Solarmodulen". Ist dieser aktiviert, werden die Eingabemasken für den Schwellwert deaktiviert. Hier standen jedoch noch 0 und 1 als Wert, Minimumwert wird nun aber als 16 angegeben. Es kommt jedoch keine Fehlermeldung, wenn die nicht angezeigten Werte falsch sind. Der Umrichter schaltet dann jedoch ab. Nach Korrektur der Werte auf 16, Speichern und anschließendem Aktivieren von "Speisung aus Solarmodulen" schaltet der Wechselrichter nicht mehr ab.

_Originally posted by @E1yot in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/808#issuecomment-2068167500_

schlimmchen, 2024-04-21:
@E1yot Ich bin verwundert darüber. Hast du die 2023.03.23 Release eingespielt und versucht den Schalter "Inverter wird an Solarmodulen betrieben" einzuschalten, oder war das eine Vorabversion bzw. eine selbstgebaute Firmware auf einem Zwischenstand?

Es ist so, dass die alten Werte keine Rolle spielen sollten beim Absenden des Formulars mit dem DPL Einstellungen, weil die Werte nicht nur unsichtbar sind, sondern tatsächlich nicht Teil des DOM, daher sollte da auch keine Überprüfung beim Abschicken passieren.

Diesen Fehler hatte @spcqike schon in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/755#issuecomment-1999572633 bemerkt, und das Problem sollte wie gesagt nicht auftreten.

Manos1966, 2024-04-21:
Hi @schlimmchen 

Beide auf 16 ?
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Dynamic-Power-Limiter-(Nulleinspeisung)-%E2%80%93-OHNE-BATTERIE#important-note-for-users-updating-their-system-from-previous-versions

E1yot, 2024-04-21:
Die DTU war schneller da, als das Solarset. Daher hatte ich bereits ein wenig "gespielt". Als die Anlage installiert war, habe ich das Update durchgeführt und erstmal ohne Limiter gefahren, um ein paar Daten zu sammeln und das Zusammenspiel zu beobachten. Den Limiter hatte ich zwar schon ein paar Mal eingeschaltet, aber immer das Problem der Abschaltung (innerhalb von Sekunden). Heute habe ich intensiver getestet und das Problem mit den Werten gefunden. Nach dem die Werte angepasst wurden, gab es keinen Stop des Inverters. Ich hätte ja auch erwartet, dass die ausgeblendeten Werte keine Rolle spielen. Gibt es evtl Abhängigkeiten von den Werten, die nicht offensichtlich sind? Werden ggf. beim Speichern noch weitere Parameter geändert? Den Quelltext habe ich noch nicht studiert, daher hier erst einmal nur die Meldung.
Ich habe bisher nur die offiziellen Builds verwendet, Hardware ist die AllianceApps OpenDTU Fusion, Powermeter kommt über MQTT.