hoylabs/OpenDTU-OnBattery #907: WR detail view is missing

Rolober, 2024-04-22:
### What happened?

Hello
on live view I go no details of the WR, only "Warte auf Daten". Also the "P" is missing. Could you help?

Rolober
[image]

### To Reproduce Bug

alway missing

### Expected Behavior

n/a

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	2024.03.23

Rolober, 2024-04-22:
Why work the FW version 2024.01.26 with the HMS?

Rolober, 2024-04-24:
thx, migsun. No, the same issue appears. Missing of panels detail etc.

[image]

schlimmchen, 2024-05-03:
Continued in #912.

closed 2024-05-03 by schlimmchen