hoylabs/OpenDTU-OnBattery #912: only last Version 2024.01.26 support HMS 4t 2000 Inverter

Rolober, 2024-04-22:
### What happened?

the current and down to 2024.2.x Firmware doesn't support the HMS 4t 2000 valid. the live view Shows not the complete Inverter data and always the message 'warten auf daten' apeaers. Is this Inverter out of support?
Kindle regards
Roland

### To Reproduce Bug

yes, always there    

### Expected Behavior

Support of HMS

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

last one

### Relevant log/trace output

```Shell
n/a
```

DonJohnLong, 2024-04-22:
I have a hms-2000 4t and it works fine... did you check your wires and pin mapping?
[image]

Rolober, 2024-04-22:
thx, if i flash a newer Version, the view can't display the values correct.
[image]

schlimmchen, 2024-05-03:
> Pre-Compiled binary from GitHub

Is this really the case? You are not building your own firmware?

How does the radio status look like with the current firmware version?

[image]

Do you have other inverters with which you are communicating okay?

Did you try to press Ctrl+F5 in your browser to force a reload of the web application?

Does this problem occur on multiple devices and in incognit/private browsing mode?

Rolober, 2024-05-04:
It is a pre compiled image generic ESP32 and I tried with different browsers, always the same issue.
I got the information from the FB Opendtu on batterie, that only this last image are working well. Newer images has the  issue "warten auf Daten". That means, other users have the same issue. If I install the version 2024.01.26, I got all values from the hoymiles inverter HMS 2000 t4. But this version could not perfect the role "Nulleinspeisung", newer versions are better.

schlimmchen, 2024-05-04:
What ESP32 are you using? Are you able to test yesterday's release?

I strongly believe that this is an upstream issue and I also guess that it has been resolved already. Can you take some time to skim (closed) issues in the upstream project https://github.com/tbnobody/OpenDTU/issues?q=is%3Aissue+is%3Aclosed to check if the issue is known there and was fixed?

However, this would not explain why @DonJohnLong does not have this problem. This is puzzling... Maybe it is an issue with the serial number? Then again, this would mean that there was a regression based on the serial number?

Still, please try to find hints regarding your issue in the upstream project. This should not be onBatter-specific.

krelac3, 2024-05-05:
Exactly same issue here. 

Brand new HMS-2000-4T, same device firmware 1.0.16 as pic on top of issue (production week 33). Downgraded the ESP firmware and seems only 2024.01.26 shows the data from inverter, any newer firmware shows "wait for data..." forever.

schlimmchen, 2024-05-05:
After you put in the serial number to register the inverter with the OpenDTU, is the inverter correctly detected as an HMS-2000-4T series?

What's printed on the serial console? Maybe there is a hint about the cause of this issue?

krelac3, 2024-05-05:
Inverter serial number is correct within OpenDTU, with fw version 2024.01.26 all seems correctly detected and data from inverter is visible. Integration with VenusOS on rpi works good as well.

With latest version no go. 

Logs attached (replaced actual serial number with XXX_MY_HOYMILES_XXX).

 
[opendtu_capture_2024.03.23.txt](https://github.com/helgeerbe/OpenDTU-OnBattery/files/15213711/opendtu_capture_2024.03.23.txt)
[opendtu_capture_2024.01.26.txt](https://github.com/helgeerbe/OpenDTU-OnBattery/files/15213712/opendtu_capture_2024.01.26.txt)

schlimmchen, 2024-05-05:
The logs don't help, I can't see an issue there.

What happens if you install the most recent upstream firmware OpenDTU v24.4.24?

What other OnBattery features do you use? VE.Direct? A Battery interface? Are those cards shown as expected in the live view?

Are you logged into OpenDTU?

What do you see when accessing `http://<ip>/api/livedata/status` and `http://<ip>/api/livedata/status?inv=<inverter serial>` directly?

schlimmchen, 2024-05-05:
@krelac3 Are you also using a release FW from github rather than building your own firmware?

krelac3, 2024-05-05:
I stand corrected. 

All works fine and as expected on either openDTU or OpenDTU-OnBattery latest firmware. Seems my local running tinyproxy on loopback made an issue although there were no issues when running older firmware versions. Not sure what changed in last 6 months or so within firmware. In any case, bypassing DTU's ip in proxy sort it out and data from inverter shows up. 

Thank you for the assistance.

schlimmchen, 2024-05-05:
@Rolober Is there a chance you experience a similar issue, i.e., where your network setup breaks something? Have you tried a different browser, Ctrl+F5, clearing the browser cache, or private browsing mode to check is the issue is gone then?

Rolober, 2024-05-06:
Hi
I use a reverse proxy on my PI, for outside connection. If I use the internal IP, all values now are there. I don't know why this happens, but many thanks for your help Bernhard.
Roland

closed 2024-05-06 by Rolober

PhilJaro, 2024-05-06:
Just for your Information, you have to activate websocket support on your proxy (on Nginx this is disabled per default), otherwise the values can't get pushed onto the webapp.