hoylabs/OpenDTU-OnBattery #913: 2x MPPT + Shunt funktionieren nicht

kloppy1984, 2024-04-23:
### What happened?

Ich versuche das System mit 2x Victron MPPT und 1x Shunt zum laufen zu bekommen,
Ich hab die Version von gestern den 22.04.2024 genommen.

Das System mit 1x MPPT und Shunt läuft Problem los.
Jetzt habe ich es auf den zweiten MPPT erweitert.

Da ihr dieses auch in Device Vorlage habt, bin ich davon ausgegangen das diese Konstellation funktioniert.

Leider fehlt mir immer ein Teilnehmer, entweder Shunt oder MPPT

 

### To Reproduce Bug

Sobald ich die config in der Hardware aktivere, Sehe ich 2x MPPT, 
aber die Batterie (Shunt) nicht mehr erreichbar:

[image]

---------------------------------------
Nun deaktiviere ich den Ve.Direct + Batterie, 
danach aktiviere ich wieder die Batterie 

Batterie (Shunt) ist wieder da:
[image]

---------------------------------------
Nun aktiviere ich den Ve.Dictect.
Es fehlt jetzt der zweite MPPT:
[image]

---------------------------------------
über den einen MPPT sehe ich die Gesamtleistung beider MPPT´s
[image]

### Expected Behavior

Ist es möglich das alle drei Victron Geräte angezeigt werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

g0000000

doublexminus, 2024-04-23:
Kann das Problem bei mir auch bestätigen.
Im Log taucht beim aktivieren der Battery das folgende auf:
`16:45:22.489 > [Battery] Serial port 2 already in use. Initialization aborted!`

Freaksplace, 2024-04-23:
Auch hier der gleiche Fehler

doublexminus, 2024-04-23:
Glaube ich habe grade in den Kommentaren zum PR mit welchem das Feature eingeführt wurde die Antwort gefunden:
https://github.com/helgeerbe/OpenDTU-OnBattery/pull/706#issuecomment-2012552252

Wenn man danach geht, ist es nach aktuellem Stand nicht möglich. 

Wenn mir @Gumbagubanga und/oder @schlimmchen das bestätigen, dann würde ich mich anbieten eine entsprechende Info in der Doku zu hinterlassen.

Für alle die zumindest die Gesamt-DC-Leistung der 2 Victrons haben möchten, schaut euch mal die folgende Diskussion an:
https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/873

Gumbagubanga, 2024-04-23:
> Glaube ich habe grade in den Kommentaren zum PR mit welchem das Feature eingeführt wurde die Antwort gefunden: [#706 (comment)](https://github.com/helgeerbe/OpenDTU-OnBattery/pull/706#issuecomment-2012552252)
> 
> Wenn man danach geht, ist es nach aktuellem Stand nicht möglich.
> 
> Wenn mir @Gumbagubanga und/oder @schlimmchen das bestätigen, dann würde ich mich anbieten eine entsprechende Info in der Doku zu hinterlassen.
> 
> Für alle die zumindest die Gesamt-DC-Leistung der 2 Victrons haben möchten, schaut euch mal die folgende Diskussion an: #873

Korrekt. Man kann aufgrund von Hardware-Limits keinen SmartShunt und 2 MPPTs gleichzeitig betreiben.

kloppy1984, 2024-04-23:
Dann bitte im Device Profile Vorlage (pin-mapping) Diese Konstellation raus nehmen!

`"name": "OpenDTU FUSION v2 + 2xVictron MPPT + Victron SmartShunt",`
Man geht davon aus, das es dann möglich ist  wenn es dort vorgeschlagen wird.

schlimmchen, 2024-04-23:
> Wenn man danach geht, ist es nach aktuellem Stand nicht möglich.

Korrekt. Mein Kommentar, den du verlinkt hast, ist noch zutreffend.

> Voraussetzung, wenn beide per Bluetooth gekoppelt sind und der TX am einem der Mppt verbunden ist.

Und eine hinreichend aktuelle Firmware, ggf. selbst gebaut, da dieses Feature noch nicht Teil des zuletzt veröffentlichten Releases ist.

schlimmchen, 2024-04-23:
Ach krass, innerhalb von einem Tag melden sich vier Leute, die das betrifft?!

@kloppy1984 @doublexminus @ferdlgogosch @Freaksplace Wer von euch hat denn einen ESP32-S3 und würde daher von der "einfachen" Anpassung in Software schon profitieren? Ich denke das kann ich kurzfristig hinpfuschen. Eine vernünftige Lösung braucht leider etwas Zeit, und die (automatische) Nutzung von SoftwareSerial, falls nötig und möglich, ist noch eine Stufe aufwändiger. Aber mit ESP32-S3 könnten wir was hinbiegen.

kloppy1984, 2024-04-23:
Ich habe einen ESP32-S3 un meiner Schublade liegen 😅

Auf Grund der Probleme, die man so liest, 
mit der Spannungsversorgung und wlan, noch nicht in Benutzung...

ferdlgogosch, 2024-04-23:
Ich habe mir heute einen esp32-S3 bestellt um für die Zukunft gerüstet zu sein. 😉
Und danke an euch für die Infos

doublexminus, 2024-04-23:
@schlimmchen
Bei mir ist einer im Zulauf ;) Wäre also schön, aber sollte niemandem den wohlverdienten Feierabend rauben ;)

Mir ist das Problem schon direkt nach dem Release aufgefallen, hatte aber leider keine Zeit weiter zu forschen :) und bin heute auf der Suche nach einem Fix auf diesen Bug aufmerksam geworden.

Ich schaue mal, welche Ecken der Doku (inkl. Pin-Mapping-Beispiele) betroffen sind und versuche mich mal an einem PR.

doublexminus, 2024-04-23:
Habe die Beschreibung hier mal um das Kapitel "Device-Profiles" mal um die Section "[Known Limitations](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Device-Profiles#known-limitations)"

schlimmchen, 2024-04-28:
[Known Limitations](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Device-Profiles) aktualisiert bzgl. #933.

closed 2024-04-29 by schlimmchen