hoylabs/OpenDTU-OnBattery #916: MPPT power calculated wrong

MalteSchm, 2024-04-24:
### What happened?

My MPPT reading (MPPT Gesamtleistung) seems to display about twice as much power. Please see screenshot below and compare the overview data with the actual Victron MPPT data 

[image]

### To Reproduce Bug

Compile from development branch 

### Expected Behavior

Display correct power

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

eb9bfd1ac62efcb28a542ce7708c0f063df48dfe

MalteSchm, 2024-04-24:
@schlimmchen 
Could this be related to this change?
https://github.com/helgeerbe/OpenDTU-OnBattery/commit/21cdc69625307a9af7e0b30128b2a34f104e5292

MalteSchm, 2024-04-25:
I use VE.direct

I did some reading and looked through the config and was able to narrow down the problem
The issue is that Victrons rx2 and tx2 are set to the same pins as rx and tx by default.

PinMapping.cpp contains:
```
            // OpenDTU-OnBattery-specific pins below
            _pinMapping.victron_rx = doc[i]["victron"]["rx"] | VICTRON_PIN_RX;
            _pinMapping.victron_tx = doc[i]["victron"]["tx"] | VICTRON_PIN_TX;
            _pinMapping.victron_rx2 = doc[i]["victron"]["rx2"] | VICTRON_PIN_RX;
            _pinMapping.victron_tx2 = doc[i]["victron"]["tx2"] | VICTRON_PIN_TX;
```

This caused both primary and secondary VE-direct input pins to be valid and both pointing to the same pin. I suspect that this means that the system reads the data twice when doing the total power calculation. The documentation states that rx2/tx2 pins should be set to -1 if not connected.

So far I used my own environment in platformio_override.ini and did not setup a pinmux config file. The way how things are configured now means that there is no way to modify rx2 and tx2 to be -1 (which would be the correct value) using defines. I therefore just created a pinmux file with this configuration. This solves the problem.

A proposed fix is here:
https://github.com/helgeerbe/OpenDTU-OnBattery/pull/920

schlimmchen, 2024-04-25:
Completed by #920.

closed 2024-04-25 by schlimmchen