hoylabs/OpenDTU-OnBattery #917: [Request] Improvement of the DPL algorith in case of shading of one of the MPPTs

greymda, 2024-04-24:
### Is your feature request related to a problem? Please describe.

Curiosly, i've noticed that the when a DPL limit is set, in my case (see the attached picture) say 104W the load is divided for the available MPPTs (HM-800 case) hence MPPT1 is limited to 52W and MPPT2 to another 52W. The expectation of the algorithm is that both MPPT can deliver, BUT this is not always the case- shades and so on.
In this particular situation, because MPPT2 can't deliver what it's being asked for (only 17W from the demanded 52W) we have grid consumption 55W (all in the picture attached)
[image]

### Describe the solution you'd like

The ask: check the real delivered energy from MPPTs and if one is much lower than the other - adjust the total DPL limit value upwards, basically what OpenDTU-OnBattery currently does when there's no input from a MPPT and scales the limit to 2x value, but this enhancement should use a variable scaling, depending on the real output of the lower MPPTs (in my case String Dois).
Hope this makes sense and you see any value from this enhancement.

greymda, 2024-04-24:
maybe in context of implementing enhancement https://github.com/helgeerbe/OpenDTU-OnBattery/issues/899 it would make sense since any change in cloudiness/etc would be reflected at a higher rate/precision.

greymda, 2024-04-25:
@schlimmchen how do you feel about this?

greymda, 2024-04-26:
Fair enough, thanks

saschaabraham, 2024-05-12:
Maybe it is posible to realise an option, to set the DLP to max limit value, if the powermeter is below zero and then decrease the inverter limit step by step to reach the wanted value. This could compensate the shading in an appropiate way. and should bounded to the solar powered option.

closed 2024-06-20 by schlimmchen