hoylabs/OpenDTU-OnBattery #921: Problems with the accessibility of an electricity meter cause interruptions in the processing of Ve.Direct

peff74, 2024-04-25:
### What happened?

Today I had a network problem with my electricity meter which is queried by the OpenDTU via HTTP(S) + JSON.
As a result, the electricity meter sometimes took a very long time to respond.
I wouldn't have noticed this at all, as the datas were arriving. However, this delay probably led to issues with reading the Victron.
This meant that the data did not appear in the live overview every second as usual, sometimes took 3-4 seconds.

### To Reproduce Bug

extend the response times of HTTP(S) + JSON

### Expected Behavior

I don't know if this is really a bug, but maybe some interesting information

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

gf0df583

SW-Niko, 2024-04-25:
Maybe same as #752 ?

schlimmchen, 2024-07-01:
Should be resolved in #1077.

closed 2024-07-10 by schlimmchen