hoylabs/OpenDTU-OnBattery #925: HM800 Disconnects and shut down?!? at Solar only, with limiter.

Solar320, 2024-04-26:
### What happened?

Hello, I have two HM-800. One is producing up to 800Wp and was limited, the second one is unlimited and has an max. of 175wp connected.

With the old DTU it happen that the 800W Inverter was shut down through the "DTU on Battery controller" and also you can see an error message at the inverter Informations which tells you the reason.

With the new Version this shutdowns should not possible.
[2024.03.23](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2024.03.23)

Now the Inverter still shuts down , the color turns yellow and at most stay in shutdown mode. It do not show anymore why it was shutdown, no additionally error messages! only "14:53:47 | 14:53:47 | 1 | Wechselrichter gestartet"

The second inverter still runs fine.

**Today this happen at full Sun in the moment when I switched on another load. Instead that the Inverter turns from low percentage to 100% it turns off! The color turns yellow, no error messages. The small inverter turns out 133W the limited 0W**

I can not provide an screenshot yet which shows it better, this screenshot shows only what happen when the inverter is in shutdown mode or not reachable(?).

Today I had the case that grid demand was over 1000W but the inverters delivered in total only the shown 133W +-20W  (only the second Inverter was active) The main Inverter switched off instead on. The other pictures shows the firmware they use, the first one is the older of the two, it is also the one which is limited and which shuts down.

The only way to bring it back to live is to manually restart the inverter over the DTU o. B. controller.

[image]
[image]
[image]

### To Reproduce Bug

My problem is that I can't provide any usefull console informations because I don't know when exactly this happen.

May be it happen in the moment when the grid demand instantly jump up, e.g. in the case when I switch on an hot plate.

So I now turn off the limiting funktion again. 

May be it is an problem with my old Hoymiles Firmware, I don't know. I can try to switch my solar panels and use the newer inverter for limiting the power.

Tomorrow I may try to reproduce the error and in paralell read the console output.

I know should be possible to redirect the serial console from VSC to an file but I don't know how at the moment.

### Expected Behavior

Two inverters are running, one is limited one is not.  Booth Inverters are read out by Open DTU on Battery and are on the same
line Phase.

The limited inverter should run from 10-100% and it should forced to never turn off. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Firmwareversion / git Hash	2024.03.23

### Relevant log/trace output

```Shell
Websocket: [/livedata][10] disconnect
13:34:29.070 > Websocket: [/batterylivedata][9] disconnect
13:34:32.943 > PowerMeterClass: TotalPower:  8.96
13:34:32.990 > Fetch inverter: 114173305377
13:34:33.124 > TX RealTimeRunData Channel: 40 --> 15 73 30 53 77 80 18 23 20 80 0B 00 66 2B 91 49 00 00 00 00 00 00 00 00 8D AB D1 
13:34:33.172 > Interrupt received
13:34:33.219 > RX Channel: 23 --> 95 73 30 53 77 73 30 53 77 01 00 01 01 3A 03 89 0B 0A 01 39 03 92 0B 23 00 09 AD | -80 dBm
13:34:33.264 > Interrupt received
13:34:33.312 > RX Channel: 75 --> 95 73 30 53 77 73 30 53 77 02 B2 5D 00 09 52 23 00 00 00 00 09 3E 13 8A 15 1D A6 | -80 dBm
13:34:33.357 > Interrupt received
13:34:33.404 > RX Channel: 75 --> 95 73 30 53 77 73 30 53 77 83 00 05 00 E4 03 E8 00 FA 00 01 D6 52 63 | -80 dBm
13:34:33.570 > RX Period End
13:34:33.621 > Yield Day reset detected!
13:34:33.667 > Yield Day reset detected!
13:34:33.714 > Success
13:34:33.761 > [DPL::updateInverter] sending limit of 69.1 % (553 W respectively), max output is 800 W
13:34:33.807 > TX ActivePowerControl Channel: 61 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 B3 00 01 FF 30 58 
13:34:33.853 > [DPL::announceStatus] waiting for a power limit command to complete
13:34:36.187 > RX Period End
13:34:38.147 > All missing
13:34:38.196 > Nothing received, resend whole request
13:34:38.242 > TX ActivePowerControl Channel: 75 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 B3 00 01 FF 30 58 
13:34:38.289 > RX Period End
13:34:38.339 > All missing
13:34:38.449 > Nothing received, resend whole request
13:34:38.551 > TX ActivePowerControl Channel: 3 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 B3 00 01 FF 30 58 
13:34:38.599 > Interrupt received
13:34:38.645 > RX Channel: 40 --> D1 73 30 53 77 73 30 53 77 81 00 00 0B 00 14 07 48 | -80 dBm
13:34:39.755 > RX Period End
13:34:39.810 > Success
13:34:39.856 > Fetch inverter: 114184855236
13:34:39.903 > [DPL::updateInverter] actual limit is 69.0 % (552 W respectively), effective 6166 ms after update started, requested were 69.1 %
13:34:39.950 > [DPL::announceStatus] waiting for sufficiently recent inverter data
13:34:39.996 > TX RealTimeRunData Channel: 23 --> 15 84 85 52 36 80 18 23 20 80 0B 00 66 2B 91 50 00 00 00 00 00 00 00 00 1D 0C FD 
13:34:40.044 > Interrupt received
13:34:40.090 > RX Channel: 40 --> 95 84 85 52 36 84 85 52 36 01 00 01 01 35 01 C1 05 6B 00 08 00 01 00 00 00 01 07 | -80 dBm
13:34:40.137 > Interrupt received
13:34:40.188 > RX Channel: 40 --> 95 84 85 52 36 84 85 52 36 02 03 D8 00 00 0F 7C 02 A7 00 00 09 44 13 8A 05 29 62 | -80 dBm
13:34:40.293 > Interrupt received
13:34:40.338 > RX Channel: 61 --> 95 84 85 52 36 84 85 52 36 83 00 01 00 38 03 E8 01 63 00 03 BC 3C 25 | -80 dBm
13:34:40.386 > RX Period End
13:34:40.432 > Success
13:34:43.023 > PowerMeterClass: TotalPower: -9.91
13:34:44.737 > Fetch inverter: 114173305377
13:34:44.795 > TX RealTimeRunData Channel: 40 --> 15 73 30 53 77 80 18 23 20 80 0B 00 66 2B 91 55 00 00 00 00 00 00 00 00 4D 33 95 
13:34:44.852 > Interrupt received
13:34:45.002 > RX Channel: 3 --> 95 73 30 53 77 73 30 53 77 01 00 01 01 36 03 A6 0B 46 01 37 03 AA 0B 5C 00 09 8B | -80 dBm
13:34:45.049 > Interrupt received
13:34:45.207 > RX Channel: 75 --> 95 73 30 53 77 73 30 53 77 02 B2 5E 00 09 52 24 00 01 00 01 09 3F 13 89 15 8C 31 | -80 dBm
13:34:45.309 > Interrupt received
13:34:45.359 > RX Channel: 75 --> 95 73 30 53 77 73 30 53 77 83 00 00 00 E9 03 E8 00 FB 00 01 C0 00 2E | -80 dBm
13:34:45.405 > RX Period End
13:34:45.451 > Success
13:34:45.616 > [DPL::announceStatus] the system is stable, the last power limit is still valid
13:34:47.254 > [DPL::updateInverter] sending limit of 71.1 % (569 W respectively), max output is 800 W
13:34:47.307 > TX ActivePowerControl Channel: 61 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 C7 00 01 E5 70 76 
13:34:47.459 > [DPL::announceStatus] waiting for a power limit command to complete
13:34:49.355 > RX Period End
13:34:49.507 > All missing
13:34:49.554 > Nothing received, resend whole request
13:34:49.602 > TX ActivePowerControl Channel: 75 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 C7 00 01 E5 70 76 
13:34:51.407 > RX Period End
13:34:51.456 > All missing
13:34:51.502 > Nothing received, resend whole request
13:34:51.553 > TX ActivePowerControl Channel: 3 --> 51 73 30 53 77 80 18 23 20 81 0B 00 02 C7 00 01 E5 70 76 
13:34:51.662 > Interrupt received
13:34:51.759 > RX Channel: 40 --> D1 73 30 53 77 73 30 53 77 81 00 00 0B 00 14 07 48 | -80 dBm
13:34:53.095 > PowerMeterClass: TotalPower: 28.63
13:34:53.436 > RX Period End
13:34:53.492 > Success
13:34:53.552 > Fetch inverter: 114184855236
13:34:53.706 > [DPL::updateInverter] actual limit is 71.0 % (568 W respectively), effective 6167 ms after update started, requested were 71.1 %
13:34:53.752 > [DPL::announceStatus] waiting for sufficiently recent inverter data
13:34:53.800 > TX RealTimeRunData Channel: 23 --> 15 84 85 52 36 80 18 23 20 80 0B 00 66 2B 91 5E 00 00 00 00 00 00 00 00 7D 40 DF 
13:34:53.910 > Interrupt received
13:34:54.014 > RX Channel: 75 --> 95 84 85 52 36 84 85 52 36 01 00 01 01 36 01 B6 05 4B 00 08 00 01 00 00 00 01 53 | -80 dBm
13:34:54.065 > Interrupt received
13:34:54.111 > RX Channel: 3 --> 95 84 85 52 36 84 85 52 36 02 03 D9 00 00 0F 7C 02 A8 00 00 09 46 13 8B 05 0A 4C | -80 dBm
13:34:54.156 > Interrupt received
13:34:54.202 > RX Channel: 75 --> 95 84 85 52 36 84 85 52 36 83 00 00 00 36 03 E8 01 63 00 03 80 07 2D | -80 dBm
13:34:54.251 > RX Period End
13:34:54.295 > Success
```

Solar320, 2024-04-27:
[image]
Kann keiner englisch?

Ich habe jetzt folgendes getan, ich habe die Wechselrichter vom Netz genommen, danach die Solaranschlüsse umgetauscht und wider eingeschaltet.
Erst ist nichts passiert und beide Wechselrichter haben wider produziert.

Danach habe ich auch in den Limiter einstellungen die wechselrichter getauscht und abgespeichert. Sofort danach wurde der kleine Wechselrichter (1. WR in der Liste ) durch die DTU abgeschaltet und so belassen. Diesmal habe ich die DTU dabei aber erwischt weil es einen entsprechenden Abschalteintrag gibt.
Das heist, den ersten Wechselrichter in der Liste erwischt es immer bei den unmotivierten Shutdowns, obwohl der momentan gar nicht geregelt wird!

[image]

Wie man am Foto sieht wird er von der Steuerung auch nicht wider eingeschaltet. Das mache ich jetzt manuell.

Wenn der Fehler dann verschwunden ist mache ich es halt dann immer so, das ich nur den zweiten Wechselrichter in der Liste regele und den ersten eben nicht.

Manos1966, 2024-04-27:
- Why do you have the Battery Option enabled? Why?

- Did you have a look at this? https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Dynamic-Power-Limiter-%28Nulleinspeisung%29-%E2%80%93-OHNE-BATTERIE#important-note-for-users-updating-their-system-from-previous-versions

Solar320, 2024-04-28:
> * Why do you have the Battery Option enabled? Why?
> 
>     * Did you have a look at this? https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Dynamic-Power-Limiter-%28Nulleinspeisung%29-%E2%80%93-OHNE-BATTERIE#important-note-for-users-updating-their-system-from-previous-versions

[image]

????????????????????where do you see an enabled battery option?   There is no Battery Option if you tick on the "solar powered Inverter" option! I have the recent Version of Open DTU on Battery.   I give Up with you....I will search now by myself. May be I install an battery this year and then I hope the inverter shut down will be irrelevant.

Manos1966, 2024-04-28:
> I give Up with you....I will search now by myself.

I think it is a good choice. Good luck with your search 👍 

ALthough, if I were you, I would stop playing the "beleidigte Leberwurst" and start wondering why I see the Battery Information on my Live-Daten... 

But, as you said yourself, I give up with you, too

[image]

schlimmchen, 2024-05-03:
> Danach habe ich auch in den Limiter einstellungen die wechselrichter getauscht und abgespeichert. Sofort danach wurde der kleine Wechselrichter (1. WR in der Liste ) durch die DTU abgeschaltet und so belassen.

Wenn du in den Einstellungen des DPL einen anderen Wechselrichter wählst, dann wird der zuvor geregelte abgeschaltet.

> Kann keiner englisch?

Doch, schon, aber deine Fehlerbeschreibung ist leider nicht verständlich. Also ich begreife nicht, was dein Problem ist.

Der DPL kann nur einen Wechselrichter steuern.

closed 2024-05-03 by schlimmchen