hoylabs/OpenDTU-OnBattery #926: DEBO OLED 2.42

lopo2005de, 2024-04-27:
### Is your feature request related to a problem? Please describe.

Erst einmal ganz vielen Dank für OpenDTU!!!
Das kleine 0.9 OLED ist wirklich sehr klein.
Ich würde gern das DEBO OLED 2.42 von JOY-IT anschliessen, per I2C sollte das kein Problem sein.
Bei der gleichen Auflösung müsste ich vermutlich die Schriftgrösse gar nicht ändern.
Nur:
JOY-IT verwendet den Treiber SSD1309.
Muss der noch in das System eingepflegt werden oder wird so etwas per pin_mapping.json eingerichtet? 

### Describe the solution you'd like

DEBO OLED 2.42 von JOY-IT

spcqike, 2024-04-27:
https://github.com/helgeerbe/OpenDTU-OnBattery/blob/4bc4defe665946d1c096206fb70682e31a004dd0/docs/DeviceProfiles/wt32-eth01.json#L54

Gibt es scheinbar bereits. Musst du nur richtig einstellen.

lopo2005de, 2024-04-27:
Ist voll cool!!! Mit Ethernet würde mir auch noch gefallen!
Also: bestellen :)

ingrimsch, 2024-05-02:
Ich habe ein SSD1309 2.42" OLED seit einiger Zeit laufen - funktioniert absolut problemlos. :)

Einfach als Display Type "4" einstellen und die korrekten Pins für DATA und CLK auswählen.

schlimmchen, 2024-05-03:
Meiner Einschätzung nach ist hier von Seiten OpenDTU-OnBatery nichts zu tun. Schreibe wieder, falls dem nicht so ist.

closed 2024-05-03 by schlimmchen