hoylabs/OpenDTU-OnBattery #930: Web-API http://192.168.xxx.xxx/api/livedata/status liefert nicht das erwartete Ergebnis

Manos1966, 2024-04-28:
### What happened?

As in the example under https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Web-API
the expected answer from http://192.168.xxx.xxx/api/livedata/status should be:

[image]

Instead, I see only as answer:
[image]

I do not receive the information 
  "AC": {
        "0": {
          "Power": {
            "v": 0,
            "u": "W",
            "d": 1
          },
etc, etc, etc,

Full detail of the WEB-API answers I receive are here:
https://pastebin.com/zmKTC5ra
https://pastebin.com/SsSF5KVA

### To Reproduce Bug

Do a  http://192.168.xxx.xxx/api/livedata/status 
request and look at the results

### Expected Behavior

The answer should be as the example described in the wiki
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Web-API

[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.03.23

spcqike, 2024-04-28:
Ich frage mich grade …. Wenn der Code vor vielen releases geändert wurde, das Wiki aber nicht aktualisiert wurde …. Ist es dann ein Bug? Oder ist das Wiki outdated?

I’d say: it’s a feature, not a bug.

Manos1966, 2024-04-28:
Ist es normal, dass keine ; am Ende dieser Linien steht?

		dict = {url: "http://" + configs[i].controllerIp + "/api/livedata/status", index: i}
		Timer.set( 2500, true, controllerTimer, dict);
		HTTPServer.registerEndpoint( "pwr" + (i+1) , VirtualPowerMeterReadings, i)
	}

schlimmchen, 2024-04-28:
> The answer should be as the example described in the wiki

No, the wiki should reflect the actual current style of implementation. This changed a while ago. The inverter details can be queried at an inverter-specific (serial?) API URL.

@Manos1966 Would you like to do this yourself? Do you need more info to understand the changed behavior?

Manos1966, 2024-04-28:
@schlimmchen Ich werde es probieren, aber danach muss du ein Auge drauf werfen.

Weiss du welche Meldung bei den Releases-Liste das bekannt gegeben hat?

Ich habe das gefunden, bin ich aber nicht sicher ob das richtig ist.
https://github.com/helgeerbe/OpenDTU-OnBattery/commit/054a67757525919dd44b828c1be7790a0415daf1

schlimmchen, 2024-04-28:
https://github.com/tbnobody/OpenDTU/releases/tag/v24.2.12

Genauer: https://github.com/tbnobody/OpenDTU/commit/c27ecc3620f656fbdfaa7e9f0f3deca6a371a373 (erster Link im Changelog des o.g. Release).

Manos1966, 2024-04-28:
@schlimmchen wirf ein Auge (bei Gelegenheit) bitte (und gegebenfalls anpassen 😄 ) 
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Web-API

closed 2024-04-28 by Manos1966