hoylabs/OpenDTU-OnBattery #935: [Request] Drop support for devices with 4MB flash (?)

schlimmchen, 2024-04-29:
I'm exploring the possibility of discontinuing support for devices with 4MB flash. Do you think increasing the minimum to 8MB would pave the way for new features or advantages in the near future? It might be worth communicating this potential change widely, such as in changelogs, the wiki, GitHub page, and other prominent spaces. For example: "Starting January 1st, 2025, ODTUoB will no longer support 4MB flash. Further details can be found here."

Personally, I believe phasing out support for these smaller devices could catalyze further development. What are your thoughts?

Best regards,

_Originally posted by @swingstate in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/923#issuecomment-2083397730_

schlimmchen, 2024-04-29:
Copying my own comment from #923:

> I just scrolled Wikipedia and now I think that requiring ESP32 with at least 8MB of flash will be very harsh. The default is 4MB, and most people will end up with such a device, and they will be frustrated to find out that they can't use such a board at all.

It is my personal interest and motivation to help make this project fun to use for everybody. Even though a voice in my head says "screw it, let's go for 8MB+ devices in the future", another voice goes "don't risk people becoming frustrated with this project because we decided to drop supporting their setup".

@swingstate idea to plan this in advance could mitigate some frustration.

spcqike, 2024-04-29:
Another possibility could be to provide different builds like tasmota does. One for NRF, one for CMT, one with oled or without, with ve.direct or without,…. Depending on what parts need much diskspace. 

As for me, right now I only use a CMT and solar only. No battery, no Victron, no BMS. All this code and libraries could be kept out in my case. 

Or maybe one could also use the custom „define“ tags like Tasmota does. So everyone could build a firmware as needed. 

Do we have an overview of which part needs how much space?

Manos1966, 2024-04-29:
1. Is the memory issue related to the amount of SW modules that are loaded?
For example, I see the standard version includes:
- Victron MPPT
- CANbus
- JKS
- Huawei

Personally I use Victron MPPT and CANbus for my test systems thus, (for example): Would removing JKS and Huawei resolve to (significant) less memory used?

2. An  ESP32 with 8MB of flash. What is this animal, where can I find it today, **what is the price difference compared to the standard ESP32 with 4Mb?** 
Every time (year 2024) I do a search for an ESP32 with 8Mb, I get more confused.
But, this can change in the future... in fact, I am sure it will.
So, it is a question of timing: When will the ESP32.8Mb become more common in the market (than the ESP32.4Mb)?
Maybe within 12 months? (wild guess from my part).
I got my first Hoymiles September 2022. 
February 2023 I had my first DTU, running on an ESP8266
Today (April 2024), nobody bothers to talk about ESP8286 any more.
So, it took approx. 16 months to change a generation.

Assuming that the price difference between ESP32.4Mb and ESP32.8Mb, will be similar to the price difference between ESP8266 ans ESP32, I would dare say that, within 12 months the majority hardware will be ESP32.8Mb

So, as swingstate pointed out, I believe that supporting a 4Mb version during 12 months will be more than enough.

SW-Niko, 2024-04-29:
Any change to improve OTA?
In a way to blow up the boot loader with more functions to avoid to store 2 compete DTU images in the flash?

greymda, 2024-04-30:
I am taking a veeeeery wild guess but i assume the ratio of users on 8Mb flash is small, very small. I'd say it's good to start spreading the idea, but to cut the support now would be very noisy. My 2 cents, obviously.

P.S. owner of three 4MB ESPs lol

spcqike, 2024-04-30:
> That is something that I think is luxurious. If corruption occurs, which certainly happens very seldom, we could require users to re-initialize using the bootloader and USB instead. What do you think about that?

good point.

Do you think it would be possible, to have something like a minimalistic second firmware, that only connects to wifi (or starts AP) and let you re-upload / OTA to the desired firmware? something like "tasmota-minimal", which is only 262Kb. 

i think, some users may have trouble hardware-flashing their boards, which they bought online. so a backup firmware makes sense, just in case. but maybe it mustn't be a complete copy of the firmware but just a minimal one that allows another OTA attempt.

schlimmchen, 2024-04-30:
Yes, agreed. However, I have little hope that this minimal Firmware will be indeed very small. And again: Maintaining it and building it is probably a headache. And it needs a minimal WebApp as well, another headache. I like this idea as well, but it's a pain. And will this minimal firmware ever be updated? If not, when does it stop working in some setups because of newer technology or WLAN encryption and such. The main firmware will be updated and stay compatible. The minimal firmware is left as is, and when the day comes and the flash process was not successful, the minimal firmware does start, but cannot connect to the WiFi? Lame.

> i think, some users may have trouble hardware-flashing their boards, which they bought online.

Notice that changing the partition layout will require flashing the factory firmware bin using the USB interface in any case. If we think we need to avoid this, we are in trouble, as touching the partition layout is suddenly not an option.

Let me use this occasion to introduce a new feature. It's not "officially announced" yet, but @helgeerbe and I gave approval and it is online. However, I did not even test it yet. https://solar.metacontrol.eu/opendtu-onbattery-webflasher/ Given that this web flasher is available, there is very little excuse to not update an existing ESP32. Do you agree?

swingstate, 2024-04-30:
Agree. Great to see this web flasher, one less headache.

Manos1966, 2024-04-30:
As I also asked...

Please, one AMAZON Link and one from Asia, if possible.

Matti07, 2024-04-30:
> any links, preferably international?
Unfortunately not international:
https://www.berrybase.de/espressif-esp32-s3-devkitm-1-n8-esp32-s3-mini-1-8mb-flash
Just ask google
"ESP32-S3-DevKitM-1-N8"

swingstate, 2024-04-30:
> As I also asked...
> 
> Please, one AMAZON Link and one from Asia, if possible.

[Link](https://www.amazon.de/Espressif-ESP32-S3-DevKitC-1-N8R2-Entwicklungsplatine/dp/B09MHP42LY/ref=sr_1_3?__mk_de_DE=ÅMÅŽÕÑ&crid=30EKZHVLHNJZR&dib=eyJ2IjoiMSJ9.aH-JDEqlBa1iP4tTsrbFFTLwDwXiYaktHeH9-DbuJKX4TwALqvrv8ML6_EhDDnHPy9kk6A8WYdTDEPmkurCd9wbmDAhL6MHrQ-d8DZrqtmN7hK6puqKWdJk90b0GsA6uco10Tx975CwsTe_mCYwWU7ZhaHb2JXE9MuS40tM_seAngsBbAJVX0pcJ62TFSe5ieBT6GsL_avzAvRudzAZTScbjuOHxQbPFuN0JivQC3xk_hS9oP19brURIOLMSKnZU-havePk0VC5-2l3vhqdpGRStitBal5xo0DaOnnmrvuo.AM9NAk4eMiFZ1bSwNJwSIhdM9gjF7oqQu1grM7YSbu4&dib_tag=se&keywords=ESP32-S3%2Bn8&qid=1714504470&s=ce-de&sprefix=esp32-s3%2Bn8%2Celectronics%2C141&sr=1-3&th=1)

[Link](https://de.aliexpress.com/item/1005006266375800.html?spm=a2g0o.productlist.main.1.22fe7d64kr72Ch&algo_pvid=371a7f96-3197-4195-96fa-2846de69128d&algo_exp_id=371a7f96-3197-4195-96fa-2846de69128d-0&pdp_npi=4%40dis%21EUR%2119.36%216.57%21%21%21146.70%2149.77%21%40211b441e17145045761451658eaa4e%2112000036532657519%21sea%21DE%212051066232%21&curPageLogUid=I6NAAxdLmqcI&utparam-url=scene%3Asearch%7Cquery_from%3A)

spcqike, 2024-04-30:
If one wants to cancel 4Mb in the long run, does it make sense to use a specific module with 8Mb or more? Like there is N8, N8R2, N8R8, N16R8 and maybe more. Where Nx is the flash and Rx the PSRAM. Is it advised to use N8R8 with 8Mb each or is N8R2 (or even N8) ok to use?

Manos1966, 2024-04-30:
Thanks swingstate! These are the numbers I was loooking for!
5,50EUR for a 8MB version
5,80EUR for a 16MB version 🤩 
(plus 2EUR transport cost)

@schlimmchen if the prices are at this level already, why not go straight for the 16MB model?
So, the only question should be "how many months should the 4MB be supported before End-of-Life?"
- My experience from product Life-cycles would say 12 months.
- Probably you could dare announce "End of 2024" (8 months).

[image]

eu-gh, 2024-05-01:
> I am taking a veeeeery wild guess but i assume the ratio of users on 8Mb flash is small, very small. I'd say it's good to start spreading the idea, but to cut the support now would be very noisy. My 2 cents, obviously.

I'd second that, since almost all popular ESP32 devboards all only feature 4MB (without even mentioning it). 
> P.S. owner of three 4MB ESPs lol

Times three here :rofl: Which I would replace if it was the only way of keeping up with recent versions, but it would make my heart bleed rendering these capable little workhorses obsolete. Also, I'm certain the pinout will change again and I have to resolder everything :tada: 
> out of curiosity tried to find such a board (8 MB) and not successful. any links, preferably international?

If you don't like waiting, there are several offers shipped by Amazon for around 8€ per piece, search term [esp32-s3 n16r8].
> That is something that I think is luxurious. If corruption occurs, which certainly happens very seldom, we could require users to re-initialize using the bootloader and USB instead. What do you think about that?

From the very beginning, I was really amazed by the level of refinement that went into this (and, of course the original openDTU) project, which puts a lot of commercially developed products to shame. On the other hand, the main target group are (and probably always will be) tech-savvy people, which should be either able to help themselves, or to search for solution online. Using a wired connection for every update would be kind of a bummer, but for recovery purposes, it's totally justified, if not industry standard.
> I really dont like the idea of adding config switches to the firmware to disable functionality. The overhead, especially when maintaining, is a real headache. The amount of different firmware versions will explode very fast, and there will be a at least one that cannot compile for different reasons for most of the time.

A thought I just had was the usage of a modular package system, as implemented by other projects with similar problems, like OpenWrt. I'm not even sure if such thing is possible on the ESP platform, and that this is an absolutely humongous task that also introduces many other challenges, but it could keep the overall flexibility, future development and hardware limitations in some sort of balance.

swingstate, 2024-05-03:
@schlimmchen, do you have any other thoughts?

What do you think about the idea of freezing the current development state (the timing of which is yet to be determined) that is compatible with the 4MB boards? This frozen state would only receive maintenance efforts on a best-effort basis with lower priority, and no new features would be added. Instead, the focus would shift to the 8 or 16MB boards. We would communicate this 'breaking change' 3 to 6 months in advance."

gitisgreat2023, 2024-05-04:
Is this only about the amount of flash, or is the availability of PSRAM also a reason? It would be good to communicate this from the beginning, when the 4MB support is dropped. Whether the baseline and foreseeable future also makes use of the PSRAM or not. There are 8 and 16MB flash versions with 0 and 0,2 and 8MB of PSRAM. Which one is the preferred baseline?

[image]

The typical cost of a N16R8 (that is 16MB of flash, 8MB of PSRAM) is about 6€ at aliexpress. (note using bank switching more than 4MB of PSRAM can be used, that's the himem feature). So without understand all the software issues, wouldn't it be better to go for a "non-zero" PSRAM baseline S3 model?

schlimmchen, 2024-05-04:
# Interim Conclusion

## Flash

The consensus actually seems to be to drop support for devices with 4MB flash. This must be announced beforehand. Giving a time-frame, however, is tricky. Let's say we manage to keep the 4MB flash devices alive for now, but a new feature from upstream comes along which pushes the 4MB over the edge. What then? I would like to make *no* promises. Do the announcement and urge people to upgrade their hardware, as future releases may stop including builds for devices with 4MB flash at any time, depending on whether or not new firmware fits the old flash layout. No effort will be made to reduce the binary size (which becomes more and more time-consuming to achieve, if at all possible).

I don't see an item in the web UI that tells me how much flash my ESP32 has. This should be available somewhere. Otherwise, some (few) people will buy new hardware even though they might not need to and some (many) will ask how they know whether or not they need to upgrade. Ripping the ESP32 board out of their setup to connect the ESP32 using serial and use `esptool` is not a viable option.

## PSRAM

Let's keep this optional. I guess it makes sense to explain people that any and more PSRAM is better, and that it makes little difference in the price, but might make a big difference regarding how many features they can use at once. Example: the prometheus web API endpoint might need so much memory, depending on the amount of features enabled, that it becomes difficult to serve it reliably. I suppose that even without PSRAM, most users should be good to go.

## ESP32-S3

If we start urging people to use a particular ESP32, they should choose an ESP32-S3 with the USB-port directly connected to the USB-subsystem of the ESP32-S3. That way, 3 hardware UARTs are available.

## Community Boards: OpenDTU FUSION and @swingstate's PCB (which other relevant boards are there?)

I asked people from AllianceApps, who confirmed that all OpenDTU FUSION boards (at least starting from v2.x) are equipped with 8MB flash (and 2MB PSRAM, to be precise) at least.

As far as I can tell, @swingstate's PCB uses an ESP32-S3 with 16MB flash. I guess it is early enough to make him use N16R2 (2MB PSRAM) modules (or N8R2).

# Proposed Next Steps

1. Relieve urgency in dropping support for 4MB flash devices by removing the FirebaseJSON lib from the firmware (this is the last comparatively easy thing we can do to save some flash -- that I know of).
2. Announce that support for 4MB flash devices will be discontinued in the near future.
3. In that announcement, add high quality information on how to obtain a supported and future-proof ESP32 board.
4. Point to OpenDTU FUSION and @swingstate's PCB in particular.
5. Once the day comes that 4MB flash devices will not receive the next update, announce the breaking change: The next release will only work on 8MB flash devices and **all** users must upgrade their partition layout (use USB connection, backup config, blabla... all will be explained in detail).

Can I get some thumbs up or criticism? @helgeerbe What are your thoughts on this? Are you prepared to upgrade your hardware or would it annoy you :wink: (or are you on a 8MB ESP32?)

greymda, 2024-05-04:
just to get a confirmation, is this a viable option?   WROOM-1-N16R8 ESP32-S3-DevKitC-1 module 
https://sl.aliexpress.ru/p?key=U2Hrslf

swingstate, 2024-05-04:
Agree to the steps @schlimmchen shared above. My PCB is based on the ESP32-S3-WROOM-1-N16R8, 16MB Flash / 8MB PSRRAM. I should also hear from TÜV next week on next steps. Would like to progress faster with the board evaluation but doing a CE cert right is a pain for a single man show. 

Regarding point #2, I'd suggest to announce the drop of 4MB for a certain quarter in the future, for example Q4. That should provide enough time for users to get a new ESP. Entering the quarter we should have visibility on which release would be the "final" for the 4MB boards.
Also, having the basic hardware specs being displayed in the WebGUI in an easier to read way (the system page is rather technical) would make a lot of sense. Maybe the GUI should also show a note on 4MB devices that support will end.

swingstate, 2024-05-04:
> just to get a confirmation, is this a viable option? WROOM-1-N16R8 ESP32-S3-DevKitC-1 module https://sl.aliexpress.ru/p?key=U2Hrslf

should work, however looks like the PINs are not soldered. Suggest you get a board with the PINs already soldered unless you are keen to do that and have the equipment.

schlimmchen, 2024-05-04:
> just to get a confirmation, is this a viable option?

I can't read that, it Russian, dude :grin: From what I see, it fits. Please try to find out whether or not there is an additional UART<->USB transceiver on there, or if the USB lines of the ESP32 are directly connected to the USB plug. The latter is preferred such that the third hardware UART is available for use.

greymda, 2024-05-04:
oh, i’m using the english version of the ali, have no idea why it copied me the russian one. apologies!

will try finding that but the description is usually pretty poor on ali listings.

eu-gh, 2024-05-04:
> future releases may stop including builds for devices with 4MB flash at any time, depending on whether or not new firmware fits the old flash layout.

So changing the flash layout to not include a second full-size OTA partition is no longer an option?
> The latter is preferred such that the third hardware UART is available for use.

Would this open up the possibility to use a CMT2300 module and the Huawei CAN interface on a single board? Because that would at least cut my need for new ESPs in half, if 4MB support gets dropped for real.

schlimmchen, 2024-05-04:
> So changing the flash layout to not include a second full-size OTA partition is no longer an option?

Exploration is required to determine the feasibility. Currently I think that it will not be possible for a simple reason: As far as I can tell the ESP32 executes instructions from flash (on a full-sized computer, instructions are copied to RAM and executed from there. since this cannot be the case on an ESP32, my best guess is that ESP32 executes code in-memory (flash)). In turn this means: Without complicated locking mechanisms or without complicated "copy some instructions to RAM and execute from there", we are simply not able to update the currently running firmware in-situ.

If you show us how this actually could work without significant implementation effort, we can reconsider this. Until then, I regret putting this option on the table, for the fact that I think to have underestimated the overhead. Also: I underestimated the willingness of the most active users to throw support for 4MB flash devices out the window.

Even more so: I know that a new project exists whose author aims to *only* support 8MB+ flash devices from the very beginning. The overall trend seems to be that 4MB ESP32 are simply outdated.

> Would this open up the possibility to use a CMT2300 module and the Huawei CAN interface on a single board?

Nope. That's an SPI issue, not UART.

Manos1966, 2024-05-04:
Wink mit dem Zaunpfahl 😇 

[image]

Manos1966, 2024-05-05:
OK, I added a discreet comment about the future need of ESP32s with larger memory:

https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/ESP32-Versions-and-Memory

poolcat4711, 2024-05-05:
I just ordered and will update all my boards with this one --> https://www.lilygo.cc/en-pl/products/t-eth-lite  (cheaper then ali). So hell yeah, up the compatibility matrix and refrain from the nonsense work making it fit.

schlimmchen, 2024-05-05:
That thing looks nice, however, how is that receiving software? Does one need to connect an external TTL-UART<->USB adapter to flash the initial software?

poolcat4711, 2024-05-05:
> Does one need to connect an external TTL-UART<->USB adapter to flash the initial software?

Yep, exactly (1Euro ish @ ali). Its a one-time procedure - same thing like the WT32-ETH01 init (which currently services opendtuob here). Just connect 5 cables, flash, done. And BTW - thanks a million for the amazing webflasher that I'll give a first shot then =)

greymda, 2024-05-05:
is the web flasher limited to german ips? it’s unavailable for me, for whatever reason.

greymda, 2024-05-06:
@MetaChuh as per the usual, it was not working for days and now it is 😀

Solar320, 2024-05-29:
Hallo, ich habe ein paar von den Dingern. Die Variante im ersten Bild mit dieser Antennenform auf dem im unteren Bereich P2N8
draufsteht hatten bei mir einen Auth error beim Verbinden mit dem Wlan netzwerk.

[image]

Die Version hier , mit der selben Antenne und im unteren Bereich aufgedruckten MON16R8 scheint zu funktionieren.

[image]

Die folgende Version mit einem anderen Antennensystem funktioniert bisher am besten.

[image]

Ich denke das die fehlerhaften Versionen inzwischen ohnehin nicht mehr verkauft werden. Die hatte ich ende 2023 gekauft.
Nur mal so zur Info, falls jemand ein ähnliches Problem damit hat.

Solar320, 2024-05-30:
Ok, da muss ich mal suchen ob ich was dazu finde.  Auf die schnelle finde ich hierzu nix bei Github. Ich vermute das das eine art Mainboard ist auf dem die ESP-S3 draufpassen und diverse ADUM chips für Canschnittstellen und außerdem die jeweiligen Funkmodule usw.....

schlimmchen, 2024-05-30:
Du hast *nichts* gefunden?! :astonished: 

https://shop.allianceapps.io/products/opendtu-fusion-community-edition
https://www.opendtu.solar/3rd_party/opendtu_fusion/

Mit diesem Board wirst du jedenfalls wenig Frust haben, weil die einfach funktionieren, insbesondere der RF Teil.

Solar320, 2024-05-30:
Naja nicht direkt bei Github. Über google jetzt schon ;)

Das Board sieht sehr gut aus und ist auch relativ günstig, allerdings habe ich schon NRF24 und CMT Module gekauft.
Auch mehrere ESP32-S3 N16R8 habe ich schon, welche die auch funktionieren ;)  

Meine DTU existiert derzeit noch als Drahthaufen auf dem Tisch, da ich es noch nicht geschafft habe ein 3D Druck Gehäuse dafür
zu machen. (zu viele Nebenprojekte.)

[image]

Das ESP32-S3-WROOM-1U scheint sogar N16R8 zu haben, wenn es davon nicht noch weitere Varianten gibt.

Mal sehen, vieleicht kaufe ich so ein Board um mir den ganzen Verdrahtungsaufwand und vor allen Platz im Gehäuse zu sparen.
Die ganzen IPEX zu SMA kabel habe ich bereits daheim, auch die passenden Antennen. Ich müsste nachher nur schauen ob auch eine Verbindung zu dem Victron MPPT 100/20 direkt vom Board aus möglich ist, incl. Potentialtrennung. Links unten auf dem Board scheint mit da ein verdächtiger Anschluß zu sein. So, muss erstmal weg. Mfg Carsten

schlimmchen, 2024-05-30:
> Meine DTU existiert derzeit noch als Drahthaufen auf dem Tisch

Genau deshalb gibt es die FUSION Platine.

Breadboards bzw. diese Kabelbrücken sind super zum Entwickeln bzw. Evaluieren, aber totaler Mist auf Dauer, weil das einfach nicht zuverlässig ist.

> Ich müsste nachher nur schauen ob auch eine Verbindung zu dem Victron MPPT 100/20 direkt vom Board aus möglich ist, incl. Potentialtrennung.

Dafür gibt es #955. Oder auch jetzt schon [im Shop mit nur einem ADUM1201](https://shop.allianceapps.io/products/opendtu-fusion-can-iso-shield-beta).

DonJohnLong, 2024-05-30:
Hab bei mir letztens auch auf einen S3 N16R8 umgebaut, um für die Zukunft gerüstet zu sein.

![20240530_185752](https://github.com/helgeerbe/OpenDTU-OnBattery/assets/130904315/47a785e8-2527-49c6-8c0e-e9a38410a204

Ich finde die FUSION Platine auch ziemlich geil, aber den Bastelspass, wollte ich mir dann doch nicht entgehen lassen.

DonJohnLong, 2024-05-30:
Ich hab bei mir letztens auch auf eine S3 N16R8 umgerüstet, um für die ODoB Zukunft gerüstet zu sein. 😄 

[image]

Die Fusion Platine finde ich auch ziemlich geil, würde aber persönlich den Bastelspass vermissen.

Solar320, 2024-05-31:
> Genau deshalb gibt es die FUSION Platine.
>
> Breadboards bzw. diese Kabelbrücken sind super zum Entwickeln bzw. Evaluieren, aber totaler Mist auf Dauer, weil das einfach nicht zuverlässig ist.
>
>
> Dafür [...]

Ja das habe ich mitbestellt, leide rist das noch die alte Version, die neue hätte CAN und einen extra seriellen. Hauptsache die Serielle ist darauf isoliert, für die anderen Schnittstellen könnte ich eigene Isolatoren verwenden wenn ich die mal brauchen sollte.

schlimmchen, 2024-05-31:
Du hast eine Sache missverstanden: Die jetzt schon angebotenen CAN/ISO shields haben in der Tat nur einen ADUM1201 für VE.Direct, aber sie haben sehr wohl schon einen CAN Transceiver drauf.

Solar320, 2024-06-01:
Hab ich jetzt auch gesehen.   Bei der V2 version sollte aber dann jede schnittstelle auch einen eigenen Isolator haben, gerade bei so sachen wie Batteriewechselrichtern wo durchaus hohe potentialunterschiede auftreten könnten bzw. bei Potentialunterschieden zwischen verschiedenen angeschlossenen Geräten ist das ganz praktisch und man erspart sich viel ärger. Na mal sehen wann die Platine ankommt, dann muss ich mich halt mal aufraffen und noch ein Gehäuse drucken.

Matti07, 2024-06-01:
> Breadboards bzw. diese Kabelbrücken sind super zum Entwickeln bzw. Evaluieren, aber totaler Mist auf Dauer, weil das einfach nicht zuverlässig ist.

@schlimmchen
Völlig richtig. Ich will demnächst mal diese Teile ausprobieren:
**Permanent PCB Breadboard mit 830 Kontakten, schwarz**
https://www.berrybase.de/permanent-pcb-breadboard-mit-830-kontakten-schwarz
Liegen hier schon rum, aber noch keine Zeit gehabt ...

schlimmchen, 2024-06-01:
Du solltest dir das Basteln nicht verbieten lassen oder madig reden lassen, wenn du Spaß dran hast. Aber lass mich nochmal auf das [OpenDTU Fusion Board](https://shop.allianceapps.io/products/opendtu-fusion-community-edition) hinweisen, was dir viel Zeit ersparen kann -- wenn du das möchtest.

Matti07, 2024-06-02:
Sorry, hatte ich mißverständlich formuliert. Die **Permanent PCBs** sind für meine anderen Basteleien (Multikopter, Lichtsteuerung etc. pp.) vorgesehen.
Für so wichtige Projekte wie OpenDTU-OnBattery wären die für mich keine Option.
Das OpenDTU Fusion Board habe ich schon, ich warte nur noch auf das CAN/Iso Shield.

schlimmchen, 2024-06-10:
Thanks to everyone for their respective contribution to this discussion, helping to reach a decision which seems to be widely supported. The announcement is in #1025.

closed 2024-06-10 by schlimmchen