hoylabs/OpenDTU-OnBattery #943: Regelschwingen unter bestimmten Konditionen

uli-bs, 2024-05-01:
### What happened?

Die DTU überwacht 3 Inverter (HMS 1200, 600, 300)
An dem 1200er und dem 300er hängen ohne Steuerung Module, das funktioniert soweit auch gut
An dem 600er hängt ein 24V 100Ah Akku und der macht Nachts zeitweise Probleme.

Eingestellt ist ein Bezug von 40W mit 30W Hysterese, sodass der Bezug, solange der Akku mitspielt, zwischen 10W und 70W schwanken sollte, die Abschaltspannung beträgt 24,1V

Der vom Stromzähler gemeldete Verbrauch wird deutlich überkompensiert und führt zu Regelschwingungen, der WR wird dabei teilweise außerhalb der Spezifikationen (600W) betrieben.

Normalerweise laufen der nachgeschaltet Shelly (obere Kurve -rot-) und die DTU( obere Kurve -blau-) recht synchron, Im Fehlerfall passiert aber, was auf dem Bild zu sehen ist. Die untere Kurve ist dabei der Stromzähler.

[image]

Mit einer anderen Zeitauflösung sieht das so aus:
[image]

### To Reproduce Bug

Betrieb am Akku

### Expected Behavior

Änderung des Regelverhaltens

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	2024.03.23

spcqike, 2024-05-01:
Siehst du irgendwo die Effizienz des wecgseleichters? Ich vermute, da dies bereits mehrfach mit 24V Akkus aufgetreten ist, dass der HM unsauber regelt und sich verhaspelt. Dabei eine Effizienz von 40-60% berechnet. Und damit falsch misst bspw einspeist.

Feldsalat, 2024-05-02:
Bei 24V Systemen läuft dir die Leistung oberhalb von ca. 200W Sollwert je MPPT immer davon! Wenn du auf 400W max begrenzt, hört das gezeigte Verhalten auf. Habe ich mit HM300 ,400 und 1200 in gleicher Weise.

uli-bs, 2024-05-02:
@spcqike: Auf den Wirkungsgrad, bzw. ob der absinkt kann ich mal achten...

@Feldsalat: Bisher habe ich das Limit auf 480W (2 x 10A x 24V) gesetzt, da auf 400W zu gehen, macht den Kohl nicht fett, probiere ich aus...

poolcat4711, 2024-05-04:
Falls er mal wieder tillt, öffne mal die konsole und lass sie mitlaufen. Das kannst du dann bitte hier einstellen. Ich kenne dieses extreme überschwingen, da ich meinen "powerlimiter" selbst geschrieben habe. Der Grund fürs überschwingen war bei mir, dass die änderung am wr noch nicht beim zähler angekommen ist und es sich deswegen hochgeschaukelt hat.

markusdd, 2024-05-05:
ich würde hier jetzt rein vom draufgucken auch auf eine recht simple fehlende bzw zu späte Rückkopplung tippen.

1-am-r00t, 2024-05-06:
könnte https://github.com/helgeerbe/OpenDTU-OnBattery/issues/931 eine potentielle lösung für das problem sein bzw dazu beitragen?

uli-bs, 2024-05-06:
Bisher ist das nicht wieder so extrem aufgetreten, allerdings hatte ich die Max.leistung auch, wie von @Feldsalat empfohlen, auf 400W gedrosselt und auch erst heute ist wieder soviel Ladung im Akku, das es auftreten könnte - mal sehen. 
Von der Sache her ist es mir auch egal, wann die Akkuleistung verbraucht wird (weniger über längeren Zeitraum, oder mehr und kürzer, nur möglichst nicht unnütz).

DrSpaltenstein, 2024-05-07:
Moinsen, ich habe ein ähnliches Problem, meiner (800er) scheint immer bei ca 297,7W hängen zu bleiben. Ich weiß aber das ich in dem Moment vielleicht nur 150W verbrauche. Mein Zähler hat eine Rücklaufsperre, zeigt daher keine negativen Werte an. Target ist bsp 20W, hyst 1W also müsste er ja trotzdem in der Lage sein eine Differenz zu bilden und abzuziehen.
Hin und wieder schafft er es dann und regelt sich auch sauber ein. Aber so halbwegs stabil läuft es nur wenn ich einen hinreichend großen Netzbezug (50W oder so) vorgebe, aber das ist ja genau nicht Sinn der Sache.
Stromzähler wird über tasmota uns JSON ausgelesen.

Anbei mal das log.

13:53:36.216 > [DPL::loop] ******************* ENTER **********************
13:53:36.740 > [DPL::calcPowerLimit] power meter: 0 W, target consumption: 20 W, inverter output: 281 W, solar power (AC): 0
13:53:36.804 > [DPL::calcPowerLimit] match power meter with limit of 261 W
13:53:36.935 > [DPL::setNewPowerLimit] input limit: 261 W, lower limit: 20 W, upper limit: 800 W, hysteresis: 1 W
13:53:36.995 > [DPL::setNewPowerLimit] inverter max: 800 W, inverter is producing, requesting: 261 W, reported: 272 W, diff: 11 W
13:53:37.138 > [DPL::updateInverter] sending limit of 32.6 % (261 W respectively), max output is 800 W
13:53:37.962 > [DPL::updateInverter] actual limit is 32.0 % (256 W respectively), effective 2030 ms after update started, requested were 32.6 %
13:53:42.805 > PowerMeterClass: TotalPower:  0.00

13:53:44.499 > [DPL::loop] ******************* ENTER **********************
13:53:45.125 > [DPL::calcPowerLimit] power meter: 0 W, target consumption: 20 W, inverter output: 297 W, solar power (AC): 0
13:53:45.227 > [DPL::calcPowerLimit] match power meter with limit of 277 W
13:53:45.286 > [DPL::setNewPowerLimit] input limit: 277 W, lower limit: 20 W, upper limit: 800 W, hysteresis: 1 W
13:53:45.435 > [DPL::setNewPowerLimit] inverter max: 800 W, inverter is producing, requesting: 277 W, reported: 256 W, diff: 21 W
13:53:45.536 > [DPL::updateInverter] sending limit of 34.6 % (277 W respectively), max output is 800 W
13:53:48.099 > [DPL::updateInverter] actual limit is 34.0 % (272 W respectively), effective 4091 ms after update started, requested were 34.6 %
13:53:53.145 > PowerMeterClass: TotalPower:  0.00

13:53:57.038 > [DPL::loop] ******************* ENTER **********************
13:53:57.587 > [DPL::calcPowerLimit] power meter: 0 W, target consumption: 20 W, inverter output: 297 W, solar power (AC): 0
13:53:57.658 > [DPL::calcPowerLimit] match power meter with limit of 277 W
13:53:57.727 > [DPL::setNewPowerLimit] input limit: 277 W, lower limit: 20 W, upper limit: 800 W, hysteresis: 1 W
13:53:57.823 > [DPL::setNewPowerLimit] inverter max: 800 W, inverter is producing, requesting: 277 W, reported: 272 W, diff: 5 W
13:53:57.926 > [DPL::updateInverter] sending limit of 34.6 % (277 W respectively), max output is 800 W
13:53:58.951 > [DPL::updateInverter] actual limit is 34.0 % (272 W respectively), effective 2064 ms after update started, requested were 34.6 %

SW-Niko, 2024-05-08:
Hallo @DrSpaltenstein ,
Wenn dein Powermeter keine negativen Werte liefert dann frage ich mich woher OpenDTU-OnBattery weiß das gerade zu viel Energie eingespeist wird! Eigentlich müsste bei dir der Wert immer weiter bis auf 800W steigen. Aber anscheinend hast du Glück das irgendwann bei requesting: 277 W, reported: 256 W, diff: 21 W das Ganze zum Stillstand kommt.
Oder habe ich da einen Denkfehler?

SW-Niko, 2024-05-09:
Kommando zurück ... doch ein Denkfehler. Du hast Pech das es nicht funktioniert.
Weil du target consumption mit einem positiven Wert (20W) konfiguriert hast solle es theoretisch funktionieren.
Du hast nur den Nachteil das die Leistung in maximal 20W Schritten reduziert wird. Es dauert also etwas bis der Inverter die Leistung reduziert. Theoretisch ....
In Realität hast du Pech, weil jetzt auch noch die Fehler und Toleranzen die beteiligten Geräte in Spiel kommen.
In deinem Log kannst du sehen das der Inverter die eigentlich geforderten Werte nicht einstellt oder nicht einstellen kann oder die Messung im Inverter so große Toleranzen hat.
13:53:45.435 > [DPL::setNewPowerLimit] inverter max: 800 W, inverter is producing, requesting: 277 W, reported: 256 W, diff: 21 W

Ich würde als erstes mal die neueste OpenDTU-OnBattery version installieren. Wenn ich das richtig in Erinnerung habe gab es vor kurzem einen Fix der dein Problem vielleicht beheben könnte.

DrSpaltenstein, 2024-05-09:
Moinsen, ja habs auch mal anhand Quelltext von Hand durchgespielt, das mit den Null ist halt blöd, er zieht dann im besten Fall die Differenz zwischen target consumption und 0 ab. Aber eben nur im besten Fall wenn die Toleranz klein genug ist...Das funktioniert nur manchmal, meist bleibt er irgendwo hängen und saugt die Batterie leer. Da müsste tatsächlich noch irgend ein I-Anteil in den Limiter integriert werden damit er solche Fehler besser ausbügeln kann.
Einfach doof das der Zähler das nicht wenigstens ausgibt, zeigt zwar im Display "-" an wenn er rückspeist, aber die Power ist halt 0 und scheinbar wird das "-" auch nicht über das SML Protokoll übertragen sonst könnte man das ja als "Hinweis" nehmen die Leistung stärker zu reduzieren. Hatte noch das DC- Kabel direkt am HM800 über ein Y Kabel angeschlossen um wirklich gleiche Werte an den zwei MPPTs zu erhalten, damit lief es besser aber immer noch nicht zuverlässig (20240323 firmware).
Hab mir jetzt n shelly bestellt...

SW-Niko, 2024-05-09:
Ich bezweifle das ein höherer I-Anteil in der Regelschleife wesentlich dazu beiträgt die Toleranzen der P-Anteile auszubügeln.
Mit einem Shelly geht das sicherlich viel besser. Aber ich würde die neueste FW testen.
Es könnte sein das #872 dein Problem löst. Hast du einen 24VDC Akku?

schlimmchen, 2024-12-02:
> Edit: auch heute keine Auffälligkeiten, bei reduzierter Leistung.

Von uli-bs am 06. Mai zurückgemeldet und seither auch nichts mehr gehört.

Das Problem, dass Inverter an 24V Akkus nicht begreifen, was sie gerade tatsächlich produzieren, ist bekannt und kann auch nicht umgangen werden im DPL. Lediglich das drosseln, wie von uli-bs umgesetzt, hilft da. #899 kann auch helfen, aber das ist ein anderes Thema und wird im erwähnten Issue verfolgt.

closed 2024-12-02 by schlimmchen