hoylabs/OpenDTU-OnBattery #947: Load current of Victron charge controllers neither displayed nor published

schlimmchen, 2024-05-02:
### What happened?

Charge controllers with a load output terminal can output significant power on this terminal. This power is not observable in the live view or MQTT.

### To Reproduce Bug

Need a charge controller with load output, see #867.

[image]

### Expected Behavior

Live view and MQTT (something else as well?) should show the load current/power.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

1234

SW-Niko, 2024-10-29:
I am in the process of implementing the value in Live view, MQTT and HASS.
If I read the code correctly, MPPT devices without a load output are currently shown as Load Output = OFF in the live view.

1. Can anyone confirm this? (I just have MPPT with load output)
2. Should I change it so that there is no display if there is no load output?

schlimmchen, 2024-10-30:
1. Yes. Mine have no load output, but "Load Output: OFF" is part of the VE.Direct message and hence this info is shown in the live view.
2. No, please leave it. If you know a way to reliably determine whether or not the charge controller has a load output, then display "Load Output: N/A". However, I think you cannot know without maintaining a hard-coded table, in which case we will just display what the charge controller tells us: "Load Output: OFF".

SW-Niko, 2024-10-31:
Ich muss jetzt mal die Sprache wechseln....

Also ... wenn ich die Victron Dokumentation richtig verstehe, dann werden die Daten LOAD (Lastausgang) und IL (Laststrom)
nur übertragen wenn der MPPT einen Lastausgang hat. Siehe Index 6 und 7.

[image]

Testen ob das wirklich so ist kann ich nicht. (Meine 2 MPPTs haben einen Lastausgang)
Ich vermute weiter das bei dir nur deshalb OFF angezeigt wird weil die Variable dazu mit 0 initialisiert wird und mangels Daten vom MPPT auch niemals umgestellt wird.

Wenn ich richtig liege, dann ist es sehr einfach herauszufinden ob der angeschlossene MPPT einen Lastausgang hat oder nicht.

So richtig gefällt mir die Anzeige "Load Output: N/A" nicht, weil es etwas in-consequent zu den anderen Daten ist.
Bis jetzt zeigen wir Informationen, die wir haben, einfach nicht an und übermitteln sie auch nicht über MQQT und HASS.
Darum ist mein persönlicher Favorit:
Wenn Lastausgang vorhanden ist dann LOAD und IL anzeigen und wenn nicht dann nicht.

Kannst du mal überprüfen ob die LOAD oder IL Daten bei dir über die TEXT-Nachrichten ankommen?

schlimmchen, 2024-10-31:
> Kannst du mal überprüfen ob die LOAD oder IL Daten bei dir über die TEXT-Nachrichten ankommen?

Naja, das hab ich doch selbstverständlich schon getan bevor ich meine Aussage dazu gemacht habe :wink: Das sind alle Daten, die von meinen Ladereglern ohne "echten" Lastausgang kommen:

```
20:01:58.776 > [VE.Direct MPPT 11/12] Text Data 'PID' = '0XA058'
20:01:58.778 > [VE.Direct MPPT 11/12] Text Data 'FW' = '163'
20:01:58.783 > [VE.Direct MPPT 11/12] Text Data 'SER' = 'HQ2238X6DPU'
20:01:58.786 > [VE.Direct MPPT 11/12] Text Data 'V' = '51700'
20:01:58.789 > [VE.Direct MPPT 11/12] Text Data 'I' = '0'
20:01:58.791 > [VE.Direct MPPT 11/12] Text Data 'VPV' = '180'
20:01:58.794 > [VE.Direct MPPT 11/12] Text Data 'PPV' = '0'
20:01:58.797 > [VE.Direct MPPT 11/12] Text Data 'CS' = '0'
20:01:58.801 > [VE.Direct MPPT 11/12] Text Data 'MPPT' = '0'
20:01:58.805 > [VE.Direct MPPT 11/12] Text Data 'OR' = '0X00000001'
20:01:58.810 > [VE.Direct MPPT 11/12] Text Data 'ERR' = '0'
20:01:58.812 > [VE.Direct MPPT 11/12] Text Data 'LOAD' = 'OFF'
20:01:58.814 > [VE.Direct MPPT 11/12] Text Data 'H19' = '197072'
20:01:58.817 > [VE.Direct MPPT 11/12] Text Data 'H20' = '55'
20:01:58.820 > [VE.Direct MPPT 11/12] Text Data 'H21' = '144'
20:01:58.823 > [VE.Direct MPPT 11/12] Text Data 'H22' = '145'
20:01:58.825 > [VE.Direct MPPT 11/12] Text Data 'H23' = '491'
20:01:58.827 > [VE.Direct MPPT 11/12] Text Data 'HSDS' = '139'
```

```
20:01:59.332 > [VE.Direct MPPT 10/9] Text Data 'PID' = '0XA073'
20:01:59.335 > [VE.Direct MPPT 10/9] Text Data 'FW' = '163'
20:01:59.337 > [VE.Direct MPPT 10/9] Text Data 'SER' = 'HQ2303CPMAR'
20:01:59.342 > [VE.Direct MPPT 10/9] Text Data 'V' = '51620'
20:01:59.344 > [VE.Direct MPPT 10/9] Text Data 'I' = '0'
20:01:59.347 > [VE.Direct MPPT 10/9] Text Data 'VPV' = '390'
20:01:59.350 > [VE.Direct MPPT 10/9] Text Data 'PPV' = '0'
20:01:59.353 > [VE.Direct MPPT 10/9] Text Data 'CS' = '0'
20:01:59.355 > [VE.Direct MPPT 10/9] Text Data 'MPPT' = '0'
20:01:59.358 > [VE.Direct MPPT 10/9] Text Data 'OR' = '0X00000001'
20:01:59.361 > [VE.Direct MPPT 10/9] Text Data 'ERR' = '0'
20:01:59.364 > [VE.Direct MPPT 10/9] Text Data 'LOAD' = 'OFF'
20:01:59.368 > [VE.Direct MPPT 10/9] Text Data 'H19' = '148030'
20:01:59.371 > [VE.Direct MPPT 10/9] Text Data 'H20' = '80'
20:01:59.374 > [VE.Direct MPPT 10/9] Text Data 'H21' = '208'
20:01:59.378 > [VE.Direct MPPT 10/9] Text Data 'H22' = '263'
20:01:59.380 > [VE.Direct MPPT 10/9] Text Data 'H23' = '957'
20:01:59.384 > [VE.Direct MPPT 10/9] Text Data 'HSDS' = '228'
```

Ich glaube den Grund zu erkennen: Mein MPPT hat zwar keinen direkt belastbaren Lastausgang, aber den kann man einstellen dass er mit seinem TX Pin der VE.Direct Schnittstelle etwas schaltet mit kleinem Strom. Das würde Sinn machen, wenn das als "Load Output State" kommuniziert würde, auch wenn es ein virtual load output ist.

[image]

> Wenn Lastausgang vorhanden ist dann LOAD und IL anzeigen und wenn nicht dann nicht.

Aber wie entscheidest du das? Bitte nicht mit einer Datenbank. Also anhand einfließender LOAD und IL Text-Daten?

> Ich vermute weiter das bei dir nur deshalb OFF angezeigt wird weil die Variable dazu mit 0 initialisiert wird und mangels Daten vom MPPT auch niemals umgestellt wird.

Ist das so? Was sagt der Code? Der Code sagt, du hast recht. Die Variable gibt es immer und daher wird die auch verwendet. Das ist natürlich Mist bei Werten, die nicht immer empfangen werden. Und bei der Verarbeitung des Load Currents IL wird das dann auf jeden Fall lästig, denn das überträgt mein Laderegler beispielsweise definitiv nicht. Die Strategie, die wir bisher schon benutzen ist ein `std::pair<uint32t, T>` für den jeweiligen Typ `T`, wobei das erste ein Zeitstempel ist und angenommen wird, dass der Wert nur dann tatsächlich mal empfangen und gesetzt wurde, wenn der Zeitstempel ungleich Null ist. Das sollten wir dann auch für LOAD und IL anwenden (noch mehr?).

Noch ein Hinweis: RELAY ist wieder etwas anderes. Das ist beispielsweise am MPPT 250|70 meines Vaters dran. Das ist kein Lastausgang, da kommt kein Strom raus, es ist nur ein Relaiskontakt. Dafür gibt es #1037, ein anderes Thema.

SW-Niko, 2024-11-01:
Virtueller Lastausgang. Jetzt wird die Sache rund. 👍 
Gut das wir uns ausgetauscht haben. Ich denke, jetzt kann ich mit dem implementieren anfangen.
Ich muss feststellen ... es ist gar nicht so einfach, funktionierenden Code zu schreiben, wenn man die richtige Hardware nicht oder nur teilweise hat. 

> Aber wie entscheidest du das? Bitte nicht mit einer Datenbank. Also anhand einfließender LOAD und IL Text-Daten?

Ja, mach ich mit den LOAD und IL Text-Daten über std::pair<uint32t, T>
Damit bleiben wir beim bereits verwendeten Konzept und das gefällt mir. 😄

closed 2024-11-03 by schlimmchen