hoylabs/OpenDTU-OnBattery #950: Cannot install Firmware 2024.05.03 via OTA --> [HTTP ERROR] Bad Request

mtlnet, 2024-05-04:
### What happened?

After Upload of of new Firmware 2024.05.03 via OTA I get the error message "[HTTP ERROR] Bad Request". Previous updates in the past were possible without problems. Last installed version is 2024.03.23. SketchFS is 1920 KB (used 1904 KB).

### To Reproduce Bug

Try to upload new Firmware 2024.05.03 (opendtu-onbattery-generic_esp32.bin) via OTA on the web interface.

### Expected Behavior

The DTU reboots and loads the new Firmware.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

d2990bd

saschaabraham, 2024-05-04:
Same here, git-hash of dtu: [f32b85c](https://github.com/helgeerbe/OpenDTU-OnBattery/commits/f32b85c)

danny1605, 2024-05-04:
I have the same problem with 
opendtu-onbattery-generic.bin 2024.05.03

[image]

ferdlgogosch, 2024-05-04:
Hatte das selbe Problem, mehrmals probiert immer selb Fehlermeldung. 
Dann machte ich einen dtu Neustart, danach hat es funktioniert.

ButterBetzi, 2024-05-04:
same here for me

danny1605, 2024-05-04:
> Hatte das selbe Problem, mehrmals probiert immer selb Fehlermeldung. Dann machte ich einen dtu Neustart, danach hat es funktioniert.

Hat bei mir leider mit einem Neustart nicht funktioniert.

mtlnet, 2024-05-04:
> Hatte das selbe Problem, mehrmals probiert immer selb Fehlermeldung. Dann machte ich einen dtu Neustart, danach hat es funktioniert.

Hat bei mir auch nicht geklappt. Weder soft- noch hardreboot.

ferdlgogosch, 2024-05-04:
Ich wollte noch hinzufügen das ich die Version,
opendtu-onbattery-generic_esp32s3_usb.bin
In Verwendung habe.

MaXxGrr, 2024-05-04:
Hier bei mir der gleiche Fehler - Version 2024.03.23 mit generic_esp32. Neustart hat nix gebracht... 

Grüazi, MaXx

schlimmchen, 2024-05-04:
Schade, dass es für so viele Leute nicht klappt.

@ferdlgogosch Dass bei dir ein Neustart geholfen hat, deutet darauf hin, dass dein Arbeitsspeicher zu stark ausgelastet oder fragmentiert war.

An die anderen: Ich kann das Problem auf meinem ESP32-WROOM-32D nachvollziehen. Ein bisschen konnte ich schon schauen, und meine erste Vermutung lautet, dass zwar das Bauen des firmware binary klappt, aber dass die endgültige Datei dennoch größer ist als es die Partition erlaubt. Leider bin ich mir nicht im Klaren, wo jetzt der Fehler liegt. Da muss ich erst genauer hinschauen.

Mir ist das Problem nicht aufgefallen, weil ich ein OTA Update nur auf einem ESP32-S3 probiert habe, und da klappt das, weil da mehr Luft ist in der Partition für die Firmware. Ansonsten habe ich mich darauf verlassen, dass die endgültige Firmware-Datei in die entsprechende Partition reinpassen wird, wenn platformio sagt, dass alles okay ist.

Das tut mir ehrlich Leid, dass ich euch eine Firmware hingelegt habe, die sich für die meisten User gar nicht installieren lässt. Auch die von platformio/esptool generierte factory-bin lässt sich nicht starten, wenn man sie seriell einspielt.

Ich bin mir einigermaßen sicher, dass sich das nicht ohne Weiteres umgehen lässt. Daher habe ich eine fette Warnung oben an das Release gepackt und die generic und generic_esp32 Binaries aus dem Release entfernt.

Ich versuche euch schnellstmöglich eine neue aktuelle Firmware zu veröffentlichen.

MaXxGrr, 2024-05-04:
Kein Problem - hat ja nix kaputt gemacht... Haben wir halt ein paar Tage mehr Zeit uns auf die Features und Fixes zu freuen :-D 

Grüazi, MaXx

Snoopy-HSS, 2024-05-05:
Gott sei dank.
Ich dachte schon ich werd blöd.
Auch per Serial Flash geht die Firmware nicht, kommt nur BOOT Loop

PR #958 (2024-05-06): Remove redundant libraries (FirebaseJson and Crypto)
The firmware for environments `generic` and `generic_esp32` (and probably more) finally became to large, see #950.

This changeset removes the FirebaseJson and the rweather/Crypto libs from the build and implements the functionality they provided using existing libs. This saves nearly 20kB of flash memory. The firmware now works again on non-S3 ESP32.

Closes #950.
closed 2024-05-06 by schlimmchen