hoylabs/OpenDTU-OnBattery #952: [Request] Monitor the connection quality with the Inverter(s)

Manos1966, 2024-05-04:
### Is your feature request related to a problem? Please describe.

I place this point for discussion, not sure whether it is relevant under normal conditions...
I have the experience of testing in environments where many DTUs (Hoymiles DTUs + OpenDTUonBattery) are active.

Depending on the transmition strength and the (shiort) distance between DTU and the inverter, you may experience a lot of "re-transmissions" (You can notice this by monitoring the messsages in the Console).

When such "re-transmissions" occur, there are delays to reaction-time of other processes. 
This could lead to the impression that OpenDTUonBattery has slow reaction time.

As most users keep the Battery near the inverter and the OpenDTUonBattery is connected to the Battery and Victron, the OpenDTUonBattery is very close to the inverter.

Not many users (in my opinion) are aware of this transmission-quality problem.

### Describe the solution you'd like

Open to suggestions.
I see two key areas:
- Reception of data from the Inverter
- Transmission of DPL data to the inverter
An visual indicator that could give an idea how good (fast, number of re-transmits) communication of these two areas is.

### Describe alternatives you've considered

I know it is funny but, it happened: Putting a brick between the OpenDTUonBattery and the inverter solves the problem.
Same as putting the OpenDTUonBattery in another room.

schlimmchen, 2024-05-04:
Good idea, I had it for months now. My approach would be to calculate the ratio of successful transmissions and (retransmits + failures). However, this was never important enough for me attempt, especially since we would need to touch the upstream code.

This request might actually be better placed upstream? All OpenDTU Users could benefit from such an indicator.

Manos1966, 2024-05-04:
Good Idea, I will place it upstream.

Would upstream also solve the DPL transmition-quality monitoring as well or, does this have to be done here?

schlimmchen, 2024-05-04:
Why would we need a separate indicator for the DPL? An no, upstream cannot do something related to the DPL, as the DPL is not an upstream feature.

Manos1966, 2024-05-04:
We need to know how many re-transmits were needed to send the updated Watt-Limit to the Inverter.

ALTHOUGH, we could deduce that, if the quality of Data Transmission from Upstream is poor, the same poor quality will be the case for the transmission of data from the DPL.

Manos1966, 2024-05-04:
Done: https://github.com/tbnobody/OpenDTU/issues/1963

closed 2024-05-04 by Manos1966