hoylabs/OpenDTU-OnBattery #966: Keine Daten vom Shelly3Em Powermeter mit Tasmota seit Update der FW auf Version 24.05.06

daToni56, 2024-05-07:
### What happened?

Powermeter ok mit FW 240323
http://192.168.1.71/cm?user=admin&password=ToEs2018&cmnd=Status%208
StatusSNS/ENERGY/Power/[0]
Nok mit FW 240506
Unable to find a value for phase 1 with JSON path "StatusSNS/ENERGY/Power/[0]"

### To Reproduce Bug

Fehler ist mit FW 240506 immer vorhanden
Retour auf FW 240323 wieder OK

### Expected Behavior

JSON Abfrage korrigieren

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.05.06

### Relevant log/trace output

```Shell
nicht notwedig
```

### Anything else?

danke

rudolfp1979, 2024-05-07:
have the same problem with https + json  [HttpPowerMeter] Unable to find a value for phase 1 with JSON path "/data/[2]/tuples/[0]/[1]"

rudolfp1979, 2024-05-07:
oh no remove FirebaseJson lib from firmware (save 17.5k of flash)
we used this library solely to interpret the answer of an HTTP web
server as JSON and find a particular value using a path expression in
the HTTP power meter implementation.

since we ran out of flash memory on non-S3 ESP32, we need to cut some
corners. removing FirebaseJson is the last low-hanging fruit that we
currently know of. we can get rid of it by using ArduinoJson (which is
already integral part of the firmware) and implementing a custom logic
to extract a value based on a path expression.

other than the FirebaseJson path "finder", the new implementation
only knows how to access sub-keys delimited by a forward slash. in
particular, accessing array members is not supported any more. I am
hoping that this is simply not an issue. if so, we will have users
complaining and we can add this functionality in a later release.

schlimmchen, 2024-05-07:
> I am hoping that this is simply not an issue. if so, we will have users complaining and we can add this functionality in a later release.

Well... Seems this is indeed in active use after all.

It should not be a big deal.

Please excuse my ignorance and stand by for a maintenance release.

Can somebody who is affected by this problem please share the JSON response from the respective HTTP server, so I can replicate the issue and test a fix? I guess I can craft a sample JSON myself, but it would be very helpful if I had a real-world-example, especially one that worked before but not after the release.

rudolfp1979, 2024-05-07:
here but i can reduce it
{ "version": "0.8.4", "generator": "vzlogger", "data": [ { "uuid": "a0e4ebc0-af8d-11ee-ada0-fb80aab23460", "last": 1715092105560, "interval": 0, "protocol": "sml", "tuples": [ [ 1715092105560, 2099441.2000000002 ] ] }, { "uuid": "bee5edd0-af8d-11ee-8404-81b945ee2f30", "last": 1715092105560, "interval": 0, "protocol": "sml", "tuples": [ [ 1715092105560, 175362.60000000001 ] ] }, { "uuid": "72f7bc00-af8d-11ee-bc18-4581c02378d5", "last": 1715092105560, "interval": 0, "protocol": "sml", "tuples": [ [ 1715092105560, 297.0 ] ] } ] }

to

{ "version": "0.8.4", "generator": "vzlogger", "data": [ { "uuid": "72f7bc00-af8d-11ee-bc18-4581c02378d5", "last": 1715092450558, "interval": 0, "protocol": "sml", "tuples": [ [ 1715092450558, 398.0 ] ] } ] }

but it still has two arrays in it.

rudolfp1979, 2024-05-07:
what about a second compile path with old firebasejson till people got a workaround?

PR #967 (2024-05-07): Fix: restore JSON array access for HTTP power meter
this implements accessing array members in an ArduinoJSON object following the FirebaseJson syntax. the FirebaseJson lib was previously removed to save flash memory, and logic was implemented to find a JSON node using the FirebaseJson path syntax, restoring the functionality. however, array access was not implemented.

this change also addresses leading and trailing and double forward slashes in the path expression.

moreover, much more expressive error messages are now generated in case the path could not be resolved.

Closes #966.
closed 2024-05-07 by schlimmchen