hoylabs/OpenDTU-OnBattery #971: SMA Energymeter als Erzeugungszähler

iroc-fabi, 2024-05-09:
### Is your feature request related to a problem? Please describe.

Die Gesamt PV Leistung stimmt nicht. 
[image]

### Describe the solution you'd like

Neben den drei HMS1600 betreibe ich noch zwei SMA Wechselrichter.
Alle fünf WR sind an einem Energymeter angeschlossen, der als Erzeugungszähler konfiguriert ist.

OpenDTU unterstützt ja bereits den Homemanager, diesen nutze ich auch. Siehe Screenshot. 
Toll wäre jetzt noch die Gesamt PV Leistung in openWB zu sehen. 
Man könnte die Daten auch per modbus aus den WR abrufen.

Snoopy-HSS, 2024-05-23:
Problem ist, man müsste 1-N Wechselrichter abfragen. Und nicht alle hängen per Speedwire im Netzwerk...
Das ein Emeter vor den WR hängt ist auch eher unüblich...

spcqike, 2024-05-24:
Ich würde es generell erst einmal zur Diskussion stellen, ob das ein typische Aufgabe für openDTU ist.

meiner Meinung nach ist die openDTU nur zur Auswertung und Steuerung der Hoymiles (und Derivate) gedacht. Ich sehe keinen Grund, warum es die Zentrale des gesamten Haushalts sein soll. 

Wozu genau wäre es denn nötig, dass die Leistung aller Wechselrichter, auch von nicht Hoymiles und daher nicht DTU Geräten, erfasst wird?

iroc-fabi, 2024-05-24:
Ich könnte die Daten auch per mqtt aus ioBroker an OpenDTU senden.
Zusätzlich ist auch openWB in Betrieb. Die WB holt sich auch die Daten per mqtt etc.

iroc-fabi, 2024-05-24:
> Problem ist, man müsste 1-N Wechselrichter abfragen. Und nicht alle hängen per Speedwire im Netzwerk...
> Das ein Emeter vor den WR hängt ist auch eher unüblich...

Du hast recht. Das ist äußerst unüblich. Musste ich aber letztes Jahr wegen den drei zusätzlichen Hoymiles WR machen. 
klar, ich hätte mir die Gesamtleistung auch in ioBroker zusammenrechnen lassen können, dann wären die Daten aber nicht im Sunnyportal 😉