hoylabs/OpenDTU-OnBattery #974: DPL: Maximales Leistungslimit wird bei Skalierung nicht beachtet

RoFranz, 2024-05-11:
### What happened?

Werden nicht alle Eingänge des Wechselrichters bedient, wird das maximale Leistungslimit nicht beachtet.

### To Reproduce Bug

1. Leistungslimit auf 1300 stellen (HMS 1600-4t) -> entspricht 81.25%
[image]

2. Werden alle 4 Eingänge bedient (angeschlossene Akkus sind ON) -> keine Skalierung -> Leistungslimit wird eingehalten
[image]

3. werden nur 2 Eingänge bedient (ein Akku OFF) -> Skalierung erfolgt korrekt -> Leistungslimit wird überschritten (100%) 
[image]

### Expected Behavior

Auch bei einer Skalierung wird das eingestellte Leistungslimit nicht überschritten

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2024.05.07

### Anything else?

Einfach danke an die fleißigen Programmierer und die Bereitschaft auch solche kleinen Bugs zu verbessern.
DANKE!

spcqike, 2024-05-11:
Ich verstehe das Problem noch nicht ganz. 

800W AC sind doch <1300W Limit. Dass der DPL bei 2 von 4 aktiven Eingängen entsprechend höhere Werte einstellt, damit überhaupt ansatzweise 800W raus kommen, ist doch Sinn der Skalierung.

RoFranz, 2024-05-11:
Danke für die Nachfrage und entschuldige, wenn ich es nicht gleich verständlich erklären konnte.
An dem Wechselrichter sind zwei Akkus (BP 2500 version 1), welche maximal 400W je Ausgang ausgeben können - ansonsten erfolgt "Notabschaltung".  Wenn der Wechselrichter mit 100% als Limit betrieben wird, können am Eingang des Wechselrichters durchaus mehr als die rechnerisch 400W (1600W durch 4) eingespeist werden. Zudem wie du bereits gesagt hast, kommt es sowieso zu einer Differenz zwischen Ausgangsleistung Akku <-> Eingangsleistung WR. Somit kommt es häufig zu einer Überlastung der Akkuausgänge (weil der WR mehr zieht als eingestellt). Dieses Zusammenspiel ist aber auch nur "mein" Problem und kann mittels einer IF-Abfrage innerhalb der DPL.cpp bisher auch gelöst werden.
Aber GENERELL: Sollte das eingestellte Leistungsmaximum des WR nicht unabhängig von der Einspeisung NIE überschritten werden. Ansonsten macht doch das Leistungslimit keinen Sinn, wenn es nicht GENERELL gilt?

spcqike, 2024-05-11:
> Aber GENERELL: Sollte das eingestellte Leistungsmaximum des WR nicht unabhängig von der Einspeisung NIE überschritten werden. Ansonsten macht doch das Leistungslimit keinen Sinn, wenn es nicht GENERELL gilt?

ist glaube ich ne Philosophie Frage. Für mich war das bisher ein maximales AC ausgangslimit. Und das wird, soweit ich weiß, eingehalten. Mit Skalierung und auch ohne.

RoFranz, 2024-05-11:
Ich denke dir für die Korrektur meines Blickwinkels!
Das es sich lediglich auf die AC Seite bezieht, steht auch bei euch so in der Erklärung.
Wer lesen kann, ist klar im Vorteil!
Danke!

closed 2024-05-11 by RoFranz