hoylabs/OpenDTU-OnBattery #998: documentation - example 4 misleading

typxxi, 2024-05-22:
### What happened?

"With Battery and AC charging via Huawei Rectifier (using the Huawei AC charger in combination with the CMT2300A radio board for HMT- HMS- Inverters is not supported at the moment)"

for HMT- HMS- Inverters is not supported 
is misleading, cause I guess it should mean that it is supported for HMT inverters, but HMS inverters are not supported

### To Reproduce Bug

as mentioned above

### Expected Behavior

as mentioned above

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

independent

### Anything else?

I suggest to clarify the example and its purpose which hoymiles are supported now or not cause in the picture itself there is no further hint like "hoymiles but no hms inverter.

And then it would also make sense to describe the purpose a bit more cause Null Einpeisung is mentioned everywhere but I bet the rectifier is used for such a big battery during winter time to charge the battery in the night with tibber / octopus. 

I know that HUAWEI charger quite well from a growatt installation where that is used as battery charge booster where the Growatt hybrid is limited to 3 kW only. And if you use the Huawei then you get about 6 kW for a 15 kWh batter which is less then 1/2 C rating and quite OK.

Finally: This would be a very good example in all those cases where you need many different sizes of solar panels and shorter strings that would not be sufficient enough to charge a 48 V battery from Victron mppt which requires about 57 volts at least.
But for a hoymiles HMS-2000-4T it would be easy to support 4 different kind of panels. Then the AC DC charger is needed to charge the battery with the surplus and you would be able to solve the topic with 1 inverter instead of 4 struggling mppt. ( in such case with many rather small spots each with a slightly different orientation and pitch. We needed to buy for each spot a different panel and even used ones to benefit from those spots the best way (max output). Where do you get a 130 cm x 30 cm panel ? Only from 2002 where there had been a few 50 Wp panels and still working properly. 50 Wp more or less in every spot makes in our uses case in total a 340 Wp difference or about 300 kWh a year. Those panels had cost us 50€ in total, but the mppt became the real challenge and mess if you want to charge a 48 V system where the HMS-2000-4T is a great solution.

Manos1966, 2024-05-22:
> for HMT- HMS- Inverters is not supported
is misleading, cause I guess it should mean that it is supported for HMT inverters, but HMS inverters are not supported

Hmm... I see what you mean.
Would "for HMT/HMS- Inverters is not supported" solve the issue?

Manos1966, 2024-05-22:
> 4 struggling mppt. ( in such case with many rather small spots each with a slightly different orientation and pitch. We needed to buy for each spot a different panel and even used ones to benefit from those spots the best way (max output). 

I think you may find this useful:
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Victron-MPPT-solar-charger#how-many-panels-can-i-connect-to-one-victron-mppt-

Manos1966, 2024-05-26:
@typxxi Have we finished with this?

Then please CLOSE it.

typxxi, 2024-05-30:
I would suggest to make sure that these are 2 different productlines that are not supported yet by adding "AND" . 
I had misunderstood that or misread it cause I had mixed it up with the HM series (I had thought there were only 2 series cause I was not aware that the HMT was a third series for the 3 phase series . I am new to Hoymiles.

"for HMT- and HMS-inverters" should make it clear that these are not supported

Manos1966, 2024-05-30:
> "for HMT- and HMS-inverters" should make it clear that these are not supported

done 👍

schlimmchen, 2024-08-18:
As far as I can tell, this issue is resolved, so I will be closing it. Please come forward if I misunderstood.

closed 2024-08-18 by schlimmchen