hoylabs/OpenDTU-OnBattery #999: [Request] Support for OpenDTU Fusion CAN/Iso Shield with Huawei PSU

drahdiwaberl, 2024-05-22:
### Describe the solution you'd like

there's a new - potentially simpler and safer - option for adding a CAN-bus with
https://github.com/markusdd/OpenDTUFusionDocs/blob/main/CANIso.md
using a TI SN65HVD230DR

it would be nice to have the option to configure a Huawei PSU connected to this.

drahdiwaberl, 2024-05-22:
there's an esphome component for connecting that way
https://github.com/mb-software/esphome-huawei-r4850

AndreasBoehm, 2024-08-08:
I think that this should not be too big of a deal but we will need someone that can write code and has the hardware. I thought about buying a Huawei PSU but I am not sure if i will actually use it.

drahdiwaberl, 2024-08-08:
assuming you're the one who could write code? ;)

schätz mal du würdest es easy wieder anbringen.. meins rennt schon ziemlich viel, hab aber auch einiges an ac-überschuss aus 800+1500W microwechselrichtern - mal schaun wie's dann mit weniger sonne und dafür extremeren schwankungen beim dynamischen import-preis ausschaut..

schlimmchen, 2024-08-09:
> assuming you're the one who could write code? ;)

eeeeer, yes?! one of only two actually active contributors right now. sooo... where is your pull request?

LennartF22, 2024-10-06:
@schlimmchen If there is need for it, I could implement this. I don't have a Huawei PSU, but I could do some mocking on the CAN bus and at least make sure that there is nothing obvious broken before giving a build to someone to test.

schlimmchen, 2024-10-06:
> If there is need for it

@LennartF22 Well, I am not sure if "need" is the right word here, but this is a valid request and implementing it would be highly appreciated, so I would certainly want to merge it. I would love to use the AC charging feature myself one day and using the OpenDTU Fusion CAN shield would definitely be a great plus. So please go ahead :blush: I can probably contribute some matching updates to the documentation.

drahdiwaberl, 2024-10-06:
and well (again), it would be the only way to allow NRF+CMT+huaweiAC, isn't it?

btw i'm thinking about getting a second huaweiAC to spread load on two phases - which a. will probably make my next feature request ;) and b. allow me to more easily test if anyone starts working on this..

schlimmchen, 2024-10-06:
> and well (again), it would be the only way to allow NRF+CMT+huaweiAC, isn't it?

No, as @LennartF22 is also hard at work with the SpiManager, which will eventually allow to use this combination -- as far as I understand.

LennartF22, 2024-10-06:
@drahdiwaberl With the new `SpiManager` it is possible to support up to three devices on the same SPI master peripheral. The driver for the SPI devices must support this though:

- nRF: not supported yet, but in the works
- CMT: supported
- W5500 (Ethernet): supported
- Huawei-CAN (via MCP2515): not supported yet, but in the works

Devices that don‘t support shared SPI buses yet still need a dedicated SPI bus. Therefore, nRF + CMT (+ W5500) + Huawei-CAN currently need 3 SPI buses, which is one too many. In the future, nRF + CMT + Huawei-CAN will work via the same SPI bus, so that one bus is completely free for 1/3 more devices, depending on whether these additional devices support sharing.

Still, implementing support for ESP32’s internal CAN controller makes sense, not only because of the Fusion CAN/ISO shield, also because it’s a bit absurd that one must add an external CAN controller while the ESP32 already comes with it. Only when a second CAN bus is needed, one should have to add an additional, external CAN bus controller.

@schlimmchen I will start working on the implementation next week. I just wanted to make sure that nobody is working on that yet.

drahdiwaberl, 2024-10-06:
> because it’s a bit absurd that one must add an external CAN controller while the ESP32 already comes with it.

word!  🤗

also - while it's nice that they exist and are available - all these extra boards and cheap components are potential points of failure in the whole system which makes the systems less reliable and errors possibly hard to find..

schlimmchen, 2024-12-23:
@LennartF22 Are you far along with adjusting the MCP2515 driver for SpiManager? I started but did not get far, yet. The MCP2515 lib is a hot mess (my opinion) and I am planning to re-implement the parts that are needed to support the Huawei PSU. skippermeister's fork includes a version compatible with SpiManager, however it's unrecognizable from the original lib and I am unsure if it even works, as his TWAI implementation was broken in at least two places... Maybe I can copy'n'paste some tricky bits.

drahdiwaberl, 2024-12-23:
might add this https://github.com/mjpalmowski/CAN-BUS-control-R4875G1-with-ESPHome-and-MQTT

also using SN65HVD230 transceiver, but seems to have gotten further with reversing the can bus, even seems to have worked out how to identify and likely even assign IDs to multiple devices on the bus..

schlimmchen, 2024-12-23:
@Manos1966 pointed out that project to me as well. I will gladly take a look. For now, I am happy to be able to soon release VP203 support to talk to the Huawei PSU at all.

closed 2025-01-06 by schlimmchen